thesofproject/sof #3966: [BUG] [TGL] [tgl-012-drop-stable] Wrong source/sink settings during ALSA conformance test

RDharageswari, 2021-03-29:
Name of component: SOF/ALSA

Brief: When doing ALSA test for SOF built sound-open-firmware-private branch, codec_adapter receives wrong source/sing parameters.

Detailed:
Usual way of doing alsa conformance test is running python script:
`./alsa_conformance_test.py -D hw:0,0 --rate-err-criteria 100 --log-file alsa_test_log.txt`
Output of this command is attached: alsa_test_output_.txt.
From alsa_test_log.txt is is clear what sequence of executions of alsa_conformance_test performs python script.
Also it is clear that test `snd_pcm_hw_params` fails when setting sample format S24_LE and S32_LE, but is passing on S16_LE.

INFO:root:Set format S16_LE
INFO:root:Execute command: alsa_conformance_test -P hw:0,0 -d 0.1 -r 48000 -c 2 -f S16_LE -p 240
INFO:root:pass
INFO:root:Set format S24_LE
INFO:root:Execute command: alsa_conformance_test -P hw:0,0 -d 0.1 -r 48000 -c 2 -f S24_LE -p 240
INFO:root:fail - snd_pcm_hw_params: Input/output error
INFO:root:Set format S32_LE
INFO:root:Execute command: alsa_conformance_test -P hw:0,0 -d 0.1 -r 48000 -c 2 -f S32_LE -p 240
INFO:root:fail - snd_pcm_hw_params: Input/output error

After investigating it for a bit I can tell that:
1. if executing same sequence of calls that python script does, but with 5 seconds time gap between them, there is no smd-pcm-hw-params error.
So this means when they are executed rapidly one after anothe, something changes.
2. building debug SOF with logs shows exactly the issue, sink and source buffers that arrive to codec_adapter have different format which is not expected.

> alsa_conformance_test -P hw:0,0 -d 0.1 -r 48000 -c 2 -f S16_LE -p 240
source and sink are 16 bit
> alsa_conformance_test -P hw:0,0 -d 0.1 -r 48000 -c 2 -f S24_LE -p 240
source is 16 bit and sink is 24 bit
> alsa_conformance_test -P hw:0,0 -d 0.1 -r 48000 -c 2 -f S32_LE -p 240
source is 16 bit and sink is 32 bit

this looks like a bug, no matter the time difference between alsa_conformance_tests execution, result should be the same, as do the stream parameters that comes to codec_adapter.
[alsa_test_log.txt](https://github.com/thesofproject/sof/files/6218625/alsa_test_log.txt)
[alsa_test_output.txt](https://github.com/thesofproject/sof/files/6218627/alsa_test_output.txt)

keyonjie, 2021-03-29:
Hi @RDharageswari do you know where do the source and sink format come from?

Since it works if start test with waiting more than 5 seconds, it looks to me the the previous playback doesn't stop/reset correctly and the subsequent playback in different format failed at .hw_params().

@RDharageswari can you share the FW SOF log at the failed case?

RDharageswari, 2021-03-29:
Following is the update from waves:
This issue is not 100% reproduction rate.
Got sometimes same sink and source format when running alsa test.
Adding logs to volume component didn`t help much because volume is below stream with respect to codec adapter, and after codec adapter failed no logs from Volume
WIll check without codec-adapter.
After codec-adapter is removed from topology volume component receives both sink and source buffers with the same format
[sof-log.txt](https://github.com/thesofproject/sof/files/6223609/sof-log.txt)
[dmesg-log.txt](https://github.com/thesofproject/sof/files/6223611/dmesg-log.txt)

RDharageswari, 2021-03-29:
[0001-WIP-Log-volume-source-and-sink-format.patch.txt](https://github.com/thesofproject/sof/files/6223612/0001-WIP-Log-volume-source-and-sink-format.patch.txt)

RDharageswari, 2021-03-30:
Feedback from waves is that the issue is still reproducible with the above patch

keyonjie, 2021-03-30:
Hi @RDharageswari someone should be able to debug this as the error is already in the FW logs:
```
[1903502123.750000] (       22.500000) c1 idc                ......./intel/cavs/idc.c:166  idc_do_cmd()
[1903502142.447917] (       18.697916) c1 codec_adapter 1.1   ......../codec_adapter.c:189  codec_adapter_prepare() start
[1903502164.270833] (       21.822916) c1 codec_adapter 1.1   ........../codec/waves.c:241  ERROR waves_effect_check() source 1 sink 2 sample format mismatch
[1903502188.333333] (       24.062500) c1 codec_adapter 1.1   ........../codec/waves.c:645  ERROR waves_codec_prepare() failed -22
[1903502212.031250] (       23.697916) c1 codec_adapter 1.1   ......../codec/generic.c:227  ERROR codec_prepare() error -22: codec specific prepare failed, codec_id 0x57410100
[1903502235.625000] (       23.593750) c1 codec_adapter 1.1   ......../codec_adapter.c:219  ERROR codec_adapter_prepare() error ffffffea: codec prepare failed
[1903502272.656250] (       37.031250) c0 pipe         1.9       src/audio/pipeline.c:690  ERROR pipeline_prepare(): ret = -5, dev->comp.id = 0
[1903502294.635417] (       21.979166) c0 ipc                       src/ipc/handler.c:285  ERROR ipc: pipe 1 comp 0 prepare failed -5
[1903502316.927083] (       22.291666) c0 pipe         1.9       src/audio/pipeline.c:964  pipe reset
```

slawblauciak, 2021-03-30:
@stolx hi, could you please help to figure this one out if it's an issue with the Waves module?

juimonen, 2021-03-30:
so this topology is defining a format range for the pcm like from 16 bit LE to 32 bit LE. So alsa conformance test is testing that it supports all these formats. If there is no any other component before the codec adapter that will do the conversion, the codec adapter needs to accept different formats.

stolx, 2021-03-30:
@slawblauciak @juimonen
Waves codec implementation supports all frame sizes of SOF 16, 24_4LE and 32.
But we expect the source and sink to have the same format.
The issue is that sometimes during ALSA test we get sink and source of different format and check in our codec_adapter API implementation fails.
The exact place of the check:
https://github.com/thesofproject/sof/blob/main/src/audio/codec_adapter/codec/waves.c#L239

So this bug is actually two independent questions:
1. Why this happens only sometimes during ALSA tests, maybe some race conditions are involved?
2. Should codec that implements codec_adapter interface support source and sink of different format?

keyonjie, 2021-03-30:
@stolx usually the input format come from the ALSA PCM param while the output format could inherit from the BE DAI so they can be different, you need to check if you have made the BE DAI in your tplg file fixed, if so, you might need to add a format converter (e.g. volume) around your codec component.

cujomalainey, 2021-03-31:
The codec adapter should be passing the [params down the sink](https://github.com/thesofproject/sof/blob/main/src/audio/codec_adapter/codec_adapter.c#L283) I wonder if we have an order issue here where params and prepare and being called in different orders since we [cache the params](https://github.com/thesofproject/sof/blob/main/src/audio/codec_adapter/codec_adapter.c#L289) in the params callback then update at least one of our buffers in the [prepare callback](https://github.com/thesofproject/sof/blob/main/src/audio/codec_adapter/codec_adapter.c#L266). Or better yet @stolx can you try scheduling the waves component on core 0 and retesting? I wonder if there is a race condition between the cores (see log to see the tasks flipping back and forth)
```
1664379254.895833] (     5121.250000) c0 pipe         2.16      src/audio/pipeline.c:556  pipe params dir 0 frame_fmt 1 buffer_fmt 0 rate 48000
[1664379273.072917] (       18.177084) c0 pipe         2.16      src/audio/pipeline.c:560  pipe params stream_tag 1 channels 2 sample_valid_bytes 3 sample_container_bytes 4
[1664379442.083333] (      169.010422) c1 idc                ......./intel/cavs/idc.c:58   idc_irq_handler(), IPC_IDCTFC_BUSY
[1664379464.062500] (       21.979166) c1 idc                ......./intel/cavs/idc.c:166  idc_do_cmd()
[1664379596.250000] (      132.187500) c0 dai          2.15           src/audio/dai.c:351  dai_playback_params() dest_dev = 2 stream_id = 0 src_width = 4 dest_width = 4
[1664379613.697917] (       17.447916) c0 dai          2.15           src/audio/dai.c:357  dai_playback_params() fifo 0x77010
[1664379646.250000] (       32.552082) c0 pipe         2.16      src/audio/pipeline.c:683  pipe prepare
[1664379691.562500] (       45.312500) c1 idc                ......./intel/cavs/idc.c:58   idc_irq_handler(), IPC_IDCTFC_BUSY
[1664379714.010417] (       22.447916) c1 idc                ......./intel/cavs/idc.c:166  idc_do_cmd()
[1664379735.625000] (       21.614584) c1 codec_adapter 2.12  ......../codec_adapter.c:189  codec_adapter_prepare() start
[1664379757.239583] (       21.614584) c1 codec_adapter 2.12  ........../codec/waves.c:241  ERROR waves_effect_check() source 0 sink 1 sample format mismatch
[1664379781.197917] (       23.958334) c1 codec_adapter 2.12  ........../codec/waves.c:645  ERROR waves_codec_prepare() failed -22
[1664379804.531250] (       23.333334) c1 codec_adapter 2.12  ......../codec/generic.c:227  ERROR codec_prepare() error -22: codec specific prepare failed, codec_id 0x57410100
[1664379827.760417] (       23.229166) c1 codec_adapter 2.12  ......../codec_adapter.c:219  ERROR codec_adapter_prepare() error ffffffea: codec prepare failed
[1664379862.291667] (       34.531250) c0 pipe         2.16      src/audio/pipeline.c:690  ERROR pipeline_prepare(): ret = -5, dev->comp.id = 11
[1664379884.322917] (       22.031250) c0 ipc                       src/ipc/handler.c:285  ERROR ipc: pipe 2 comp 11 prepare failed -5
[1664379906.406250] (       22.083334) c0 pipe         2.16      src/audio/pipeline.c:964  pipe reset
```

stolx, 2021-04-01:
When everything is on core 0 all tests are "pass"

lgirdwood, 2021-04-02:
Some observations:
1) when it works after 5 seconds it means that FW has done a D0 -> D3 -> D0 transition (and all configurations are reset to defaults).
2) works on core 0 and doesn't work on other cores usually means cache coherency.

It could be that the params are entering the DSP via core 0 (for the IPC handling) and then they propagate through the pipeline, but there needs to be a writeback and invalidate everytime the config crosses a core.

cujomalainey, 2021-04-02:
@lgirdwood that is likely true, given the majority of the pipeline was scheduled on core 0 and this single component was scheduled on core 1.

lgirdwood, 2021-04-13:
@keyonjie @slawblauciak any update here, I'm suspicious that cache coherency could also be a factor in some of the other tgl-012 issues. ?

mrajwa, 2021-04-26:
Is this issue still present? Today I checked on my end, the FW branch I used was tgl-012-drop-table but with linux kernel (5.9.0).
I don't reproduce this issue. Codec adapter was scheduled on core 1 while the rest of the pipeline on 0(master) core. Each time the format has changed, the stream params followed that.
I asked @zrombel to verify this on the slim driver as well, @zrombel once you have the result please share them with us here.

cujomalainey, 2021-04-26:
Will check again, but I have not see a specific fix for this on tgl-012-drop-stable yet

mrajwa, 2021-04-26:
@cujomalainey, I have just started working on codec adapter performance improvements (including your feedback regarding double state machine, config apply etc.) So better freeze your tests for a day, until I push my pattches.

cujomalainey, 2021-04-26:
Noted, thanks

lgirdwood, 2021-05-10:
Could this also be linked to #4104

cujomalainey, 2021-05-12:
I dont think so since this fails before we get to playback stage. But I think I have an idea what might be happening here. I have two possible theories. Unfortunately the author of the `idc` module is no longer around to answer.

First theory
1. Caches not being invalidated when handling params. Looking through the code I don't see any cache coherency when running params callbacks. Given it looks like params are inferred from buffers this seems like a possible problem.
2. idc sync on params order. I have to run more tests on this one, but if idc is not blocking downstream components from running their callback till upstream components on separate cores are done then they will pull bad params from the buffer.

From what I have seen, changing a whole pipeline removes the above issues which would align with the nature of the bugs. Still need to do more testing on the whole pipeline change though.

lgirdwood, 2021-05-20:
@slawblauciak @mwasko fyi - lets get in for rc3 and TGL13 once @mrajwa is back from vacation.

lgirdwood, 2021-05-27:
@dbaluta @mrajwa @mwasko @RDharageswari any update here ? Have we solved this in TGL13 or in main ?

keyonjie, 2021-06-02:
@dbaluta can you re-test this with tgl-013-drop-stable branch?

cujomalainey, 2021-06-02:
@stolx FYI

keyonjie, 2021-06-04:
@stolx @RDharageswari please hep to re-try as we don't have a way to reproduce it.

cujomalainey, 2021-06-04:
I can try either today or monday hopefully

cujomalainey, 2021-06-08:
forgot to write here as well my results, initial tests look promising

keyonjie, 2021-06-08:
> forgot to write here as well my results, initial tests look promising

Thanks for the feedback, waiting for your final result.

keyonjie, 2021-06-11:
@cujomalainey please help to close it if it doesn't reproduce with tgl-013-drop-stable.

cujomalainey, 2021-06-11:
In my testing so far this has been good, i think its safe to close unless we see it pop up again, but i think it might be worth adding single core components into the testing pipeline to detect this in CI in the future

closed 2021-06-11 by cujomalainey

keyonjie, 2021-06-11:
Thanks @cujomalainey , yes component level multi-core will be next declared support, maybe delivered with Zephyr together.