thesofproject/sof #3987: [BUG] Fail to compile waves codec support with main branch

dbaluta, 2021-03-30:
**Describe the bug**

Selecting CONFIG_WAVES_CODEC=y and compiling the SOF firmware I get:

```
/home/daniel/work/sof_dir/sof/src/audio/codec_adapter/codec/waves.c:12:10: fatal error: MaxxEffect/MaxxEffect.h: No such file or directory
 #include "MaxxEffect/MaxxEffect.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
```
@cujomalainey  @stolx is this as expected?

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
Always.

**Expected behavior**
Code should compile fine even the proprietary waves algorithm is not provided. Eventually the code linking should fail! Using a stub library should allow WAVES codec support to compile file

**Impact**
This is just annoying because I want to test compile changes on waves support. Also, CI should be able to compile the code.

cujomalainey, 2021-03-30:
This is WAI. Like the maxim lib, you an external lib to compile this

cujomalainey, 2021-03-30:
Although i agree, for future we should be stubbing this out for compile testing, but that is not our focus right now

closed 2021-03-30 by cujomalainey