thesofproject/sof #4007: [BUG] sof-hda-dsp not recognized by ALSA after upgrade (1.6.1 to 1.7)

tchici, 2021-04-07:
**Describe the bug**

Unable to set my laptop's internal speakers or headphone jack as an output because they are not being recognized by ASLA.

**To Reproduce**

No steps to reproduce. Audio device is not recognized immediately after boot.

**Reproduction Rate**

100%

**Expected behavior**

I expected to be able to output audio to my laptop's internal speakers or headphone jack and be able to control the output with something like `alsamixer`.

**Impact**

Not able to hear anything (unless plugging in an external USB audio device).

**Environment**

Apologies in advance. I'm unsure of how to answer this part. I am basically using the default Arch Linux kernel (5.11.11-arch1-1), and I am not sure which branches it is based off (the kernel and its modules). Also, I am not sure what a topology file is or what platform means in this context. I think platform is Whiskey Lake?

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**

Here is `alsa-info.sh` output:

After upgrade:
```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.4.65
!!################################

!!Script ran on: Wed Apr  7 02:32:35 UTC 2021

!!Linux Distribution
!!------------------

Arch Linux \r (\l) DISTRIB_ID=Arch DISTRIB_DESCRIPTION="Arch Linux" NAME="Arch Linux" PRETTY_NAME="Arch Linux" ID=arch HOME_URL="https://www.archlinux.org/" DOCUMENTATION_URL="https://wiki.archlinux.org/" SUPPORT_URL="https://bbs.archlinux.org/" BUG_REPORT_URL="https://bugs.archlinux.org/" LOGO=archlinux

!!DMI Information
!!---------------

Manufacturer:      HP
Product Name:      HP ENVY Laptop 13-aq0xxx
Product Version:   Type1ProductConfigId
Firmware Version:  F.13
System SKU:        6BG33EA#ABU
Board Vendor:      HP
Board Name:        85E3

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI0003:00/status 	 15
/sys/bus/acpi/devices/ACPI000C:00/status 	 15
/sys/bus/acpi/devices/ELAN2513:00/status 	 15
/sys/bus/acpi/devices/HPIC0003:00/status 	 15
/sys/bus/acpi/devices/HPQ6001:00/status 	 15
/sys/bus/acpi/devices/INT0E0C:00/status 	 15
[... 1982 lines cut ...]
[    3.898221] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.898228] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
--
[    4.016485] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    4.043904] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    4.043912] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.043915] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.043919] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    4.043921] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    4.043923] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    4.295882] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    4.295892] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    4.296100] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
--
[    4.296174] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[    4.302716] snd_hda_codec_hdmi ehdaudio0D2: Monitor plugged-in, Failed to power up codec ret=[-13]
[    4.319989] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input34
[    4.320048] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input35
[    4.320101] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input36
[    4.320153] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input37
[    4.320208] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input38
[    4.407748] wlan0: authenticate with 84:d8:1b:c8:8f:c0
```

Tell me if more information is needed (and where to get it from).

fbernier, 2021-04-07:
Also experiencing loss of sound since going from 1.6.1 to 1.7. More details can be found [here](https://bbs.archlinux.org/viewtopic.php?pid=1966481#p1966481)

AviusX, 2021-04-07:
Happens to me too. Just downgraded back to 1.6.1 and everything works again. Upgrading to 1.7 breaks all audio and the only detected sound card is "Dummy Output".

tumido, 2021-04-07:
Same here, Arch with 5.11.11 kernel on Thinkpad Carbon X1 7th gen

journalctl:
```
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 4
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-cnl.ri failed with error -2
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-cnl.ri failed err: -2
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
Apr 07 10:10:36 stonefly kernel: sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -2
```

Downgrade helped.

w4v3, 2021-04-07:
Adding `snd_hda_intel.dmic_detect=0` to the kernel parameters also fixed this without downgrading for me on Arch 5.11.11, Lenovo Yoga 7i.

heftig, 2021-04-07:
This looks like a downstream packaging bug; please close.

balazs4, 2021-04-07:
same story but on `Lenovo X1 Carbon Gen 8`

```bash
# lspci | grep audio
00:1f.3 Audio device: Intel Corporation Comet Lake PCH-LP cAVS
```

Downgrading to 1.6.x helped.

tchici, 2021-04-07:
> This looks like a downstream packaging bug; please close.

Looks like you're right. I wasn't sure how to confirm this myself at first since I had no idea how to install this directly from github. All there is under deployment is a todo.

For reference, here is the downstream bug report: https://bugs.archlinux.org/task/70327

closed 2021-04-07 by tchici

plbossart, 2021-04-07:
> Adding `snd_hda_intel.dmic_detect=0` to the kernel parameters also fixed this without downgrading for me on Arch 5.11.11, Lenovo Yoga 7i.

it also removes support for your microphones. Not so good.