thesofproject/sof #4018: Processing loop in codec_adapter

dbaluta, 2021-04-09:
Hi @mrajwa 

I need your opinion on how to fix / improving processing loop in codec_adapter.

The code is like this:

```
397 »       while (bytes_to_process) {
398 »       »       /* Proceed only if we have enough data to fill the lib buffer
399 »       »        * completely. If you don't fill whole buffer
400 »       »        * the lib won't process it.
401 »       »        */
402 »       »       if (bytes_to_process < codec_buff_size) {
403 »       »       »       comp_dbg(dev, "codec_adapter_copy(): source has less data than codec buffer size - processing terminated.");
404 »       »       »       break;
405 »       »       }
406 
407 »       »       buffer_invalidate(source, codec_buff_size);
408 »       »       codec_adapter_copy_from_source_to_lib(&source->stream, &codec->cpd,
409 »       »       »       »       »       »             codec_buff_size);
410 »       »       codec->cpd.avail = codec_buff_size;
411 »       »       ret = codec_process(dev);
412 »       »       if (ret) {
413 »       »       »       comp_err(dev, "codec_adapter_copy() error %x: lib processing failed",
414 »       »       »       »        ret);
415 »       »       »       break;
416 »       »       } else if (codec->cpd.produced == 0) {
417 »       »       »       /* skipping as lib has not produced anything */
418 »       »       »       comp_err(dev, "codec_adapter_copy() error %x: lib hasn't processed anything",
419 »       »       »       »        ret);
420 »       »       »       break;
421 »       »       }
422 »       »       codec_adapter_copy_from_lib_to_sink(&codec->cpd, &local_buff->stream,
423 »       »       »       »       »       »           codec->cpd.produced);
424 
425 »       »       bytes_to_process -= codec->cpd.consumed;
426 »       »       processed += codec->cpd.consumed;
427 »       »       produced += codec->cpd.produced;
428 »       }
429 
430 »       if (!produced && !cd->deep_buff_bytes) {
431 »       »       comp_dbg(dev, "codec_adapter_copy(): nothing processed in this call");
432 »       »       goto end;
433 »       } else if (!produced && cd->deep_buff_bytes) {
434 »       »       goto db_verify;
435 »       }
436 
437 »       audio_stream_produce(&local_buff->stream, produced);
438 »       comp_update_buffer_consume(source, processed);
```

mrajwa, 2021-04-09:
Hi Daniel, do you have some actual problem we can reference to? I would like to get some context here.

dbaluta, 2021-04-09:
@mrajwa I have updated the description. Please read again, maybe is more clear. Will try to summarize:

1) First problem is that if the processing loops goes more than 1 iteration it will copy wrong data to the input buffer because we update the source and local_buff outside the loop. This is fixable by moving the update of source/local_buff inside the loop.

2) Which is more problematic, is that if the processing loop goes more than 1 iteration and the processing is heavy (e.g calling codec_process() for AAC takes 0.6ms) then calling codec_process() twice inside a codec_adapter_copy() it will take 1.2ms more than 1ms which I set for the pipeline period. 

Actually, my main question is it if you see any problems if we remove the processing loops so that we only call process() only once per codec_adapter_copy().

mrajwa, 2021-04-14:
Yes I think there will be no problem with just a single processing integration on each copy.

dbaluta, 2021-04-14:
Yes, I tested that! One iteration scenario works fine! I have though the following problem (this is not a limitation of codec adapter):

-> codec_adapter_copy()
     -> codec_init_process() [0.8ms]
     -> codec_process()  [0.6ms]

You see doing one init_process + process() in a single copy takes more that 1ms for AAC. I could fix that by using a 2ms period but doing this just because init_proces() is very heavy with some codecs is really not ideal.

I think that just doing only codec_init_process() in the first copy and then breaking would be better. In order to avoid skipping one period for the codecs which do not require a call to init_proces() I think we should somehow make codec_init_process() a NOP when codec_init_process() is not needed.

mrajwa, 2021-04-14:
Interesting that init_process() takes so much time. I guess it is the decoding stage right? 1,4 ms for the whole processing procedure is still acceptable assuming you move CA component on slave core.

dbaluta, 2021-04-14:
Yes, init_process() is decoding stage where the algorithm parses the input header, check for sync words, figures out codec parameters (channels, sample rate etc).

Unfortunately, i.MX8 integration of the DSP is single core. 

Do you see any problems by doing the codec_init_process() in the first copy() and then skipping codec_process() for the first copy().

Something like this:
```
codec_adapter_copy()
     if (!init) {
             codec_init_process();
             return 0;
      }

    codec_process();
```

mrajwa, 2021-04-14:
> Unfortunately, i.MX8 integration of the DSP is single core.

Why is that?

> Do you see any problems by doing the codec_init_process() in the first copy() and then skipping codec_process() for the first copy().

Don't you need to call codec_init_process() prior to any processing of new input? In general by skipping `codec_process()` you lose valuable time so I don't think it is a good idea.

paulstelian97, 2021-04-14:
> > Unfortunately, i.MX8 integration of the DSP is single core.
> 
> Why is that? I mean why can't you utilize more cores?
> 

Maybe that's an idea for future hardware revisions, but the current one in i.MX8 (all of the supported platforms) is single core.

> Don't you need to call codec_init_process() prior to any processing of new input? In general by skipping `codec_process()` you lose valuable time so I don't think it is a good idea.

We need to ensure we don't break real time requirements while we have only a single core. So we have to do the bulk of the processing in a single period and if we try to do high sample rates longer periods will overload the SDMA (and will also fuck with buffer computation in topology code for some reason).

dbaluta, 2021-04-14:
1) There are no more cores. The HIFI4 DSP hardware chip has one single core.

2) codec_init_process() is called only once before decoding starts.

mrajwa, 2021-04-14:
> There are no more cores. The HIFI4 DSP hardware chip has one single core.

Ohh ok.

What you can also do is change the pipeline period so you have 2ms processing window. Maybe you can also try long deep buffering. Yet another solution would be to schedule separate task for processing on that component.

closed 2021-05-05 by dbaluta