thesofproject/sof #4042: undefined reference to `XT_ROUND_S'

hongshui3000, 2021-04-16:
**Describe the bug**
build the code with xcc, the architecture is hifi4. But I searched the header file but did not find the definition of XT_ROUND_S.
Only found "XT_FIROUND_S", but I think the definition is different.
```
C:/usr/xtensa/XtDevTools/install/tools/RI-2020.5-win32/XtensaTools/bin/xt-ld.exe: zephyr/zephyr_prebuilt.elf section `.rodata' will not fit in region `dram1_0_seg'
C:/usr/xtensa/XtDevTools/install/tools/RI-2020.5-win32/XtensaTools/bin/xt-ld.exe: region `dram1_0_seg' overflowed by 158248 bytes
modules/sof/libmodules_sof.a(pcm_converter_hifi3.c.obj):(.literal.pcm_convert_f_to_s16_lin+0x0): undefined ro\pcm_converter/pcm_converter_hifi3.c:878: undefined reference to `XT_ROUND_S'
ninja: build stopped: subcommand failed.
```

paulstelian97, 2021-04-16:
Bug also affects us on the HIFI4 part. The workaround we have done at NXP is use RG-2017.8 instead of the 2020 toolchain.

dbaluta, 2021-04-16:
what xcc version are you using?

hongshui3000, 2021-04-16:
Yes i use RI-2020.5.Besides changing the tool version, are there other solutions?

hongshui3000, 2021-04-16:
Can someone help me look at the description of XT_FIROUND_S, I want to find out if there is any macro or function that can replace it

dbaluta, 2021-04-16:
@hongshui3000 Before going this route can you try to:

```
#define XT_ROUND_S(x, y) x
```

and see if you get into other problems?

I think the best way is to upgarde SOF to a newer toolchain.

hongshui3000, 2021-04-16:
@dbaluta 
It can work like this 
```
#define XT_ROUND_S(x, y) x
```
I also think sof should be upgraded

hongshui3000, 2021-04-16:
```
#define XT_ROUND_S(x, y) x
```
Build fail：
```
E:/work/modules/audio/sof/src/audio/pcm_converter/pcm_converter_hifi3.c:782: error: incompatible types in assignment        
E:/work/modules/audio/sof/src/audio/pcm_converter/pcm_converter_hifi3.c: In function 'pcm_convert_f_to_s32_lin':
E:/work/modules/audio/sof/src/audio/pcm_converter/pcm_converter_hifi3.c:880: error: perhaps a TIE conversion macro (from xtfloat to ae_int32x2) is missing
```

hongshui3000, 2021-04-16:
I am not sure if XT_TRUNC_S can replace XT_ROUND_S
```
Converts the contents of the lower 32 bits of floating-point operand fs from single-precision to
signed integer format, rounding toward zero. The converted integer value is first scaled by a
power of two constant value encoded in the immediate field, with 0..31 representing 1.0, 0.5,
0.25, …, 1.0/2147483648.0. The scaling allows for a fixed point notation where the binary
point is at the right end of the integer for i=0 and moves to the left as i increases until for i=31
there are 31 fractional bits represented in the fixed point number.
C syntax:
 int XT_TRUNC_S (float fs, immediate i);
```

dbaluta, 2021-04-16:
@hongshui3000 what are you exactly trying to do? Compile SOF with Zephyr for Intel platforms right?

Also, you might want to remove pcm_converter_hifi3.c from compilation. We got into a lot of trouble when trying to enable for RI-2020. 

@paulstelian97 I think we should brainstorm and spend some days to really make sof work with a newer compiler. RI-2020 is a good start. 

We need to have different /src/arch/xtensa/xtos directories depending on toolchain.

lgirdwood, 2021-04-16:
@singalsu any comments here ?

lgirdwood, 2021-04-16:
@lyakh fyi

singalsu, 2021-04-16:
It's likely this ROUND.S instruction:

[image]

paulstelian97, 2021-04-16:
@singalsu Any direct replacement that works on the new toolchain? Or do we need a HIFI4 version of this?

hongshui3000, 2021-04-17:
@dbaluta 
I am using a brand new hifi4 processor, which is not in the sof library.

I consulted cadence ，Their reply to me：
```
Try comparing your previous and the present HiFi 4 DSP configurations (i.e., respective config.html)
Most likely, the optional FPU is not part of the configuration where XT_ROUND_S missing.  
In such a case, you need to use standard C library functions for conversions between float and integer formats.
```

hongshui3000, 2021-04-17:
Before finding a good solution, I first did the following processing. I think it should not affect the original code
```
#if !defined(XT_ROUND_S)   --------------------------------------------------------------------------here
/**
 * shift d value left (for positive a) or right (for negative a)
 * and take care about overflows
 */
static inline int32_t _pcm_shift(int32_t d, int32_t a)
{
	int64_t dd = d;

	if (a > 32)
		a = 32;
	else if (a < -32)
		a = -32;

	dd = a >= 0 ? dd << a : dd >> -a;
	if (dd > INT32_MAX)
		dd = INT32_MAX;

	return (int32_t)dd;
}
/**
 * \brief convert float number to fixed point
 *
 * Do not relay on compiler built-in float<=>int conversion in generic
 * implementation, because "floating types float, double, and long double whose
 * radix is not specified by the C standard but is usually two"
 * ~https://gcc.gnu.org/onlinedocs/gcc/Decimal-Float.html
 *
 * \param src integer number to convert, it is int32_t to omit software float
 *            operations library inclusion by compiler, when in whole topology
 *            only external component needs float input.
 * \param pow additional exponent component,
 *            number of fractional bits in fixed point value.
 *            Use '0' for normal conversion to integers
 * \return (int32_t)src * 2**pow
 */
static int32_t _pcm_convert_f_to_i(int32_t src, int32_t pow)
{
	int32_t exponent, mantissa, dst;

	exponent = (src >> 23);
	exponent = (exponent & 0xFF) + pow - 127; /* exponential */
	mantissa = BIT(23) | (MASK(22, 0) & src); /* mantisa + 1.0 [Q9.22] */
	/* calculate power */
	dst = _pcm_shift(mantissa, exponent - 23);
	/* add 0.5 to round correctly but assert it doesn't lead to overflow */
	if (exponent - 22 < 9 || (src & BIT(31)) == BIT(31))
		dst += _pcm_shift(mantissa, exponent - 22) & 1;
	/* copy sign to dst */
	dst = (dst ^ (src >> 31)) + (int)((unsigned int)src >> 31);

	return dst;
}
#endif
/**
 * \brief HiFi3 enabled PCM conversion from IEEE-754 float to 16 bit.
 * \param[in] psrc source linear buffer.
 * \param[out] pdst destination linear buffer.
 * \param[in] samples Number of samples to process.
 */
#if defined(XT_ROUND_S) --------------------------------------------------------------------------------------------here
static void pcm_convert_f_to_s16_lin(const void *psrc, void *pdst,
				     uint32_t samples)
{
	const xtfloat *in = psrc;
	ae_int16x4 *out = pdst;
	xtfloat x;
	int y;
	int i = 0;

	while (i < samples) {
		/* load one 32 bit sample */
		XT_xtfloat_loadip(x, in, sizeof(x));

		/* shift and round */
		y = XT_ROUND_S(x, 15);

		/* store one 16 bit sample */
		AE_S16_0_IP(y, (ae_int16 *)out, sizeof(ae_int16));

		++i;
	}
}
#else
static void pcm_convert_f_to_s16_lin(const void *psrc, void *pdst,
				     uint32_t samples)
{
	const int32_t *src = psrc; /* float */
	int16_t *dst = pdst;
	int i;

	/* s16 is in format Q1.15 so during */
	/* conversion add 15 from exponent */
	for (i = 0; i < samples; i++)
		dst[i] = sat_int16(_pcm_convert_f_to_i(src[i], 15));
}
#endif
```

dbaluta, 2021-04-17:
@hongshui3000 will have a look on Monday. What platform are you trying SOF on?

SOF supports NXP imx8 platforms which have Hifi4 DSP.

hongshui3000, 2021-04-17:
Currently I work on my own hifi4 platform (not Intel or nxp soc) .I probably browsed the configuration of my nxp RT600 (dsp part), which is similar to my current configuration.So I think most people who use hifi4 are likely to encounter my problems

lgirdwood, 2021-04-19:
> ```
>     /* compute d+n=log2(e)*x */
> #if defined(XT_FIROUND_S)
>     y=XT_FIROUND_S(XT_MUL_S(x,c[0].f));
> #else
>     y=XT_FLOAT_S(XT_ROUND_S(XT_MUL_S(x,c[0].f),0),0);
> #endif
> ```
> 
> 
> Who has a better solution to be compatible with hifi4？

Lets do what Cadence do in the NN libs here. They know what they are doing and we should copy their method here.

@hongshui3000 your code looks fine to me, but @singalsu needs to review all HiFi code. Please send a PR and we can merge.

lgirdwood, 2021-04-19:
@hongshui3000 please also send us your patch and we can merge.

hongshui3000, 2021-04-20:
@dbaluta 
 I am working for a company 
@lgirdwood 
Ok, I will provide PR for your reference

hongshui3000, 2021-04-24:
The PR has been accepted and should be closed

closed 2021-04-24 by hongshui3000