thesofproject/sof #4044: Is DYNAMIC_INTERRUPTS mandatory in the sof system?

hongshui3000, 2021-04-17:
Is DYNAMIC_INTERRUPTS mandatory in the sof system?
I plan to call the sof library in zephyr, maybe my soc does not support dynamic interrupt.
What is the reason for the need for dynamic interrupt enable in sof? 

If CONFIG_LIBRARY is enabled, is it necessary to enable dynamic interrupts?

file : modules\audio\sof\zephyr\wrapper.c
```
/* Confirm Zephyr config settings - TODO: Use ASSERT */
#if !CONFIG_DYNAMIC_INTERRUPTS
#error Define CONFIG_DYNAMIC_INTERRUPTS
#endif
```

lyakh, 2021-04-20:
@hongshui3000 the reason is that SOF requests interrupts dynamically, so, there isn't much point in not having them enabled with SOF. However, I guess, we could do something like #4063 . Note, that's a draft PR for now, let's see if others like it too or if you'd prefer to enable dynamic interrupt support for your platform.

dbaluta, 2021-04-20:
I think we need something like this with i.MX also.

Cc: @iuliana-prodan

iuliana-prodan, 2021-04-20:
> I think we need something like this with i.MX also.
> 
> Cc: @iuliana-prodan

DYNAMIC_INTERRUPTS are not imx specific, but they are required by SOF. So, we should enable them.

lgirdwood, 2021-04-20:
TBH, dynamic irqs can probably be deprecated once Zephyr is fully integrated and we use the static Zephyr model. Its a simplification for us.

hongshui3000, 2021-04-24:
The PR has been accepted and should be closed

closed 2021-04-24 by hongshui3000