thesofproject/sof #4045: [BUG]"M_PI" does not exist

hongshui3000, 2021-04-17:
**Describe the bug**
When I use zephyr (minimal library) to call sof, I found that there is no definition of "M_PI".
Cause build error

I added the following code：
file:modules\audio\sof\src\math\trig.c
```
#ifndef M_PI
#define M_PI       3.14159265358979323846   // pi
#endif
```

lgirdwood, 2021-04-19:
@singalsu an comment here ?

singalsu, 2021-04-19:
@lgirdwood It seems that src/math/trig.c gets the value M_PI from C #include <math.h>. Should SOF avoid that and define own value?

lgirdwood, 2021-04-19:
We should use the C library versions of all these symbols.

hongshui3000, 2021-04-20:
@lgirdwood 
Then I provide PR to Zephyr, but I am not sure that they will be accepted

lgirdwood, 2021-04-20:
> @lgirdwood
> Then I provide PR to Zephyr, but I am not sure that they will be accepted

Fwiw, we work alongside the Zephyr developers (who also work at Intel and NXP). Dont worry, this will get resolved. 

@nashif @andyross care to give us some direction on how you see Zephyr using trig ? We can carry this in the SOF headers if required.

andyross, 2021-04-20:
The root cause is that the Zephyr minimal libc is a partial implementation and mostly skipped anything to do with floating point (because Zephyr targets often don't have hardware FPUs, or as is the case with the ADSP support only weird vectorized instructions sets without general C support -- there just hasn't been demand basically).  We support newlib as a C library which is much more compliant, though there are some integration glitches there too (specifically: don't trust that anything that needs to be reentrant or threadsafe is, it almost certainly isn't).

For the specific case of M_PI and related constants, it's trivial just to copy the right values in.  But it's likely we'll run into similar problems later depending on how deep the floating point usage gets.  If we start having to pull in yet another copy of the sunpro trigonometry functions, then it's time to start moving to newlib.

lgirdwood, 2021-04-21:
@hongshui3000 it looks like we will need to carry this in SOF in the short term. Could you send a PR to fix it (and make sure the comments state the reasons). Thanks !

hongshui3000, 2021-04-22:
I think Zephyr should be able to accept my PR, if not. I will submit a PR in sof. @nashif @andyross  Any comments?

lgirdwood, 2021-04-27:
@hongshui3000 fyi - we had a call with @nashif and @andyross last night. Issue now fixed on SOF side.

closed 2021-04-27 by lgirdwood