thesofproject/sof #4046: [BUG]Why do I need to enable CONFIG_CAVS_TIMER?

hongshui3000, 2021-04-17:
**Describe the bug**
This code will prevent all xtensa architecture processors from using sof (zephyr), except for the Intel CAVS platform.I don’t understand the necessity of doing so.

file:modules\audio\sof\src\schedule\timer_domain.c
```
/*
 * Currently the Zephyr clock rate is part it's Kconfig known at build time.
 * SOF on Intel CAVS platforms currently only aligns with Zephyr when both
 * use the CAVS 19.2 MHz SSP clock. TODO - needs runtime alignment.
 */
#if CONFIG_XTENSA && !CONFIG_CAVS_TIMER
#error "Zephyr uses 19.2MHz clock derived from SSP which must be enabled."
#endif
```

lgirdwood, 2021-04-19:
Good point. It's a bug. @lyakh can you fix.

lyakh, 2021-04-20:
@hongshui3000 would #4062 also work for you?

hongshui3000, 2021-04-22:
> @hongshui3000 would #4062 also work for you?

work

lgirdwood, 2021-04-22:
Thanks for confirming it works @hongshui3000 will close now.

closed 2021-04-22 by lgirdwood