thesofproject/sof #4048: [BUG] __typeof__ or typeof

hongshui3000, 2021-04-18:
**Describe the bug**
I use (zephyr + sof )to build the project, and I find that my compiler does not seem to be able to correctly recognize typeof. I include the typeof header file in zephyr and it will report an error. 

I took a preliminary look at the sof code, and I found that there are a lot of __typeof__ and typeof in it. And both are in use. I'm not sure where the problem is. I use xt-xcc (xtensa) to compile the code 
log：
```
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h: In function 'audio_stream_avail_frames':
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:381: warning: implicit declaration of function 'typeof'
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:381: error: expected ';' before '__a'
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:381: error: '__a' undeclared (first use in this function)
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:381: error: (Each undeclared identifier is reported only once
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:381: error: for each function it appears in.)
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:382: warning: control reaches end of non-void function
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h: In function 'audio_stream_copy':
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:574: error: expected ';' before '__a'
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:574: error: expected ';' before '__a'
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:574: error: '__a' undeclared (first use in this function)
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:574: warning: type defaults to 
'int' in declaration of '__b'
../../zephyr/../modules/audio/sof/src/include/sof/audio/audio_stream.h:574: error: expected ';' before '__a'
```

lgirdwood, 2021-04-18:
@marc-hb - looks like another alignment thing to take care of.

lgirdwood, 2021-04-19:
@lyakh fyi - not sure if you have seen this - I guess not.

marc-hb, 2021-12-25:
@hongshui3000 is this still a problem? SOF+Zephyr has changed a lot. If yes please provide reproduction steps with git versions of everything. It not please close.

lgirdwood, 2021-12-25:
@andyross whats the preference in Zephyr ? we can align.

closed 2021-12-28 by hongshui3000

andyross, 2022-01-02:
Sorry, just noticing this.  I don't think Zephyr has any particular consensus here.  We use this only in a few places, and where we do it's always used as `__typeof__`, presumably because the bare symbol isn't exposed in some `--std=...` mode or another.  But I don't see any attempt at fallback code for toolchains that don't have it, so it doesn't look fully baked to me.