thesofproject/sof #4050: [BUG] glitch in 17_06_TestMultiCoreCompSrcPlb8000Hz16b16b1ch test

kkarask, 2021-04-19:
**Describe the bug**
In test  17_06_TestMultiCoreCompSrcPlb8000Hz16b16b1ch glitch was found.

**To Reproduce**
Run test with Diagnostic driver 17_06_TestMultiCoreCompSrcPlb8000Hz16b16b1ch.

**Reproduction Rate**
Sporadic.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: BRANCH: main HASH: be065e55a87b5a7b4a731856b4e7d6419de7f60e

2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP
3) Test environment:
BRANCH: master HASH: b7ed915d911acda055bbb822800b76557e9fe7de

**Screenshots or console output**
[image]

[2021_04_18_00_45_36.zip](https://github.com/thesofproject/sof/files/6335141/2021_04_18_00_45_36.zip)

kkarask, 2021-05-11:
I run stress tests and found glitch in the whole wav file in test:  17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch on WHL platform.
Reproduction rate: 1/75.

Topology:
```
 pipe_plb core 0
                    +--------------------+
    +----------+    | +------+   +-----+ |   +-----+
    | Host plb |----->|Buffer|-->| Dai |---->|codec|
    +----------+    | +------+   +-----+ |   +-----+
                    +--------------------+    |  |
                                              |  |
                    pipe_cap core 1           |  |
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +-----+
    | Host cap |<-----|Buffer|<--| Dai |---->|codec|
    +----------+    | +------+   +-----+ |   +-----+
                    +--------------------+
```
Branch: main, hash: 37d97d4357b34ba2f5f59aa575740d27f2a031c7

[17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch.zip](https://github.com/thesofproject/sof/files/6458539/17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch.zip)

lgirdwood, 2021-05-13:
Duplicate of simpler https://github.com/thesofproject/sof/issues/4076

kkarask, 2021-06-22:
Problem with multi core was found also in test 17_07_TestMultiCoreSrcCap48000Hz32b32b8ch in capture pipeline where src component is on core 1:

```
pipeline playback core 0
           +------------------+
    +----+ | +---+  +-------+ |
    |Host|-->|Buf|->|SSP Dai|----------------->+
    +----+ | +---+  +-------+ |                |
           +------------------+                |
        pipeline capture core 0, src core 1    |
           +--------------------------------+  |
    +----+ | +---+  +---+  +---+  +-------+ |  |
    |Host|<--|Buf|<-|Src|<-|Buf|<-|SSP Dai|<---+
    +----+ | +---+  +---+  +---+  +-------+ |
           +--------------------------------+
```

Reproduction rate: 3/75

Branch: main, Hash: d5d2b41c63ab0715c5f122ac1cfcdbaed8df40b1

[image]

[ICL_17_07_TestMultiCoreSrcCap48000Hz32b32b8ch.zip](https://github.com/thesofproject/sof/files/6694463/ICL_17_07_TestMultiCoreSrcCap48000Hz32b32b8ch.zip)

wszypelt, 2023-01-02:
TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt