thesofproject/sof #4052: [BUG] glitch in test 22_05_TestDsmMulticoreSsp48000Hz32b32b2ch

kkarask, 2021-04-19:
**Describe the bug**
Glitch was found in test 22_05_TestDsmMulticoreSsp48000Hz32b32b2ch

**Topology**

'''
 +----------------------------------------------+
    | +------+   +---+   +- -+   +---+   +-------+ |
    | | Host |-->|Buf|-->|Dsm|-->|Buf|-->|SSP Dai|-------+
    | +------+   +---+   +---+   +---+   +-------+ |     |
    +------------------+   ^   +-------------------+     |
                       |   |   |                         |
                       | +---+ |                         |
                       | |Buf| |                         |
                       | +---+ |                         |
                       +---^---+                         |
    +----------------------|------------------------+    |
    | +------+   +---+   +-|--+   +---+   +-------+ |    |
    | | Host |<--|Buf|<--|Dmux|<--|Buf|<--|SSP Dai|<-----+
    | +------+   +---+   +----+   +---+   +-------+ |
    +-----------------------------------------------+

'''

**To Reproduce**
Run test with Diagnostic driver 22_05_TestDsmMulticoreSsp48000Hz32b32b2ch

**Reproduction Rate**
Sporadic.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF:  BRANCH: main HASH: b7ed915d911acda055bbb822800b76557e9fe7de
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP
3) Test environment:
  BRANCH: master HASH: b7ed915d911acda055bbb822800b76557e9fe7de

**Screenshots or console output**
[image]

[2021_04_18_00_53_06.zip](https://github.com/thesofproject/sof/files/6335809/2021_04_18_00_53_06.zip)

lgirdwood, 2021-04-19:
@kkarask I'm assuming all is good when DSM is disabled ? It looks like we have missed a deadline and slipped some data. @slawblauciak fyi.

lgirdwood, 2021-04-19:
@kkarask I also assume that your topology is all core 0 and can you confirm if you are using low power clocks ?

keyonjie, 2021-04-20:
@kkarask what is "SOF: BRANCH: main HASH: b7ed915d911acda"? I can't find this commit in the main branch?

kkarask, 2021-04-20:
@keyonjie Sorry, my mistake. Correct HASH is: be065e55a87b5a7b4a731856b4e7d6419de7f60e

@lgirdwood There are no glithes in other Multi core tests apart from only one issue: [(https://github.com/thesofproject/sof/issues/4050)]

All components use core = 0 except DSM = 1 in test 22_05_TestDsmMulticoreSsp48000Hz32b32b2ch. 
Regarding power clocks could you please check dma_trace_log.

slawblauciak, 2021-04-27:
@kkarask can you please re-check if the issue is reproducible with the current main branch? There was a relevant fix on a different platform, I wonder if it also applies to ICL.

kkarask, 2021-04-27:
Hi, @slawblauciak 
Unfortunately the problem still exists on current main branch on platforms: TGL and ICL. There are glitches in whole wav files. 
Logs:
TGL 22_06_TestDsmMulticoreSdw48000Hz24b32b2ch :
[2021_04_27_10_01_42.zip](https://github.com/thesofproject/sof/files/6383782/2021_04_27_10_01_42.zip)

ICL 22_05_TestDsmMulticoreSsp48000Hz32b32b2ch:
[2021_04_27_10_29_41.zip](https://github.com/thesofproject/sof/files/6383786/2021_04_27_10_29_41.zip)

lgirdwood, 2021-04-27:
DSM and multicore is showing up in a few of these failures recently. @kkarask what's your take here since you are doing the bulk of the test runs.
1) Are we seeing multicore fail sporadically without DSM ? (points to scheduling)
2) Are we seeing DSM fail on core 0 (if possible to run on 0 - points to DSM)
It would be good to know if these issues are more likely scheduling work on other cores or are DSM overruns that then impact the pipeline.

kkarask, 2021-04-28:
@lgirdwood I will run stress tests and give you answer tomorrow. We will see how fix https://github.com/thesofproject/sof/pull/4088 works.

kkarask, 2021-04-29:
All results from stress tests will be available tomorrow. I will comment about reproduction rate.

lgirdwood, 2021-04-29:
@kkarask thanks, it occurred to me that the pcm converter/copier component is also in this pipeline and also the simple pipelines with similar glitches so could be getting stuck or some confusion between DMA and module R/W pointers.
Btw, I can't read the logs.

kkarask, 2021-05-12:
I run stress tests and found glitches with reproduction rate: 4/75.
Logs:
[22_05_TestDsmMulticoreSsp48000Hz24b32b2ch.zip](https://github.com/thesofproject/sof/files/6465529/22_05_TestDsmMulticoreSsp48000Hz24b32b2ch.zip)

lgirdwood, 2021-05-13:
Duplicate of simpler https://github.com/thesofproject/sof/issues/4076

juimonen, 2021-05-18:
@kkarask do you have any idea when these "glitch" bugs started to appear? So do you recall running these previously at some point without these errors? Just trying to little bit bisect the area... I can't reproduce the loopback glitches with linux driver (or it is quite difficult to do at least)...

kkarask, 2021-05-19:
@juimonen Unfortunately I don't have logs, but as long as I can see the test fail on this hash: 4c1f41a91b92cba729f68c0d2e3a33b290ef075b I'm not sure about reason because I don't have logs:( maybe this help you a bit...

mwasko, 2021-06-07:
> Duplicate of simpler #4076

This issue is specific to scenario with DSM component on Core 1 while the rest of the pipeline is on Core 0. The issue is not reproducing in case when entire pipeline is on Core 1.

mwasko, 2021-06-07:
> @kkarask do you have any idea when these "glitch" bugs started to appear?

@juimonen the issue was not reproducible on tgl-11-drop-stable branch. However when we merged scheduling refinements from tgl-11 to main branch then we started to catch quite a lot of multi-core problems (including this one). Most of the issues has already been addressed, but this one still remain - reproduced recently on tgl-13-drop-stable.