thesofproject/sof #4068: [BUG] DSP panic problem

kkarask, 2021-04-21:
**Describe the bug**
In tests 19_00_TestAsrcPlayback8000Hz32b32b2ch and 19_00_TestAsrcPlayback48000Hz16b16b2ch Panic notification occurred.

**Topology**

```
              +---------------------------------+
    +------+    | +---+  +----+  +---+  +-------+ |
    | Host |----->|Buf|->|Asrc|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +----+  +---+  +-------+ | |
                +---------------------------------+ |
                                                    |
                +------------------+                |
    +------+    | +---+  +-------+ |                |
    | Host |<-----|Buf|<-|SSP Dai|<-----------------+
    +------+    | +---+  +-------+ |
                +------------------+

```

**To Reproduce**
Run tests with Diagnostic driver 19_00_TestAsrcPlayback8000Hz32b32b2ch and 19_00_TestAsrcPlayback48000Hz16b16b2ch.

**Reproduction Rate**
100%
Problem doesn't exist before dc0ab8a449d6aea470f787a8aac8067d9c547ec2

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: BRANCH: main HASH: dc0ab8a449d6aea470f787a8aac8067d9c547ec2
2) Name of the platform(s) on which the bug is observed.
    * Platform: WHL RVP
3) Test environment:
    * SOF: BRANCH: master HASH: b7ed915d911acda055bbb822800b76557e9fe7de

**Screenshots or console output**

[2021_04_19_23_43_01.zip](https://github.com/thesofproject/sof/files/6348789/2021_04_19_23_43_01.zip)

lgirdwood, 2021-04-21:
@singalsu 100% reproducible so should be easy to find.

iuliana-prodan, 2021-04-21:
> @singalsu 100% reproducible so should be easy to find.

I can confirm that I also found this issue.
Tried to add ASRC components in an imx pipeline who looked like: PCM0 <----> ASRC <----> volume <-----> SAI1 (wm8960).
When trying to playback a wav, even with 48kHz it crashes in SOF, I only get the IPC reply in kernel
(`sof-audio-of 556e8000.dsp: error: hw params ipc failed for stream 0`) and no logs in sof-logger.

singalsu, 2021-04-22:
Thanks for reporting this! I will start checking tomorrow.

singalsu, 2021-04-23:
Yep, confirm the panic happens with normal SOF linux driver controlled device, no need to set up diagnostic driver that eases the debugging for me.

singalsu, 2021-04-26:
This avoid the panic the pipeline no more works (with SSP DAI). I wonder if dd->chan has got deprecated.

```
diff --git a/src/audio/dai.c b/src/audio/dai.c
index 44c033671..07c80228f 100644
--- a/src/audio/dai.c
+++ b/src/audio/dai.c
@@ -1161,7 +1161,13 @@ static int dai_ts_config(struct comp_dev *dev)
        cfg->direction = dai->direction;
        cfg->index = dd->dai->index;
        cfg->dma_id = dd->dma->plat_data.id;
-       cfg->dma_chan_index = dd->chan->index;
+       if (dd->chan) {
+               cfg->dma_chan_index = dd->chan->index;
+       } else {
+               comp_err(dev, "dai_ts_config(), NULL chan");
+               return -EINVAL;
+       }
+
        cfg->dma_chan_count = dd->dma->plat_data.channels;
        if (!dd->dai->drv->ts_ops.ts_config)
                return -ENXIO;

```
@lgirdwood Do you remember if this has been changed?

lgirdwood, 2021-04-26:
@singalsu any asynchronous APIs should not assume a previous state. In this case the DAI API assumes that the timestamp is setup by prepare(). The DAI API needs to make this clear (where the timestamp and other async features APIs are ready) and survive any attempts to use the API prior to setup. i.e. we check for NULL

iuliana-prodan, 2021-04-27:
@singalsu I've tested i.MX topology with ASRC, with your patches (#4095), and I don't get the DSP panic, anymore.

But I still have some questions regarding ASRC.
It seems to me that I can only convert from 8000 11025 16000 22050 32000 44100 48000 to 48000 Hz.
If I try a downsampling from 64000 to 48000 I get:
```
asrc         1.1      src/audio/asrc/asrc.c:571  asrc_prepare()
asrc         1.1   ...../asrc/asrc_farrow.c:750  ERROR initialise_filter(), fs_in = 64000
asrc         1.1   ...../asrc/asrc_farrow.c:384  ERROR asrc_initialise(), failed filter initialise
asrc         1.1      src/audio/asrc/asrc.c:695  ERROR initialise_asrc(), error -5
```
Which seems expected, by looking in `asrc_farrow.c`, in [`initialise_filter()`](https://github.com/thesofproject/sof/blob/main/src/audio/asrc/asrc_farrow.c#L595) and the [`asrc_conversion_ratios` ](https://github.com/thesofproject/sof/blob/main/src/audio/asrc/asrc_farrow.c#L43)

But, I'm wondering why do you, in Intel's topologies, like [sof-apl-asrc-pcm512x.m4](https://github.com/thesofproject/sof/blob/main/tools/topology/development/sof-apl-asrc-pcm512x.m4) use 192000 Hz as max rate - see [here ](https://github.com/thesofproject/sof/blob/main/tools/topology/development/sof-apl-asrc-pcm512x.m4#L41).
Is this really converted to 48000 Hz? If yes, how?

Thanks!

singalsu, 2021-04-28:
@iuliana-prodan Nice that it fixes the issue for you. There's a minor document add need for that patch and then it should be good to go.
PCM512x codec usually is slave to APL so we can use the sychronous SRC to convert from 192 to 48 kHz. If the codec is master the topology could use SRC and ASRC both in same pipeline. Though this could reveal new pipeline rules issues. This is actually a good case to add to our rules study we are doing with @juimonen . With current quality ASRC coefficients the result could be better with such SRC-ASRC system because the ASRC aliases more. ASRC alone might also be higher load.

lgirdwood, 2021-04-28:
@singalsu ping - will we see the fix be ready to merge today ?

singalsu, 2021-04-28:
@lgirdwood I updated the PR, hope the doxygen part add is OK.

@iuliana-prodan @kkarask There's another problem that I noticed with the I2S devel topology that my other related PR adds. The simultaneous playback and capture with ASRC pipelines into same DAI causes and I/O error for one direction with xrun recover failed error in trace. I'll add a new issue to track the add of the access filtering code to HW timestamps.

iuliana-prodan, 2021-04-28:
> @iuliana-prodan Nice that it fixes the issue for you. There's a minor document add need for that patch and then it should be good to go.

Is there a document about ASRC? Where? 
I only know about SRC - https://thesofproject.github.io/latest/algos/src/sample_rate_conversion.html

> PCM512x codec usually is slave to APL so we can use the sychronous SRC to convert from 192 to 48 kHz. If the codec is master the topology could use SRC and ASRC both in same pipeline. Though this [...]

So, for APL, for example, you use ASRC for upsampling and SRC for downsampling?

For imx8QM we use wm8960 codec as master. 
You're saying that we can use SRC and ASRC in the same pipeline, something like:
PCM0 <----> SRC <----> ASRC <---->volume <-----> SAI1 (wm8960)?

singalsu, 2021-04-28:
@iuliana-prodan The history of ASRC is different than SRC. This was a contribution from other team than Liam's and we got permission to publish source code (I did the "SOFifying" work) but not the document that wasn't totally suitable for this need. A from scratch written new document should be possible given some time. Add an enhancement issue for SOF Docs perhaps?

If there's need for large rate changes, SRC should be better for the hard work. I assume the shipped rates conversions are where the ASRC algorithm performs well. I didn't change those from original. Also some of the Q-format choices inside may to restrict the highest rates or highest in/out ratios so those would need to be checked. For now the SRC-ASRC combination is the recommendation to cover all rates. Load and latency of SRC is low when it does simple 1/N or N fractions. But there will be likely new pipeline params walk issues when SRC and ASRC are used both. Need to test and file issues.

iuliana-prodan, 2021-04-28:
@singalsu @lgirdwood @zrombel I'm looking on Intel's topologies with SRC and ASRC and on the CI ([here](https://sof-ci.01.org/sof-pr-viewer/#/build/PR4089/build6352012), for example).
Where can I find the tests for:
11_01_TestSrcSimple16000On48000Hz16b16b2ch or 17_06_TestMultiCoreCompSrcPlb48000Hz24b32b4ch?
I can only see the logs from the CI. 
How/where can I find out the topology used, the test itself?

Thanks!

closed 2021-04-28 by lgirdwood

singalsu, 2021-04-30:
@iuliana-prodan Those test case implementations are not open source. They use the diagnostic driver (open source under SOF) and I think they are not based on any .tplg but some generated IPC sequence. It would be useful to ASRC tests in sof-test as well to test via normal topologies and driver too.

commit 7522a7b6 (2021-04-28): Audio: ASRC: Move DAI timestamping setup request to trigger start

The asrc_dai_configure_timestamp() caused a DSP panic when called from
asrc_prepare(). The reason is that DAI prepare now happens after ASRC
prepare. Call from trigger start is after pipeline prepare so it
avoids this problem.

Fixes: #4068
PR #6904 (2023-02-03): topology2: set DMIC0 pcm id to 6
UCM requires DMIC PCM ID to be 6.
Fixes: #4068