thesofproject/sof #4069: [BUG][ADL] multiple-pipeline failed on APL_UP2_NOCODEC , Unable to install hw params when arecording on DMIC

XiaoyunWu6666, 2021-04-21:
**Describe the bug**
multiple-pipeline test failed on APL_UP2_NOCODEC , show "Unable to install hw params"when doing arecord on DMIC.

**To Reproduce**
ssh ubuntu@jf-apl-up2-nocodec-1.jf.intel.com
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20
(also run  'arecord   -D hw:0,6 -c 2 -r 48000 -f S32_LE /dev/null -q' and  'arecord   -D hw:0,5 -c 2 -r 48000 -f S32_LE /dev/null -q' remotely,which may cause connection termination)

**Test log**
[image]

**Kernel log**
    
[image]

**Reproduction Rate**
100%

**Environment**
    * Kernel: 6d1b1460
    * SOF: c08f030d536e
    * Topology: sof-apl-nocodec.tplg
    * Platform: APL_UP2_NOCODEC

XiaoyunWu6666, 2021-04-21:
worked well on April 18 , failed since April 19 .
Will do bisect soon

XiaoyunWu6666, 2021-04-21:
after bisecting , found that commit 6c2cdffd0239de4fcedc95e0cdded6acb6dac184 (HEAD -> main) is the first bad commit 
Author: Keyon Jie <yang.jie@linux.intel.com>
Date:   Thu Apr 8 10:56:23 2021 +0800

    dai: dai_config refinement

    This is cleanup of the dai_config logic to make it more robust.

    The host side should send _DAI_ONFIG IPC before trigger start, it can be
    at:
    a. topology parsing and pipeline initialization stage, or
    b. before hw_params, send _DAI_CONFIG according to runtime params.

    For both #a and #b, the configured dai will be stored in the new added
    dai_data.dai_config, then set at the start of .params(), while the DMA
    channel will be requested later at .prepare() and freed at .reset().

    There is no need for the host side to send DMA_CHAN_INVALID to ask for
    freeing the DMA channel anymore, if the host side keep sending this,
    it will be ignored since the dd->chan is already freed at .reset().

    Signed-off-by: Keyon Jie <yang.jie@linux.intel.com>

keyonjie, 2021-04-21:
looks like the memory is used up here, but yes, why this happen after my commit:
```
$ sudo sof-logger -l ~/sof-apl.ldc
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[1225227754.687500] (        0.000000) c0 dma-trace             src/trace/dma-trace.c:331  ERROR FW ABI 0x3012001 DBG ABI 0x5003000 tag v1.7-rc1-271-g6d1224bb84e8-dirty src hash 0x5d021e21 (ldc hash 0x5d021e21)
[   238132.968750] (   238132.968750) c0 memory                      src/lib/alloc.c:743  ERROR failed to alloc 0x78 bytes zone 0x2 caps 0x1 flags 0x0
[   238140.989583] (        8.020833) c0 memory                      src/lib/alloc.c:547  ERROR heap: 0xbe048980 size 55296 blocks 7 caps 0x45
[   238147.395833] (        6.406250) c0 memory                      src/lib/alloc.c:549  ERROR  used 55296 free 0
[   238153.750000] (        6.354167) c0 memory                      src/lib/alloc.c:556  ERROR  block 0 base 0xbe048980 size 64 count 128
[   238159.843750] (        6.093750) c0 memory                      src/lib/alloc.c:558  ERROR   free 0 first at 127
[   238166.041667] (        6.197917) c0 memory                      src/lib/alloc.c:556  ERROR  block 1 base 0xbe04a980 size 128 count 64
[   238172.135417] (        6.093750) c0 memory                      src/lib/alloc.c:558  ERROR   free 0 first at 63
[   238178.333333] (        6.197917) c0 memory                      src/lib/alloc.c:556  ERROR  block 2 base 0xbe04c980 size 256 count 96
[   238184.427083] (        6.093750) c0 memory                      src/lib/alloc.c:558  ERROR   free 0 first at 95
[   238190.625000] (        6.197917) c0 memory                      src/lib/alloc.c:556  ERROR  block 3 base 0xbe052980 size 512 count 8
[   238196.718750] (        6.093750) c0 memory                      src/lib/alloc.c:558  ERROR   free 0 first at 7
[   238216.197917] (       19.479166) c0 memory                      src/lib/alloc.c:756  ERROR failed to alloc 0x78 bytes zone 0x2 caps 0x1 flags 0x0
[   238222.864583] (        6.666667) c0 memory                      src/lib/alloc.c:547  ERROR heap: 0xbe048980 size 55296 blocks 7 caps 0x45
[   238229.531250] (        6.666667) c0 memory                      src/lib/alloc.c:549  ERROR  used 55296 free 0
[   238270.625000] (       41.093750) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[   238280.156250] (        9.531250) c0 host         13.52         src/audio/host.c:738  ERROR host_params(): failed to alloc dma buffer
[   238285.885417] (        5.729167) c0 pipe         13.55 ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 52
[   238291.927083] (        6.041667) c0 ipc                  src/ipc/handler-ipc3.c:284  ERROR ipc: pipe 13 comp 52 params failed -12

```

XiaoyunWu6666, 2021-04-22:
Confirmed with Keyon that this is not a DMIC-related issue . 
In the CI test , we run  'TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20' which means running playback pipelines first . So the testcase failed when 'arecord -Dhw:0,6 ....' since it is the 10th pipeline. Buffer_error occurs at this time.

run ' ./multiple-pipeline.sh -f c -c 5' and  './multiple-pipeline.sh -f c -c 10' can validate this.
[' ./multiple-pipeline.sh -f c -c 5']
no error when "arecord   -D hw:0,6 -c 2 -r 48000 -f S32_LE /dev/null -q"
['./multiple-pipeline.sh -f c -c 10']
error occurs when " aplay   -D hw:0,5 -c 2 -r 48000 -f S16_LE /dev/zero -q "

keyonjie, 2021-04-22:
Hi @XiaoyunWu6666 can you help to try if https://github.com/thesofproject/sof/pull/4074 can fix this issue?

XiaoyunWu6666, 2021-04-22:
@keyonjie  Yes, #4074 can fix this issue.

keyonjie, 2021-04-22:
> @keyonjie Yes, #4074 can fix this issue.

Thanks for confirming, then we can close this now since the PR is merged.

closed 2021-04-22 by keyonjie