thesofproject/sof #4073: [FEATURE] Is it more reasonable to choose a file when compiling, or is it more reasonable to choose a file when cmake builds?

hongshui3000, 2021-04-22:
**Is your feature request related to a problem? Please describe.**
I see that all files will be added in CmakeFile, I'm not sure this is a good way. Maybe I prefer to solve these problems in the early stage of cmake build, and Solving the problem during the compilation process may introduce more macro definitions.
Current CmakeFile:
```
add_local_sources(sof eq_fir.c eq_fir_generic.c eq_fir_hifi2ep.c eq_fir_hifi3.c)
```
At first glance at this file, I thought that all c files would be used

paulstelian97, 2021-04-22:
All files **are** compiled, however each of them has conditional code so that if it doesn't apply it won't build anything. It's about what's inside those files.

hongshui3000, 2021-04-22:
> All files **are** compiled, however each of them has conditional code so that if it doesn't apply it won't build anything. It's about what's inside those files.

What I mean is that if the platform architecture increases, too many macro judgments may be introduced into the C code. It might be better if you solve it in the Cmake file. Although it will not generate code to introduce macro judgments

dbaluta, 2021-04-22:
Do, you mean for example to only compile eq_fir_hifi3.c for HIFI3 platforms right?

I think we need to have a better understanding what's the difference between hifi2, hifi3, hifi4 regarding software. 

In my opinion, we know which HIFIx version each platform has integrated and we can use the platform define (e.g CONFIG_IMX8) to select the proper code. 

I understand you want to reduce the size of the firmware for a platform, right?

hongshui3000, 2021-04-22:
If I use cmake to generate the IDE project, I think the current code will be included in the project, but it is not used in many cases. If you judge which code is needed under the current architecture in cmake, the generated project will not contain unused source code

hongshui3000, 2021-04-22:
> Do, you mean for example to only compile eq_fir_hifi3.c for HIFI3 platforms right?
> 
> I think we need to have a better understanding what's the difference between hifi2, hifi3, hifi4 regarding software.
> 
> In my opinion, we know which HIFIx version each platform has integrated and we can use the platform define (e.g CONFIG_IMX8) to select the proper code.
> 
> I understand you want to reduce the size of the firmware for a platform, right?
> 
> At least what we can do right now is something like this:
> 
> ```
> #if(CONFIG_FIR_GENERIC)
> add_local_sources(sof eq_fir.c eq_fir_generic.c)
> #endif                   
> 
> #if(CONFIG_EQ_FIR_HIFI2EP)
> add_local_sources(sof eq_fir_hifi2ep.c)
> #endif         
> ```
This may just be a matter of code reading habits
Maybe I am used to zephyr, so I think your method is very good.

closed 2021-06-03 by lgirdwood