thesofproject/sof #4076: [BUG] glitch in test 02_00_TestPtpSimpleLoopback48000Hz24b32b8ch

wszypelt, 2021-04-22:
**Describe the bug**
Glitch was found in test 02_00_TestPtpSimpleLoopback48000Hz24b32b8ch

**Topology**
```
                    pipe_plb
                    +--------------------+
    +----------+    | +------+   +-----+ |   +----+
    | Host plb |----->|Buffer|-->| Dai |---->|SSP1|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+    |  |
                                              |  |
                    pipe_cap                  |  |
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +----+
    | Host cap |<-----|Buffer|<--| Dai |---->|SSP0|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+
```

**To Reproduce**
Run test with Diagnostic driver 02_00_TestPtpSimpleLoopback48000Hz24b32b8ch
With options --playback-iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: dc0ab8a449d6aea470f787a8aac8067d9c547ec2
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP
3) Test environment:
BRANCH: master HASH: 180819001ee5c678ee2b6a53b4931c605a14e89c

**Screenshots or console output**
[image]

[2021_04_20_16_03_11.zip](https://github.com/thesofproject/sof/files/6357592/2021_04_20_16_03_11.zip)

lgirdwood, 2021-04-27:
@wszypelt @zrombel I didn't know we supported the same buffer for host DMA and DAI DMA ? I thought the simplest would be the passthrough volume that places a volume in the middle to make sure we dont use the same buffer for host and DAI DMA ?

zrombel, 2021-04-27:
This image can be a little bit confusing as "Host plb" and "Host cap" component are a part of pipelines (test documentation needs a fix). Regarding buffers: buffers shown on topology are buffers that are explicitly created by test logic by IPC "TplgBuffNew" but as far as I know one another buffer is being created implicitly by FW for data from host DMA - those buffers are not included in tests topology drawings. Please correct me if I'm wrong.

lgirdwood, 2021-05-13:
There is a lot of very similar loopback test failures using different audio pipelines with different processing options now marked with "duplicate" label.
All these tests have loopback in common so this simple example should be solved first as it's test case architecture is shared with the other more complex examples. Once this is solved , then the others can be retested and closed/updated as necessary.

wszypelt, 2022-10-13:
No longer reproduced

closed 2022-10-13 by wszypelt