thesofproject/sof #4080: [BUG] glitch in test 02_02_TestPtpSimpleLoopbackMasterRecord16000Hz32b32b4ch

wszypelt, 2021-04-22:
**Describe the bug**
Glitch was found in test 02_02_TestPtpSimpleLoopbackMasterRecord16000Hz32b32b4ch

**Topology**
```
                    pipe_plb
                    +--------------------+
    +----------+    | +------+   +-----+ |   +----+
    | Host plb |----->|Buffer|-->| Dai |---->|SSP1|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+    |  |
                                              |  |
                    pipe_cap                  |  |
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +----+
    | Host cap |<-----|Buffer|<--| Dai |---->|SSP0|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+
```
**To Reproduce**
Run test with Diagnostic driver 02_02_TestPtpSimpleLoopbackMasterRecord16000Hz32b32b4ch
With options --playback-iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: dc0ab8a449d6aea470f787a8aac8067d9c547ec2
2) Name of the platform(s) on which the bug is observed.
Platform: ICL RVP
3) Test environment:
BRANCH: master HASH: 180819001ee5c678ee2b6a53b4931c605a14e89c

**Screenshots or console output**
[image]

[02_02_~1.ZIP](https://github.com/thesofproject/sof/files/6359051/02_02_.1.ZIP)

lgirdwood, 2021-04-27:
@wszypelt @zrombel Not sure if we can use the same buffer for both host and DAI DMA - please confirm if topology diagram is correct.

wszypelt, 2021-05-11:
The same issue in the same reproduction appears in test
02_01_TestPtpSimpleLoopbackMasterPlayback16000Hz32b32b4ch

lgirdwood, 2021-05-12:
@zrombel @kkarask @wszypelt @XiaoyunWu6666 @keqiaozhang @keyonjie @slawblauciak fwiw  we are seeing a lot of glitches being reported recently in various **loopback** tests.
Can we do the following here

1)  Align these under one issue and mark the others as duplicates, they all seem to show repeated or skipped samples which is usually indicative of races between DMA and DSP buffer IO.
2) Duplicate these failures using an existing upstream topology as its difficult to reproduce for developers who do not have access to HW with the slim test framework.

lgirdwood, 2021-05-13:
Duplicate of simpler https://github.com/thesofproject/sof/issues/4076

kkarask, 2021-07-20:
Glitch was found also in test where playback pipeline is connected to master SSP, capture pipeline is connected to slave SSP (02_01_TestPtpSimpleLoopbackMasterPlayback48000Hz32b32b4ch).
Reproduction rate: 2/75.

[image]

lgirdwood, 2021-07-21:
@kkarask we think we may be getting closer to why the python driver and the no_codec CI devices are showing some issues as more reproducible.  Python driver and no codec mode are probably sending IPCs for some sequences faster that the kernel driver in codec mode (as codec mode means lots of slow & blocking I2C IO with codecs).

Can you rerun the test with #4471 locally merged and let us know if there are any changes to above result with Python driver.
@mwasko fyi.

kkarask, 2021-07-22:
@lgirdwood I rerun tests on #4471  and tests passed, but problem can still exists because reproduction rate is not high. See also related issue: https://github.com/thesofproject/sof/issues/4180

lgirdwood, 2021-07-22:
> @lgirdwood I rerun tests on #4471 and tests passed, but problem can still exists because reproduction rate is not high. See also related issue: #4180

Thanks @kkarask, do you mean that #4471 decreases the reproduction rate ? if so by how much ?

kkarask, 2021-07-23:
@lgirdwood yes, it looks like reproduction rate was decreased. After run 3000 tests result is 0.001% fail.

lgirdwood, 2021-07-23:
@lyakh fyi

closed 2025-07-24 by github-actions[bot]