thesofproject/sof #4104: [BUG] Post Processing XRUN when playing multiple times on same topology

zrombel, 2021-04-27:
**Describe the bug**
In any Post Processing test when trying to play 2 times or more on one topology (new streams are created, topology stays intact) SteamStall caused by XRUN occurs. 

**Topology**
```
                +------------------------------------+
    +------+    | +---+  +-------+  +---+  +-------+ |
    | Host |----->|Buf|->|GenProc|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +-------+  +---+  +-------+ | |
                +------------------------------------+ |
                                                       |
                +------------------+                   |
    +------+    | +---+  +-------+ |                   |
    | Host |<-----|Buf|<-|SSP Dai|<--------------------+
    +------+    | +---+  +-------+ |
                +------------------+
```

**To Reproduce**
Any python test from groups:
25_00_TestGenericProcessorSimplePlb
25_02_TestGenericProcessorCompMultiCorePlb
25_06_TestGenericProcessorPplMultiCorePlb

with parameters:
--playback_iterations=2

**Reproduction Rate**
100%

**Environment**
1) Name of the platform(s) on which the bug is observed.
    * Platform: TGL B0 RVP
2) Firmware branch name and commit 
    * Branch: main
    * Hash: between eb459078f3023f6762428f38976547576eeae59f (bug free) and 2e6cafa02fca352c9f4a085d3c44925e61a19253 (bugged)
3) Python tests branch name and commit
    * Branch: master
    * Hash: 825533f4e0d3ff5e2425abb50e9785916d7d1044

**Logs**
[25_00_TestGenericProcessorSimplePlb_StreamStall.zip](https://github.com/thesofproject/sof/files/6384533/25_00_TestGenericProcessorSimplePlb_StreamStall.zip)

lgirdwood, 2021-05-10:
@dbaluta fyi, have you seen this on your side ? I'm wondering here if we are not correctly tearing down the codec adapter pipeline at stream close.

lgirdwood, 2021-05-10:
```
[  5628978.880208] (       12.760417) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:843  ssp_trigger() cmd 1
[  5628988.880208] (       10.000000) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:631  ssp_pre_start()
[  5628998.281250] (        9.401042) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:760  ssp_start()
[  5629055.442708] (       57.161457) c0 ll-schedule        ./schedule/ll_schedule.c:403  task add 0xbe0a0100 0x1fffa4e0U
[  5629063.984375] (        8.541667) c0 ll-schedule        ./schedule/ll_schedule.c:407  task params pri 0 flags 0 start 0 period 1000
[  5629076.510417] (       12.526042) c0 ll-schedule        ./schedule/ll_schedule.c:313  new added task->start 216160631 at 216156342
[  5629084.921875] (        8.411458) c0 ll-schedule        ./schedule/ll_schedule.c:316  num_tasks 4 total_num_tasks 4
[  5630249.322917] (     1164.401001) c0 codec_adapter 1.0   ......../codec/cadence.c:429  ERROR cadence_codec_process() error ffff80c4: processing failed
[  5630260.208333] (       10.885417) c0 codec_adapter 1.0   ......../codec/generic.c:285  ERROR codec_prepare() error -32572: codec process failed for codec_id cade0101
[  5630271.093750] (       10.885417) c0 codec_adapter 1.0   ......../codec_adapter.c:417  ERROR codec_adapter_copy() error ffff80c4: lib processing failed
[  5630282.578125] (       11.484375) c0 pipe         1.5   ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -32572, start->comp.id = 2, dir = 1
[  5630294.739583] (       12.161458) c0 pipe         1.5   ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[  5630305.911458] (       11.171875) c0 ll-schedule        ./schedule/ll_schedule.c:127  task complete 0xbe0a0100 0x1fffa4e0U
```

lgirdwood, 2021-05-10:
Linked to #3966 ?

dbaluta, 2021-05-10:
@zrombel I actually did notice that:
* cplay works fine at first run
* cplay gets blocked waiting for DSP to consume the frames if you start the second cplay quick enough (1-2seconds).
* if you start cplay after 2 or more seconds everything works well.

I plan to investigate this. Initially I thought that might have something to do with DRAIN which is not yet implemented.

dbaluta, 2021-05-10:
@zrombel yes, this happens because of PM runtime suspend.

zrombel, 2021-05-11:
Unfortunately PR #4162 doesn't fix this bug. 
Logs from test with 2x playback iterations:
[MRajwa_PR4162_25_00_TestGenericProcessorSimplePlb48000Hz32b32b2ch_XRUN.zip](https://github.com/thesofproject/sof/files/6457175/MRajwa_PR4162_25_00_TestGenericProcessorSimplePlb48000Hz32b32b2ch_XRUN.zip)

lgirdwood, 2021-05-18:
@dbaluta any update from your side here ?

mrajwa, 2021-05-18:
@lgirdwood, @zrombel will be back from vacations in two days, once he's back I will get his platform and fix that

lgirdwood, 2021-05-27:
@dbaluta @mrajwa @zrombel any updates guys - we have some PRs under discussion but are we trending towards a fix ?

mrajwa, 2021-05-27:
@lgirdwood I have just got a platform for testing, will be back with update soon

mrajwa, 2021-05-29:
@lgirdwood https://github.com/thesofproject/sof/pull/4263/ this should fix the issue. However Cadence should really improve the way they document things...the reset procedure is not documented at all and this fix has been created based on library source code analysis...
That said, @kkarask could you please verify this patch against consecutive playbacks tests?
Also, I don't know if slim driver test cases verify it or not but we really should test the reset this way:
1. Initalize & prepare the codec
2. Send custom params (simple post gain is enough here)
3. Perform playback and record some time
4. Perform reset
5. Do another playback
6. Test if playback itself is fine + if we record default parameters (gain for example)

kkarask, 2021-06-02:
I'm working on it. This requires a change in test.

lgirdwood, 2021-06-03:
@kkarask thanks, let us know when results are available. @keyonjie @mengdonglin

keyonjie, 2021-06-09:
@kkarask ping for update. We are gonna to freeze the 1.8 final, if this issue still exist, we will have to push it out for 1.9.

kkarask, 2021-06-09:
#4263 works with test that @mrajwa required (25_01_TestGenericProcessorRuntimeConfig with all parameters). All iteration pass (--playback_iterations=20). 

Other problem occurred: 
Glitch in test: 25_02_TestGenericProcessorCompMultiCorePlb48000Hz16b16b4ch reproduction rate:  2/20
[image]

Glitch in test: 25_06_TestGenericProcessorPplMultiCorePlb48000Hz16b16b4ch reproduction rate: 1/20
[image]

mrajwa, 2021-06-09:
Needs to verify:
- if it happens without this PR (main branch)
- if it happens in playback iterations mode or in topology iterations as well
- if it happens in the very fist iteration

kkarask, 2021-06-10:
@mrajwa it is impossible to verify if it happens without this PR (please, see the bug description). 
All Post Processing tests passed in the very first iteration (with this PR and without)

I run all Post Processing tests with parameters:
--topology_iterations=6
--playback_iterations=6
 and found glitches in tests:

- 25_00_TestGenericProcessorSimplePlb48000Hz16b16b8ch
glitch occurs:
Topology=5, iter=2 

- 25_02_TestGenericProcessorCompMultiCorePlb48000Hz16b16b4ch 
In each topology glitches occur in two iterations (iteration 2 and 3)

- 25_02_TestGenericProcessorCompMultiCorePlb48000Hz16b16b8ch
glitches occur: 
Topology=0, iter=2 
Topology=1, iter=2
Topology=5, iter=2   

- 25_06_TestGenericProcessorPplMultiCorePlb48000Hz16b16b4ch
glitches occur:
Topology=1, iter=2
Topology=2, iter=2 
Topology=4, iter=2 
Topology=5, iter=2

All glitches are at the beginning of the wav file.

keyonjie, 2021-06-10:
@kkarask as the glitches are another issue, can you verify if the XRUN issue original reported here is fixed by #4263 ? e.g. without #4263 does the XRUN issue happen? With PR #4263 applied, does the XRUN issue disappear?

kkarask, 2021-06-11:
@keyonjie yes, #4263 fixed the original issue.

keyonjie, 2021-06-11:
> @keyonjie yes, #4263 fixed the original issue.

Thanks, then let's close this once #4263 merged and file a new one for the glitches issue.

keyonjie, 2021-06-16:
1.8 is already froze, postpone this to 1.9

abonislawski, 2022-01-25:
@kkarask is the glitch problem still valid? If yes, please open a new issue

closed 2022-01-25 by abonislawski