thesofproject/sof #4129: [BUG] Crash when format enabled in topology but not firmware

cujomalainey, 2021-04-30:
**Describe the bug**
[    5.932289] sof-audio-pci 0000:00:1f.3: error : DSP panic!                                                                                                                                  
[    5.932295] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005                  
[    5.932393] sof-audio-pci 0000:00:1f.3: error: runtime exception                                                                                                                            
[    5.932395] sof-audio-pci 0000:00:1f.3: error: trace point 00004000                                                                                                                         
[    5.932397] sof-audio-pci 0000:00:1f.3: error: panic at :0                                                                                                                                  
[    5.932399] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops                                                                                                                            
[    5.932401] sof-audio-pci 0000:00:1f.3: error: Exception Cause: IllegalInstructionCause, Illegal instruction          
[    5.932403] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00000000 SAR     0x00000000                                                                      
[    5.932405] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[    5.932406] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000                                                                      
[    5.932408] sof-audio-pci 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[    5.932410] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000  
[    5.932412] sof-audio-pci 0000:00:1f.3: stack dump from 0x00000000                                                                                                                          
[    5.932414] sof-audio-pci 0000:00:1f.3: 0x00000000: 00000000 00000000 00000000 00000000                                                                                                     
[    5.932417] sof-audio-pci 0000:00:1f.3: 0x00000004: 00000000 00000000 00000000 00000000                                                                                                     
[    5.932419] sof-audio-pci 0000:00:1f.3: 0x00000008: 00000000 00000000 00000000 00000000
[    5.932421] sof-audio-pci 0000:00:1f.3: 0x0000000c: 6d588180 ffff9e6c 47f2fdc0 ffffbcce
[    5.932423] sof-audio-pci 0000:00:1f.3: 0x00000010: a341bbe0 ffffffff 6ff29000 ffff9e6c
[    5.932425] sof-audio-pci 0000:00:1f.3: 0x00000014: 30340828 ffff9e6c 000c0800 00000000                                                                                                     
[    5.932427] sof-audio-pci 0000:00:1f.3: 0x00000018: 00000000 00000000 47f2fe50 ffffbcce  
[    5.932429] sof-audio-pci 0000:00:1f.3: 0x0000001c: 47f2fe78 ffffbcce c0587194 ffffffff                                                                                                     
[    5.932431] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers                                                                                                                 
[    6.756998] capability: warning: `main' uses 32-bit capabilities (legacy support in use)                                                                                                    
[    7.933258] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12         
[    7.933266] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context                                                                                          
[    7.933278] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005                                                                                                                  
[    7.933384] sof-audio-pci 0000:00:1f.3: error: runtime exception                   
[    7.933388] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[    7.933391] sof-audio-pci 0000:00:1f.3: error: panic at :0                
[    7.933395] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[    7.933398] sof-audio-pci 0000:00:1f.3: error: Exception Cause: IllegalInstructionCause, Illegal instruction                                                                                
[    7.933402] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00000000 SAR     0x00000000                                                                      
[    7.933406] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000                                                                      
[    7.933410] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    7.933414] sof-audio-pci 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[    7.933417] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[    7.933420] sof-audio-pci 0000:00:1f.3: stack dump from 0x00000000                                                                                                                          
[    7.933425] sof-audio-pci 0000:00:1f.3: 0x00000000: 00000000 00000000 00000000 00000000                                                                                                     
[    7.933430] sof-audio-pci 0000:00:1f.3: 0x00000004: 00000000 00000000 00000000 00000000                                                                                                     
[    7.933434] sof-audio-pci 0000:00:1f.3: 0x00000008: 00000000 00000000 00000000 00000000
[    7.933438] sof-audio-pci 0000:00:1f.3: 0x0000000c: 6d588180 ffff9e6c 47ef7ab0 ffffbcce
[    7.933442] sof-audio-pci 0000:00:1f.3: 0x00000010: 47ef7a40 ffffbcce ee1a3200 7e4394ef
[    7.933446] sof-audio-pci 0000:00:1f.3: 0x00000014: 30340828 ffff9e6c 2fd3fc28 ffff9e6c
[    7.933450] sof-audio-pci 0000:00:1f.3: 0x00000018: 47ef7b88 ffffbcce c0549237 ffffffff
[    7.933454] sof-audio-pci 0000:00:1f.3: 0x0000001c: a2569819 ffffffff 47ef7b48 ffffbcce
[    7.933465] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00                                                                                         
[    7.933468] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000                                                                                                   
[    7.933476] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003                                                                                 
[    7.933479] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers                                                                                                                 
[    7.933732] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_trigger on 0000:00:1f.3: -110                                                                                  
[    7.933740]  DMIC: ASoC: trigger FE cmd: 0 failed: -110                            

**To Reproduce**
E.g. disable FORMAT_S32LE for TGL with main

**Reproduction Rate**
every time

**Expected behavior**
hmm, hard to tell

**Impact**
Blocker to possible memory reduction methods

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: chromeos kernel
    * SOF: top of #4128 
2) Name of the topology file
    * Topology: TGL-*
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL although i expect this to affect any platform

juimonen, 2021-05-03:
I can get this reproduced actually without mux/demux... with mixer pipeline at least. So we have something fishy in the format propagation when restricting to 16 bits. looks like NULL pointer access currently. Debugging more...

juimonen, 2021-05-03:
@lgirdwood @cujomalainey I think the problem is in dai.c calling pcm_get_conversion_function() in pcm_converter. With processing reduced to 16 bit, it also compiles out everything else than 16 bit processing. If your dai is configured to 24 or 32 bit, pcm_get_conversion_function returns NULL.... and we have no checks here

I'm not sure what's the proper way to fix this:
1) just return error here if function is NULL -> you're responsible to match the dai format to pipeline format
2) remove the "reduction" from converter -> it should always be able to convert formats from rest of the pipeline to dai

juimonen, 2021-05-03:
@cujomalainey can you try out if https://github.com/thesofproject/sof/pull/4131 at least fixes the crash? we can then see what to do next...

juimonen, 2021-05-04:
#4131 is merged. we can still experiment with #4135 and/or close this...

cujomalainey, 2021-05-04:
I think we can close this and follow up in the PR for experimentation

closed 2021-05-04 by cujomalainey