thesofproject/sof #4136: [BUG] 4ch volume control is not working correctly for 16 kHz DMIC DAI

singalsu, 2021-05-04:
**Describe the bug**
The volume control is applied to first two channels for 16 kHz pipeline. The volume control in 48 kHz DMIC capture works OK.

**To Reproduce**

The output of amixer shows that volume is applied for two channels despite attempt to set four.

```
$ amixer cset name='Dmic1 2nd Capture Volume' 60,60,60,60
numid=49,iface=MIXER,name='Dmic1 2nd Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=70,step=0
  : values=60,60
  | dBscale-min=-50.00dB,step=1.00dB,mute=1

```
In comparison the volume command for Dmic0 looks like next with all four channels set to 60.

```
$ amixer cset name='Dmic0 Capture Volume' 60,60,60,60
numid=46,iface=MIXER,name='Dmic0 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=4,min=0,max=70,step=0
  : values=60,60,60,60
  | dBscale-min=-50.00dB,step=1.00dB,mute=1

```

**Reproduction Rate**
Always.

**Expected behavior**
In above example volume step applied to channels 1-4.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: Binary v1.6.1 release
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lenovo X1

**Screenshots or console output**
The audio level on first two channels is higher than last two. They should be nearly the same.

[image]

It seems that pipeline macro pipe-eq-iir-volume-capture-16khz.m4 is using a hard-coded channels map for volume. For 48 kHz it's defined by DMIC_VOL_CH_MAPS. It works for DMIC but will fail if this pipeline is used for any other than DMIC point.

lgirdwood, 2021-05-04:
@juimonen can you take a look and help @singalsu here. I've tagged for v1.8

singalsu, 2021-05-06:
Confirm that Jaska's #4142 fixes this.

commit b1775f27 (2021-05-06): topology: define a channel map macro for all topologies

Previously we had a channel map macro only in general dmic topology. As
the pipelines used with dmic are also used with other topologies the
dmic specific macro does not work anymore. Thus define generic macro in
pipeline.m4 that defaults to stereo channel map. This can be overridden
in topologies as shown in general dmic.

fixes: #4136
commit 21a057b2 (2021-05-14): topology: define a channel map macro for all topologies

Previously we had a channel map macro only in general dmic topology. As
the pipelines used with dmic are also used with other topologies the
dmic specific macro does not work anymore. Thus define generic macro in
pipeline.m4 that defaults to stereo channel map. This can be overridden
in topologies as shown in general dmic.

fixes: #4136
closed 2021-05-06 by lgirdwood