thesofproject/sof #4141: [BUG]/scripts/build-tools.sh -T doesn't build development topologies

dbaluta, 2021-05-05:
**Describe the bug**

Running /scripts/build-tools.sh -T doesn't build tplg files in tools/topology/development

Is this expected behavior? How one builds topologies in `tools/topology/development` using `build-tools.sh` script

lgirdwood, 2021-05-05:
@singalsu fyi - I assume this is a regression ? @marc-hb fyi.

marc-hb, 2021-05-05:
They're built only when building everything. Demo:

```
rm tools/topology/development/sof-apl-dmic-asymmetric.m4 
./scripts/build-tools -T # does not care about /development/
./scripts/build-tools    # tries to build /development.
```

This single line is enough to change that:
```diff
--- a/tools/topology/development/CMakeLists.txt
+++ b/tools/topology/development/CMakeLists.txt
@@ -34,6 +34,7 @@ set(TPLGS_UP
 )
 
 add_custom_target(dev_topologies ALL)
+add_dependencies(topologies dev_topologies)
 
 foreach(tplg ${TPLGS})
 	list(GET tplg 0 input)
```

However I have no idea whether that's desirable or not. I think @plbossart, @ranj063  have some informed opinions on what topologies should be built and when? A new `tools/topology/README.md` would not hurt.

>  I assume this is a regression

No idea but there hasn't been any recent change AFAICT

fredoh9, 2021-05-05:
From SOF-CI, we use "scripts/build-tools.sh" without any option. We should build all.
But I checked on the storage server, I couldn't find development tplg files. Maybe we have some bugs when copying output.

fredoh9, 2021-05-05:
Sorry for a bit off the topic. This issue started with -T option...

For the storage server, there was my misunderstanding. There is no development directory, but all files are under `tplg` directories. I don't think we have any platform to use development's topology, I think, we are keeping flat directory structure for sake of simplifying deploy script. We have more option to do so also. I created an internal issue to discuss that.

dbaluta, 2021-05-06:
@fredoh9 thanks for your explanations.
To sum up:
- `./scripts/build-tools.sh  -T` only builds topologies in the main directory
- if one wants to build topologies in `development` directory there is the option to run `./scripts/build-tools.sh` with no params.

closed 2021-05-06 by dbaluta

marc-hb, 2021-05-18:
I got tired of this confusion too and I submitted #4210

marc-hb, 2021-12-10:
> if one wants to build topologies in development directory there is the option to run ./scripts/build-tools.sh with no params.

Except for this "install" race condition I just discovered: #5067

commit e69e1c7e (2021-05-24): scripts/build-tools: some usage and other clarifications

Fixes #4141. I got tired of repeatedly having to read CMakeLists.txt
files.