thesofproject/sof #4144: [BUG] IPC timeout on GLK-Bobba w. multi-pipeline capture

plbossart, 2021-05-05:
**Describe the bug**

During multi-day tests with sof-test, I found one case of IPC timeout

````
[152083.395138] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 1 dir 0 cmd 0
[152083.395180] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[152083.400719] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 0
[152083.901780] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
[152083.901850] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[152083.901899] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[152083.902014] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[152083.902080] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x80000780 intlctl 0x40000781 rirb 00
[152083.902132] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[152083.902194] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x40000000 dsp status 0x00000000 mask 0x00000003
[152083.902246] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: waking up any trace sleepers
[152083.902342] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: trace IO error
[152083.902385] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0x180]=0x240000 successful
[152083.902388] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[152083.902442] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at soc_component_trigger on 0000:00:0e.0: -110
[152083.902496] kernel:  Headset: ASoC: trigger FE cmd: 0 failed: -110
````

**To Reproduce**
run sof-test/test-case/run-all-tests.sh

**Reproduction Rate**
very infrequent

**Expected behavior**
no IPC error

**Impact**
this shouldn't happen

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev
    * SOF: 1:7:0-cf85f
    * 
2) Name of the topology file
    * Topology: TPLG=/lib/firmware/intel/sof-tplg/sof-glk-da7219.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: GLK-bobba

[bobba-multi-pipeline-capture.tar.gz](https://github.com/thesofproject/sof/files/6429740/bobba-multi-pipeline-capture.tar.gz)

plbossart, 2021-05-05:
@fredoh9 @aiChaoSONG is this something that was observed before?

fredoh9, 2021-05-05:
I don't remember the error on GLK. When I scanned the logs there are some ipc timeout errors with only sof pr 4125 only (purposely didn't add the link). This is one of very hard problems in GLK. I'm running multipipeline testing with KASAN enabled build for two days.

lgirdwood, 2021-05-06:
@fredoh9 please also check if this is issue in v1.8-rc1

plbossart, 2021-05-06:
@fredoh9 It took me three days to find the error, and KASAN was also enabled. I was actually looking to reproduce the da7219 oops and found this IPC error instead...

plbossart, 2021-05-06:
The dmesg log looks awfully similar to the TGL RVP NOCODEC reports https://github.com/thesofproject/sof/issues/3953

````
kernel: [ 5535.360414] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 4 dir 1 cmd 0
kernel: [ 5535.360433] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
kernel: [ 5535.360641] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60050000 size 12
````
The direction is different, and one is an IPC error v. an IPC timeout, but that happens in both cases on a TRIG_STOP. I might have been part of the police in a previous life, I no longer believe in coincidence.

fredoh9, 2021-05-07:
> @fredoh9 please also check if this is issue in v1.8-rc1

Daily test with (sof v1.8-rc1) is available, internal test 3755, looks good. I have a plan to run a test KASAN enabled kernel with sof v1.8-rc1 over the weekends also. Will update if I find any issue.

lgirdwood, 2021-06-03:
@fredoh9 can we close ?

plbossart, 2022-07-20:
no update in a year, closing

closed 2022-07-20 by plbossart