thesofproject/sof #4160: [BUG] Missing topology for sound card chtmax98090 Baytrail chipset

gabriele-3, 2021-05-10:
**Describe the bug**
Sound broken in SOF mode, perfectly working in SST mode.
I'm running openSUSE Tumbleweed on a Bay Trail re purposed chromebook.
When the kernel was updated to 5.12 the internal sound card stopped to work and by installing sof-firmware didn't work.
The dmesg logs showed that `sof-byt-max98090.tplg` was not available.

I worked around the issue by symlinking the available `sof-cht-max98090.tplg` to the missing `sof-byt-max98090.tplg`
Another workaround is to pass to the kernel at boot  `snd_intel_dspcfg.dsp_driver=2`, reverting to SST

**To Reproduce**
SOF mode on Baytrail sound card chtmax98090

**Reproduction Rate**
10/10

**Expected behavior**
Internal sound card to work.

**Impact**
Showstopper: sound not working

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.12 {SHA}
    * SOF: sof-firmware-1.7-1.1{SHA}
2) Name of the topology file
    * Topology: available `sof-cht-max98090.tplg` / missing `sof-byt-max98090.tplg`{FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: Bay Trail {PLATFORM}

**Screenshots or console output: dmesg output**
\`\`\`
[    8.567986] input: PC Speaker as /devices/platform/pcspkr/input/input6
[    8.962365] snd_hda_intel 0000:00:1b.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    9.295484] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1b.0/sound/card0/input7
[    9.295588] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1b.0/sound/card0/input8
[    9.344876] iTCO_vendor_support: vendor-support=0
[    9.361516] iTCO_wdt iTCO_wdt.1.auto: Found a Bay Trail SoC TCO device (Version=3, TCOBASE=0x0460)
[    9.375878] iTCO_wdt iTCO_wdt.1.auto: initialized. heartbeat=30 sec (nowayout=0)
[    9.769967] cros-ec-dev cros-ec-dev.2.auto: cannot get EC features: -95/1
[    9.801684] sof-audio-acpi-intel-byt 80860F28:00: BYT-CR not detected
[    9.816623] sof-audio-acpi-intel-byt 80860F28:00: BYT-CR not detected
[    9.831649] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    9.833111] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.837464] sof-audio-acpi-intel-byt 80860F28:00: Firmware info: version 1:7:0-47d07
[    9.837474] sof-audio-acpi-intel-byt 80860F28:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[    9.837491] sof-audio-acpi-intel-byt 80860F28:00: unknown sof_ext_man header type 3 size 0x30
[    9.891602] sof-audio-acpi-intel-byt 80860F28:00: Firmware info: version 1:7:0-47d07
[    9.891612] sof-audio-acpi-intel-byt 80860F28:00: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[    9.976277] sof-audio-acpi-intel-byt 80860F28:00: Direct firmware load for intel/sof-tplg/sof-byt-max98090.tplg failed with error -2
[    9.976290] sof-audio-acpi-intel-byt 80860F28:00: error: tplg request firmware intel/sof-tplg/sof-byt-max98090.tplg failed err: -2
[    9.976294] sof-audio-acpi-intel-byt 80860F28:00: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[    9.976297] sof-audio-acpi-intel-byt 80860F28:00: error: failed to load DSP topology -2
[    9.976299] sof-audio-acpi-intel-byt 80860F28:00: ASoC: error at snd_soc_component_probe on 80860F28:00: -2
[    9.976303] sof-audio-acpi-intel-byt 80860F28:00: ASoC: failed to probe component -2
[    9.976323] cht-bsw-max98090 cht-bsw-max98090: ASoC: failed to instantiate card -2
[    9.976389] cht-bsw-max98090 cht-bsw-max98090: snd_soc_register_card failed -2
[    9.976401] cht-bsw-max98090: probe of cht-bsw-max98090 failed with error -2

\`\`\`

plbossart, 2021-05-10:
@gabriele-3 I don't think we expected anyone to change drivers for now.
Can you double-check what your distribution set for CONFIG_SND_INTEL_BYT_PREFER_SOF?

This was intended to provide a default but maybe some distros jumped the gun.

That said, you are right that we don't generate this sof-byt-max98009.tplg file. That's a mix I'll have to fix. Thanks for reporting it.

lgirdwood, 2021-05-12:
Now merged, we can get this in v1.8-rc2.

gabriele-3, 2021-05-13:
@Pierre-Louis
My distribution is openSUSE Tumbleweed, they set
CONFIG_SND_INTEL_BYT_PREFER_SOF = y as Tumbleweed is always bringing in the
very latest updates.

plbossart, 2021-05-13:
thanks @gabriele-3 for reporting this.
SOF on Baytrail is mostly functional, we are still chasing random channel swaps and some noise on startup on specific platforms. I had one Chromebook with this configuration but it's no longer functional so can't really test any longer unfortunately, we have to rely on the community to help us find issues and test.

closed 2021-05-13 by plbossart

gabriele-3, 2021-05-13:
@Pierre-Louis
I'm happy to help with testing, consider that I'm not  either a dev or a
coder.