thesofproject/sof #4168: [BUG] Keyword detection notification timeout occured

kkarask, 2021-05-11:
**Describe the bug**
Keyword detection notification timeout was found in test 13_00_TestKdProductionGLK

**Topology**
```
+----------+   +------------------+
|    OS    |-->|   AUDIO DEVICE   |---------------------+
+----------+   +------------------+                     |
                                                        |
           +-------------------------+                  |
+--------+ | +---+   +-----+   +---+ |                  |
|KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->-+               |
+--------+ | +---+   +-----+   +---+ |  |  pipe_cap     |
           +-------------------------+  |               |
                     +------------------+    FIFO B     |
+--------------------|--------------------------------+ |
|+----+  +------+  +---+  +---+  +---+  +---+  +----+ | |
||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-|DMIC|---+
|+----+  +------+  +---+  +---+  +---+  +---+  +----+ | |
+-----------------------------------------------------+ |
                                                        |
                                FIFO A                  |
                 +------------------------------------+ |
                 |+----+  +---+  +---+  +---+  +----+ | |
                 ||Host|<-|Buf|<-|Vol|<-|Buf|<-|DMIC|---+
                 |+----+  +---+  +---+  +---+  +----+ |
                 +------------------------------------+

       +--------------------------------+
+----+ | +---+  +---+  +---+  +-------+ |
|Host|-->|Buf|->|Vol|->|Buf|->|SSP Dai|---+
+----+ | +---+  +---+  +---+  +-------+ | |
       +--------------------------------+ |
                                          |
       +------------------+               |
+----+ | +---+  +-------+ |               |
|Host|<--|Buf|<-|SSP Dai|<----------------+
+----+ | +---+  +-------+ |
       +------------------+
```

**To Reproduce**
Run test 13_00_TestKdProductionGLK with Diagnostic driver

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: 37d97d4357b34ba2f5f59aa575740d27f2a031c7

2) Name of the platform(s) on which the bug is observed.
    * Platform: APL RVP
    
3) Test environment:
    Branch: master Hash: 7bc54a9c53346cea3dfc5bdedbc6a02e39597193

4) Tag:
    v1.8-rc1

Logs:
[13_00_TestKdProductionGLK.zip](https://github.com/thesofproject/sof/files/6458679/13_00_TestKdProductionGLK.zip)

lgirdwood, 2021-05-24:
@kkarask is the SSP loopback pipeline running when this failure occurs ?
@mrajwa fyi.

kkarask, 2021-05-24:
@lgirdwood yes, SSP pipeline is running during the test

lgirdwood, 2021-05-24:
@kkarask ok, have you been able to reproduce this at all with the kernel driver ? @slawblauciak fyi - could impact TGL13.

juimonen, 2021-05-24:
@kkarask are you sure about the "problem doesn't exists before" -part as that commit really can't have too much to do with this issue? We should really bisect the fw and try to find the actual commit. I know this is a really difficult one, but I would assume the commit should be related more likely directly to SSP or fw scheduling etc.

I can't reproduce the loopback issues with linux driver and I can't get the slimdriver very easily running in linux...

kkarask, 2021-05-25:
@lgirdwood I'm not able to reproduce it using kernel. 
@juimonen Sorry for misleading, I found keyword problem also here:  5ca0e9bba3e6da8abab7faf4577aa6683200bf41
Logs:
[13_00_TestKdProductionGLK.zip](https://github.com/thesofproject/sof/files/6538547/13_00_TestKdProductionGLK.zip)

lgirdwood, 2021-05-27:
@kkarask thanks for the confirmation, I think we need to say that this could be a bug with Slim driver since we cant reproduce it with the Linux driver. We can re-open if it proves otherwise but lets close it now.

closed 2021-05-27 by lgirdwood

zrombel, 2021-05-27:
@lgirdwood: I wouldn't be so sure about closing this bug. KD is problematic feature and some issues are visible only on slim driver but it may indicate that there is some problem in KD logic or some corner case and I think that someone should take a look at this as it may show it self in future. @kkarask please take a look at this issue and report it if it keeps on failing.

lgirdwood, 2021-05-27:
> @lgirdwood: I wouldn't be so sure about closing this bug. KD is problematic feature and some issues are visible only on slim driver but it may indicate that there is some problem in KD logic or some [...]

I agree, but this cant yet be reproduced on the Linux driver - it may be differences in timing or slim driver flow that make it reproducible on slim driver only. I would also agree that we keep testing as future FW changes make may it reproducible on the Linux driver (and hence an end user issue).

kkarask, 2021-06-08:
I would like to add that this problem is visible every day in regression tests on APL.