thesofproject/sof #4175: [BUG] Memory leak detected

kkarask, 2021-05-12:
**Describe the bug**
Memory leak detected in tests: 05_10_TestDmicMemLeak and 07_06_TestKdMemLeak

Topology for test 05_10_TestDmicMemLeak  :
```
    +----------+   +------------------+
    |    OS    |-->|   AUDIO DEVICE   |-------------+
    +----------+   +------------------+             |
                                                    |
                  +--------------------+            |
    +----------+  | +------+   +-----+ |  +------+  |
    | Host cap |<---|Buffer|<--| Dai |<---| DMIC |<-+
    +----------+  | +------+   +-----+ |  +------+
                  +--------------------+
```

Topology for test 07_06_TestKdMemLeak
```
Topology:
                pipe_plb
               +------------------+
        +----+ | +---+  +-------+ |
        |Host|-->|Buf|->|SSP Dai|----------------------------------+
        +----+ | +---+  +-------+ |                                |
               +----------------- +                                |
                                                                   |
                   +-------------------------+                     |
        +--------+ | +---+   +-----+   +---+ |                     |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                   |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap         |
                   +-------------------------+ |                   |
                                               |                   |
                             +-----------------+                   |
        +--------------------|-----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-|SSP Dai|---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ |
        +--------------------------------------------------------+
```

**To Reproduce**
Run tests 05_10_TestDmicMemLeak and 07_06_TestKdMemLeak with Diagnostic driver

**Reproduction Rate**
100/100
Problem doesn't exist before: ea5d9be6a0c3087f3f026c41c381102204f57e05

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: ea5d9be6a0c3087f3f026c41c381102204f57e05
  
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL RVP, APL RVP, WHL RVP, ICL RVP

3) Test environment:
    Branch: master, Hash: b7ed915d911acda055bbb822800b76557e9fe7de

Logs:
TGL:
[07_06_TestKdMemLeak.zip](https://github.com/thesofproject/sof/files/6464796/07_06_TestKdMemLeak.zip)
[05_10_TestDmicMemLeak.zip](https://github.com/thesofproject/sof/files/6464809/05_10_TestDmicMemLeak.zip)

WHL:
[05_10_TestDmicMemLeakWHL.zip](https://github.com/thesofproject/sof/files/6464833/05_10_TestDmicMemLeakWHL.zip)
[07_06_TestKdMemLeakWHL.zip](https://github.com/thesofproject/sof/files/6464838/07_06_TestKdMemLeakWHL.zip)

lgirdwood, 2021-05-12:
This could be two leaks, one in KWD and the other in DMIC. @singalsu can you check the dmic and @mrajwa the KWD. Thanks

mrajwa, 2021-05-12:
@slawblauciak will check KWD

singalsu, 2021-05-12:
Thanks, I'll test here 05_10_TestDmicMemLeak.

singalsu, 2021-05-18:
@kkarask There's three DMIC driver related fixed merged #4196, #4195, and #4159. Could you please re-run 05_10_TestDmicMemLeak since I'm not able to run the test with Linux as host OS.

kkarask, 2021-05-18:
@singalsu Issue still exists after marged above PR. Problem with memory leak is present at that Hash: f510435a253373c806c49fbc32d876287bcbe801

singalsu, 2021-05-25:
I could not repeat this issue with SOF in single and overlapping capture cases with arecord scripting. Maybe this should be bisected if it's a regression if there would be a good version found from the past?

lgirdwood, 2021-05-25:
@singalsu I assume we are using uncache regiosn for DMIC config now like SSP ? Please see @keyonjie recent patches.

kkarask, 2021-05-26:
@singalsu Last stable version: 47d223c09ff5da56073f3590d14a64bde513358d First memory leak detected: ea5d9be6a0c3087f3f026c41c381102204f57e05

kkarask, 2021-06-02:
@singalsu Bug free: 47d223c09ff5da56073f3590d14a64bde513358d and bugged: ea5d9be6a0c3087f3f026c41c381102204f57e05. This bug occurs every day in regression tests.

kkarask, 2021-06-08:
@singalsu Good news! Memory leak is not visible on none of the platforms.

singalsu, 2021-06-08:
@kkarask That's great news, thanks for update! Can this issue be closed?

closed 2021-06-09 by kkarask