thesofproject/sof #4184: [BUG][TGL]TGLU_RVP_HDA failed when aplaying using HDMI , dai_hw_param error on iDisp1 Pin

XiaoyunWu6666, 2021-05-13:
**Describe the bug**
When aplaying on HDMI[hw:0,3],it failed.  Unable to install hw params

**To Reproduce**
aplay   -Dhw:0,3 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q
OR
TPLG=sof-hda-generic-4ch-kwd.tplg ~/sof-test/test-case/check-playback.sh -d 10 -l 1 -r 1

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: : 93d91f00
SOF Branch: main
SOF Commit: d337c0a9bab0
TPLG= sof-hda-generic-4ch-kwd.tplg
Device:TGLU_RVP_HDA

**Screenshots or console output**
[Console log]
[image]
[dmesg]

> 
>sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
> iDisp1: ASoC: soc_pcm_hw_params() failed (-22)
> HDMI1: ASoC: dpcm_fe_dai_hw_params failed (-22)
>

kv2019i, 2021-05-19:
Do we have FW logs in any existing bug?

XiaoyunWu6666, 2021-05-19:
> Do we have FW logs in any existing bug?

hi kai , could you please check internal daily test result 4051 -check-playback-3rounds - TGLU_RVP_HDA . Or check sof-logger.txt I updated in the first comment