thesofproject/sof #4185: [BUG]Tplg load failed on Chrome ADLP_RVP_SDW

XiaoyunWu6666, 2021-05-13:
**Describe the bug**
TPLG load failed on Chrome ADLP_RVP_SDW ,there a mismatch between chrome hardware and TPLG file for ADLP_RVP_SDW [sof-adl-rt711.tplg]

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 93d91f00
SOF Branch: main
SOF Commit: d337c0a9bab0
TPLG= sof-adl-rt711.tplg
Device:ADLP_RVP_SDW

**Screenshots or console output**
[image]

XiaoyunWu6666, 2021-05-13:
This is a known blocker 
#4041 is the related pr

libinyang, 2021-05-14:
This is a known issue. We have discussion in https://github.com/thesofproject/sof/pull/4041#discussion_r621949355
We can wait for coreboot to fix it.

lgirdwood, 2021-05-14:
@libinyang @XiaoyunWu6666 if this is duplicate then please mark as duplicate

libinyang, 2021-05-17:
> @libinyang @XiaoyunWu6666 if this is duplicate then please mark as duplicate

@lgirdwood This is not a duplicated issue, but a known issue.

kv2019i, 2021-05-18:
Thanks @libinyang ... I also didn't notice this in the tplg review. This is a bit unfortunate -- if we have some coreboot versions that enable it and some that do not (without passing the info in any runtime discoverable interface), that's going to be a mess. So let's try to get the coreboot changed asap.

kv2019i, 2021-05-20:
And another PR to address the incorrect DMI info on some of the CI machines: https://github.com/thesofproject/linux/pull/2935

XiaoyunWu6666, 2021-06-02:
hi @libinyang @kv2019i  should this issue be closed?

kv2019i, 2021-06-03:
Ack @XiaoyunWu6666 , let's close this.

closed 2021-06-03 by kv2019i