thesofproject/sof #4188: [BUG] topology: SSP_CONFIG_DATA and SSP_MULTI_CONFIG_DATA are not aligned

plbossart, 2021-05-13:
**Describe the bug**

The definitions in topology are not aligned

````m4
dnl SSP_CONFIG_DATA(type, idx, valid bits, mclk_id, quirks, bclk_delay)
dnl mclk_id, quirks, bclk_delay are optional
define(`SSP_CONFIG_DATA',
`SectionVendorTuples."'N_DAI_CONFIG($1$2)`_tuples" {'
`	tokens "sof_ssp_tokens"'
`	tuples."word" {'
`		SOF_TKN_INTEL_SSP_SAMPLE_BITS'	STR($3)
`		SOF_TKN_INTEL_SSP_QUIRKS'	`ifelse(`$5', `', "0", STR($5))'
`		SOF_TKN_INTEL_SSP_BCLK_DELAY'	`ifelse(`$6', `', "0", STR($6))'
`	}'
`	tuples."short" {'
`		SOF_TKN_INTEL_SSP_MCLK_ID'	`ifelse(`$4', `', "0", STR($4))'
`	}'
`}'
`SectionData."'N_DAI_CONFIG($1$2)`_data" {'
`	tuples "'N_DAI_CONFIG($1$2)`_tuples"'
`}'
)

define(`MULTI_SSP_CONFIG',
`SectionHWConfig."'$1`" {'
`'
`	id		"'$2`"'
`'
`	format		"'$3`"'
`	'$4
`	'$5
`	'$6
`	'$7
`}'
$8
)

define(`SSP_MULTI_CONFIG_DATA',
`SectionVendorTuples."'$1`_tuples" {'
`	tokens "sof_ssp_tokens"'
`	tuples."word" {'
`		SOF_TKN_INTEL_SSP_SAMPLE_BITS'	STR($2)
`		SOF_TKN_INTEL_SSP_QUIRKS'	`ifelse(`$4', `', "0", STR($4))'
`		SOF_TKN_INTEL_SSP_BCLK_DELAY'	`ifelse(`$5', `', "0", STR($5))'
`		SOF_TKN_INTEL_SSP_CLKS_CONTROL' `ifelse(`$6', `', "0", STR($6))'
`	}'
`	tuples."short" {'
`		SOF_TKN_INTEL_SSP_MCLK_ID'	`ifelse(`$3', `', "0", STR($3))'
`		SOF_TKN_INTEL_SSP_FRAME_PULSE_WIDTH' `ifelse(`$7', `', "0", STR($7))'
`	}'
`	tuples."bool" {'
`		SOF_TKN_INTEL_SSP_TDM_PADDING_PER_SLOT' `ifelse(`$8', `', "false", STR($8))'
`	}'
`}'
`SectionData."'$1`" {'
`	tuples "'$1`_tuples"'
`}'
)
````

Some fields are only present in the multi-config case, and the parameters are off-by-one. 

**Expected behavior**

Same parameters used in all cases.

**Impact**

annoyance

**Environment**

    * SOF: 8ed78885bc4164529b4aa788558fa1a491055f64

plbossart, 2021-05-13:
@juimonen @ranj063 @kv2019i FYI

juimonen, 2021-05-14:
Do you really see this as a problem? I separated the 2 configs (single/multi) because of reviewer request. The m4 definitions trace down to dai.m4 and that should be changed also. Also I would like to keep the 2 extra parameters on top level for user to understand a little bit more what's happening. multi_ssp macro is a special case and it is used like in 0.5% of our topologies -> I could spend a week trying to make it similar -> just don't think it makes sense here.

plbossart, 2021-05-14:
at a minimum the two cases should expose the same topology tokens. I don't see a reason why
SOF_TKN_INTEL_SSP_FRAME_PULSE_WIDTH
is only enabled for the multi-config case.

lgirdwood, 2021-05-20:
@juimonen Let use the same tokens. We can set unused token to 0 or false if they are not used in 1 case. Try and spend min effort of possible here.

juimonen, 2021-05-21:
minimum effor here: https://github.com/thesofproject/sof/pull/4232

closed 2021-05-26 by lgirdwood