thesofproject/sof #4193: [BUG] Exception SofIpcErrorResponseException 8ed78885bc4164529b4aa788558fa1a491055f64

wszypelt, 2021-05-14:
**Describe the bug**
Bug was found in 19_01_TestAsrcCapture48000Hz32b32b2ch

the content of the error:
ERROR:Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':730 FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000

Last working build: master-2290

**Topology**
```
                     +------------------+
    +------+    | +---+  +-------+ |
    | Host |----->|Buf|->|SSP Dai|------------------+
    +------+    | +---+  +-------+ |                         |
                     +------------------+                         |
                                                                             |
                      +---------------------------------+  |
    +------+    | +---+  +----+  +---+  +-------+ | |
    | Host |<-----|Buf|<-|Asrc|<-|Buf|<-|SSP Dai|<--+
    +------+    | +---+  +----+  +---+  +-------+ |
                       +---------------------------------+

```

**To Reproduce**
Run test with diagnostic driver 19_01_TestAsrcCapture48000Hz32b32b2ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: 8ed78885bc4164529b4aa788558fa1a491055f64
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP, TGL RVP, WHL RVP, APL RVP
3) Test environment:
BRANCH: master HASH: a6c9c757ecfb191a795b172afa8ed05c1e2ae990

**Screenshots or console output**
[image]

[2021_05_14_00_56_06.zip](https://github.com/thesofproject/sof/files/6479025/2021_05_14_00_56_06.zip)

lgirdwood, 2021-05-14:
@wszypelt @kkarask I'm marking as a duplicate as all these recent loopback back have the same common denominator. We need to fix the simplest case first. 
@slawblauciak @mengdonglin @keyonjie has anyone been looking at the simple loopback failures at scrub ?

slawblauciak, 2021-05-14:
@lgirdwood I'm attempting to figure those out atm.

singalsu, 2021-05-18:
I'm able to replicate this issue on CML device with topology sof-cml-rt1011-rt5682-asrc.tplg. Run command "arecord -Dhw:0,0 -f dat -d 10 rec.wav"

```
[338285874.375000] (       62.333332) c0 asrc         2.9      src/audio/asrc/asrc.c:530  asrc_trigger()
[338285883.125000] (        8.750000) c0 asrc                  src/audio/asrc/asrc.c:449  ERROR No DAI sink in pipeline.
[338285891.750000] (        8.625000) c0 asrc         2.9      src/audio/asrc/asrc.c:538  ERROR No DAI found to track

```

singalsu, 2021-05-18:
Git bisect result is

```
There are only 'skip'ped commits left to test.
The first bad commit could be any of:
e21c1f31d51c9d61bd1e26f05b3be2ca325f53ad
80666b7cc7a9c70652a55086034d2bfcd35b3b76
We cannot bisect more!
```

Commit e21c1f31d51c9d61bd1e26f05b3be2ca325f53ad doesn't build, and commit 80666b7cc7a9c70652a55086034d2bfcd35b3b76 produces the corrupt asrc_new() IPC. The DAI seek fail is due to initializing ASRC to push mode for capture (token value 0) while kernel has set the IPC data for pull mode correctly  (token value 1). I wonder if the new IPC is missing a big longer length of IPC struct.

PR #4209 (2021-05-19): IPC: Handle ASRC as ASoC component type and not as process type
ASRC new() receives incorrect topology tokens values before this
fix. It occurred as pipeline DAI find fail for capture direction
ASRC.

Fixes #4193
commit 79c64754 (2021-05-19): IPC: Handle ASRC as ASoC component type and not as process type

ASRC new() receives incorrect topology tokens values before this
fix. It occurred as pipeline DAI find fail for capture direction
ASRC.

Fixes #4193
closed 2021-05-19 by lgirdwood