thesofproject/sof #4199: [BUG]DSP panic and dai_config timeout on APL_UP2_NOCODEC

XiaoyunWu6666, 2021-05-17:
**Describe the bug**
DSP panic and dai_config timeout on APL_UP2_NOCODEC ,causing multiple-pause-resume-25 and check-runtime-pm-status-15 to fail  since May 13 http://sof-ci.sh.intel.com/#/result/planresultdetail/3961

**To Reproduce**
Reproduced in CI daily test since last Thursday (May 13)
But cannot manually reproduce

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 0e1e5072
SOF Branch: main
SOF Commit: 8ed78885bc41

**Log**

http://sof-ci.sh.intel.com/#/result/planresultdetail/3961?model=APL_UP2_NOCODEC&testcase=multiple-pause-resume-25
[image]

http://sof-ci.sh.intel.com/#/result/planresultdetail/4007?model=ADLS_RVP_HDA&testcase=multiple-pipeline-playback-50
[image]

lgirdwood, 2021-05-25:
@XiaoyunWu6666 can you run with latest main as @keyonjie has merged some DAI fixes.

XiaoyunWu6666, 2021-05-26:
I am afraid not, error still occurs .
Let me run it with CI

keyonjie, 2021-05-26:
looks this could be dynamic pipeline related, can you try if reverting commit 426d7816e265 help?

XiaoyunWu6666, 2021-05-26:
i will check it later

lgirdwood, 2021-05-26:
@keyonjie this is a NULL ptr dereference at NULL + 0x5c.  Do we have the FW logs ?

XiaoyunWu6666, 2021-05-26:
> looks this could be dynamic pipeline related, can you try if reverting commit [426d781](https://github.com/thesofproject/sof/commit/426d7816e2652130efe060bbe5a12d01cef082c1) help?

i tried on ubuntu@jf-apl-up2-nocodec-1.
it can help

lgirdwood, 2021-05-26:
> > looks this could be dynamic pipeline related, can you try if reverting commit [426d781](https://github.com/thesofproject/sof/commit/426d7816e2652130efe060bbe5a12d01cef082c1) help?
> 
> i tried on ubuntu@jf-apl-up2-nocodec-1.
> it can help

ok, so it sounds like dynamic pipelines is freeing a resource, setting it to NULL and then we are reusing it.

plbossart, 2021-05-26:
we are not using dynamic pipelines for nocodec, isn't it @ranj063 ?

ranj063, 2021-05-26:
> we are not using dynamic pipelines for nocodec, isn't it @ranj063 ?

Yes, we are. I enabled it for some nocodec topologies already

ranj063, 2021-05-26:
I can look into this one. I cant imagine pause/release causing panics. It has to be something else.

plbossart, 2021-05-26:
FWIW the pause_push/release for SSPs seems completely broken, I can't figure out what the code does.

marc-hb, 2021-06-01:
Can we please partially revert the small, `apl-nocodec` bit of 426d7816e265 to get the tests passing again while leaving this bug open? I mean just this bit:

```diff
--- a/tools/topology/CMakeLists.txt
+++ b/tools/topology/CMakeLists.txt
@@ -30,7 +30,7 @@ set(TPLGS
        "sof-hda-generic-idisp\;sof-hda-generic-idisp\;-DCHANNELS=0\;-DDYNAMIC=1"
        "sof-hda-generic-idisp\;sof-hda-generic-idisp-2ch\;-DCHANNELS=2\;-DDMICPROC_FILTER1=eq_iir_coef_high>
        "sof-hda-generic-idisp\;sof-hda-generic-idisp-4ch\;-DCHANNELS=4\;-DDMICPROC_FILTER1=eq_iir_coef_high>
-       "sof-apl-nocodec\;sof-apl-nocodec"
+       "sof-apl-nocodec\;sof-apl-nocodec\;-DDYNAMIC=1"
        "sof-apl-keyword-detect\;sof-apl-keyword-detect"
        "sof-bdw-codec\;sof-bdw-rt286\;-DCODEC=RT286"
        "sof-bdw-codec\;sof-bdw-rt5640\;-DCODEC=RT5640"
```

marc-hb, 2021-06-02:
> Can we please partially revert the small, apl-nocodec bit of 426d781 to get the tests passing again while leaving this bug open?

Submitted in #4273 . See a different bug description fixed by the same revert there.

ranj063, 2021-06-02:
Like I mentioned, this issue not related to pause/release at all. Just testing playback simultaenously on PCM0 and PCM7 exposes the bug with dynamic pipelines. Our tests limit the number of simultaneous pipelines to 4, so we never test PCM0 and PCM7 together. Ideally the multiple-pipeline playback should have failed at the very first iteration if we had tested this combination.

marc-hb, 2021-06-02:
Indeed I can reproduce a severe crash with a mere `sof-test/test-case/multiple-pipeline-playback.sh -c 5` (which uses both PCM0 and PCM7) whereas `sof-test/test-case/multiple-pipeline-playback.sh -c 4` (the default) does not. Patch #4282 makes that panic go away.

The bug I described in #4273 seems different. It is not affected by the #4282 patch.

@XiaoyunWu6666 could you or someone else investigate our worrying lack of test coverage with respect to concurrent pipelines here?

plbossart, 2021-06-02:
I don't get the limitation @ranj063. we reduced the number of SSP tested due to memory requirements, why do the tests reduce concurrency between pipelines?

plbossart, 2021-06-02:
I also don't get what is common between PCM0 and PCM7, the patch in #4282 fixes problems when sharing components but PCM0 is an SSP and PCM7 HDMI. There should be nothing in common.

Confused.

ranj063, 2021-06-02:
[image]

ranj063, 2021-06-02:
@plbossart PCM0 and PCM7 are the playback pipelines that are connected at the mixer.

plbossart, 2021-06-02:
ok, thanks @ranj063. Not sure why we keep using silly numbering conventions or again using a bad mixer solution. Oh well.

PR #4282 (2021-06-03): ipc: ipc3: Do not reset pipeline components during comp_free
Pipelines can be shared between multiple PCM's. For example,
in the case of the sof-apl-nocodec, a part of pipeline 1
(mixer1.0 ->BUF1.2 ->PGA1.1-> BUF1.3 -> SSP0.OUT)
is shared between PCM0 and PCM7. Setting the source_comp for
pipeline 1 to NULL when PCM0 is stopped results in a panic.
So, hold off resetting the pipeline until the scheduler widget
is freed.

Fixes #4199
closed 2021-06-03 by lgirdwood