thesofproject/sof #4206: [FEATURE] ASRC component with i.MX

iuliana-prodan, 2021-05-18:
**Is your feature request related to a problem? Please describe.**
While trying to add the ASRC component to an i.MX pipeline I've realize that in our DAI (more specifically for i.MX, in SAI or ESAI) we don't have the timestamp_ops implemented. 
The problem is that our hardware, i.MX8QM and i.MX8QXP, doesn't have timestamp support.

**Describe the solution you'd like**
Is there any possibly to use platform timer for ASRC?
Will this impact the accuracy of the conversion?

**Describe alternatives you've considered**
By looking on the SPP and HDA timestamp_ops implementation I realized that we need to determine each argument from 
```
struct timestamp_data {
	uint64_t walclk; /* Wall clock */
	uint64_t sample; /* Sample count */
	uint32_t walclk_rate; /* Rate in Hz, e.g. 19200000 */ 
};
```
Will it be possible to compute:
- walclk and walclk_rate - can these be taken from dtb file?
8QM dtb example:
```
	clock-names = "ipg", "ocram", "core",
		"sai1_bus", "sai1_mclk0", "sai1_mclk1", "sai1_mclk2", "sai1_mclk3";
	clocks = <&clk_dummy>, <&clk_dummy>, <&clk_dummy>,
		<&sai1_lpcg 1>, <&clk_dummy>, <&sai1_lpcg 0>,
		<&clk_dummy>, <&clk_dummy>;
	assigned-clocks = <&acm IMX_ADMA_ACM_SAI1_MCLK_SEL>,
			<&clk IMX_SC_R_AUDIO_PLL_0 IMX_SC_PM_CLK_PLL>,
			<&clk IMX_SC_R_AUDIO_PLL_0 IMX_SC_PM_CLK_MISC0>,
			<&clk IMX_SC_R_AUDIO_PLL_0 IMX_SC_PM_CLK_MISC1>,
			<&sai1_lpcg 0>; /* FIXME: should be sai1, original code is 0 */
	assigned-clock-parents = <&aud_pll_div0_lpcg 0>;
	assigned-clock-rates = <0>, <786432000>, <49152000>, <12288000>, <49152000>;
```
- sample -  what is sample count? How can this be obtained?
Can we sample the FIFO level each time and do the diff to get how many samples were processed meanwhile?
 
Can this be a valid solution? @singalsu @tlauda 

CC: @dbaluta @lgirdwood @paulstelian97 @ViorelSuman

lgirdwood, 2021-05-18:
@iuliana-prodan platform timer could be used if its the same time domain that the DAI is using ?

dbaluta, 2021-05-18:
@lgirdwood in our case DAI is using DMA domain. Our question is what exactly do we need to provide as an input to ASRC module in order to work if our DAI doesn't have support for timestamping.

singalsu, 2021-05-19:
If there's no timestamp feature in HW there would be need for a timer count that is synchronized to audio clock generated by the SoC for e.g. codec that the rest of audio system is synchronized to. The DAI should count PCM frames consumed/produced by HW within timer values captured. There's likely need for stronger low-pass filtering for skew factor than with HW timestamps (minimal measure noise added). Watch out for skew settle delay and possible xruns when enhancing filtering. But it's possible and I was using such before I got the HW timestamps to work.

iuliana-prodan, 2021-05-19:
> count PCM frames consumed/produced by HW

iuliana-prodan, 2021-05-19:
> If there's no timestamp feature in HW there would be need for a timer count that is synchronized to audio clock generated by the SoC for e.g. codec that the rest of audio system is synchronized to. 

Can the platform timer be used?

> The DAI should count PCM frames consumed/produced by HW within timer values captured. 

This represents the `sample count` from `struct timestamp_data`. Have I understood correctly?

If I can use the platform timer, the `walclk_rate` will be the platform frequency, right?

singalsu, 2021-05-19:
@iuliana-prodan Yep, that's correct. If things are correct you will get in asrc_control_loop() a cd->skew value Q2.30 that is close to 1.0. When checking thinks I've output to trace the raw and filtered skew values and extracted them for Matlab plots with some scripts. It was an ugly hack so I've not shared code. Some well thought format to get trace data to Matlab and Octave or to other favorite maths tool would be nice to have.

iuliana-prodan, 2021-05-19:
@singalsu Thank you! Will make a POC and tweak it later. With your help :)