thesofproject/sof #4207: [BUG] [CML] [APL] Incorrect topologies for demux

iuliana-prodan, 2021-05-18:
**Describe the bug**
The  sof-apl-demux-pcm512x.m4 and sof-cml-demux-rt5682.m4 are incorrect since both have two inputs for a buffer (see BUF5.0 in screenshots below).

**Expected behavior**
A buffer has only 1 sink and 1 source.
See also: https://github.com/thesofproject/sof/issues/3818#issuecomment-776093909

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: latest
    * SOF: main
2) Name of the topology file
    * Topology: sof-apl-demux-pcm512x.m4 and sof-cml-demux-rt5682.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL, CML

**Screenshots or console output**
sof-apl-demux-pcm512x.m4:
[image]

sof-cml-demux-rt5682.m4:
[image]

Are there any simple demux topology examples?
For i.MX we want to start with a simple demux example.

lgirdwood, 2021-05-24:
@iuliana-prodan best to grep through the topologies for demux. I think there are users for DSM or WOV.

plbossart, 2021-05-24:
@ranj063 is there a way to flag such mistakes while creating or parsing the topology?

ranj063, 2021-05-24:
> @ranj063 is there a way to flag such mistakes while creating or parsing the topology?

not with m4 but it is possible with tplg2.0

plbossart, 2021-05-24:
what about when parsing the topology in the kernel?

ranj063, 2021-05-24:
The kernel doesnt validate the routes or does it maintain the list of connections per widget. But we can certainly reject the second route to the buffer in the FW at the time of topology parsing

plbossart, 2021-05-24:
that would be a nice addition @ranj063.

ranj063, 2021-05-24:
Sure, I can certainly take a look at this.

juimonen, 2021-05-25:
waitwait.... this is not an error IMHO. The demux pipe needs a "virtual" dai to be scheduled. Also from asoc driver point of view the demux pipeline needs 2 endpoints (host/dai) to be even loaded and not generating an error. So the "echo ref" component path is not real, it is just for making this thing work.

iuliana-prodan, 2021-05-25:
> waitwait.... this is not an error IMHO. The demux pipe needs a "virtual" dai to be scheduled. Also from asoc driver point of view the demux pipeline needs 2 endpoints (host/dai) to be even loaded and not generating an error. 

@juimonen You're saying that there is no BUF5.0, in the above screenshots? 
So, MUXDEMUX1.0 has only one input (BUF1.1) and only one output (BUF1.2)?

>So the "echo ref" component path is not real, it is just for making this thing work.

What thing to make it work? 
This part here: 
[image] 
is virtual? 
Can you, please, explain a little bit.
For example, in i.MX we only have one DAI.

juimonen, 2021-05-25:
@iuliana-prodan so in linux driver the pipeline cannot be created if it doesn't have "endpoints". Endpoints are in this case PCM4C and SSP5.IN. if this SSP5.IN and echoref are removed the pipeline just can't be created (with current driver code). So buf5 is real, it is getting data from demux, but as echo ref is "virtual" there's no data coming from it.

So it looks ugly, but there's no like easy way of getting reference capture pipeline to be scheduled/created without this "virtual dai"

plbossart, 2021-05-25:
@juimonen you are probably using the 'wrong' pipelines here, a pipeline does not need a DAI at all. See e.g. the baytrail topologies, there are host and dai pipelines. You could perfectly create a pipeline that is just a host pipeline for the echo capture.

What you describe doesn't look like a feature to me...

ranj063, 2021-05-25:
Ahh now I remember the virtual Echo Ref 5. I understand Echo Ref 5 is veritual but SSP5.IN is a real widget and we create it in the FW for no reason @juimonen

juimonen, 2021-05-26:
remove SSP5_IN ->

[  147.626415] sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  147.626597] sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  147.626603] sof-audio-pci-intel-apl 0000:00:0e.0: pcm: stream dir 1, posn mailbox offset is 790832
[  147.626771]  EchoRef: ASoC: no backend DAIs enabled for EchoRef
[  147.626812]  EchoRef: ASoC: dpcm_fe_dai_prepare() failed (-22)

Maybe you can what you want somehow differently, but I just don't know how...

lgirdwood, 2021-06-03:
ping @iuliana-prodan - any update here. Do we need a fix or clarification here ?

iuliana-prodan, 2021-06-03:
> ping @iuliana-prodan - any update here. Do we need a fix or clarification here ?

Now I understand that even if there are two inputs for a buffer (i.e. BUF5.0), there is no data coming from EchoRef since this is just a virtual component.

I didn't have the time to make a simple demux topology for imx, but it's on my todo list.

Thanks for clarifications. I believe we can close this.

closed 2021-06-03 by lgirdwood