thesofproject/sof #4230: [BUG] glitches in wav file

kkarask, 2021-05-21:
**Describe the bug**
Glitches was found in wav file in test 05_01_TestDmicSimpleStreamOperations48000hz32b32b4ch. In that test while pipeline is working pause and resume operation are performed on capture stream. There are glitches in most of streams.

**Topology**
```
  +----------+   +------------------+
    |    OS    |-->|   AUDIO DEVICE   |-------------+
    +----------+   +------------------+             |
                                                    |
                  +--------------------+            |
    +----------+  | +------+   +-----+ |  +------+  |
    | Host cap |<---|Buffer|<--| Dai |<---| DMIC |<-+
    +----------+  | +------+   +-----+ |  +------+
                  +--------------------+
```

**To Reproduce**
Run test 05_01_TestDmicSimpleStreamOperations48000hz32b32b4ch with Diagnostic driver 

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: f510435a253373c806c49fbc32d876287bcbe801
    
2) Name of the platform on which the bug is observed.
    * Platform: APL RVP
    
3) Test environment:
   * BRANCH: master HASH: a6c9c757ecfb191a795b172afa8ed05c1e2ae990

**Screenshots or console output**

[image]

lgirdwood, 2021-05-24:
@kkarask are you seeing this with TGL13 or stable-v1.8 branch using kernel (instead of slim driver) ? We have seen some issues lately that have not been reproduced with the kernel and trying to understand the delta with slim driver.

kkarask, 2021-05-25:
@lgirdwood Today I reproduced this bug on stable-v1.8 branch with reproduction rate 1/100 but without kernel. Currently I don't have proper setup of platform to reproduce it using kernel. @mengdonglin Could you help with reproduction this issue using kernel?

I also run this test on TGL13 and all 100 iteration passed.

juimonen, 2021-05-25:
@kkarask so we need to reproduce this with slim driver as with linux driver it is very difficult. So don't go now to linux driver but just keep using slim driver. And we need to bisect the dsp fw for trying to find the commit that causes the loopback issues.

1) choose some platform you have easy access to (like up2 APL)
2) use most simple loopback tests, just input output looped from SSP
3) bisect the fw and run the iteration test to see for errors

@slawblauciak can you help here?

kkarask, 2021-05-26:
@juimonen I reproduced test case and bug is somewhere between: pass: 71ac0f97e9d98df7e5679fbb6669bb07229d4170 and fail: ba6ba3cfc6e835df704e83867ccf68d7e64f72cb with reproduction rate: 1/200

juimonen, 2021-05-26:
@kkarask ok thanks! just still quite hard to imagine that any of these commits in that range would cause glitches...

lgirdwood, 2021-05-26:
@kkarask how do we test the slim driver IO logic is correct and not glitch free. I looks like we cant reproduce with the kernel and we are seeing similar loopback bugs reported with slim driver only ?

zrombel, 2021-05-27:
@lgirdwood: What slim driver does is: it maps memory region to read and write to ADSP to communicate with FW. It has no logic in it. All the logic is done in python test scripts. Second thing is that @kkarask found a regression point in FW and the slimdriver version hasn't change in that time so I don't see how diagnostic driver could be responsible for that bug. Diagnostic driver is heavily used by SOF for couple of years from now and there was no issues with it so I would search for root cause rather in FW that slimdriver it self. Maybe it's true that this bug cannot be reproduce on fullstack setup but  I think is good to investigate it as it may show some deeper problems with FW logic. We had this kind of  issues in past that was only reproducible on slim driver and actually was a FW bug that was fixed.

juimonen, 2021-05-27:
@zrombel yeah not questioning that here, we just can't used it ourselves to find the issue and can't reproduce this with linux driver... 

Did you check the commit range for bisect? quite hard to imagine any of those commits would cause this -> so is this just some random thing caused something else before?

juimonen, 2021-05-27:
ba6ba3cfc audio: codec_adapter: Make sure local_buff has enough room for processing
dfb29d522 codec_adapter: Introduce get_samples interface
da9a5ebe2 codec_adapter: Replace hardcoded codec api ids
3500d0efe include:fix zephyr conflicting types and function
b50099947 scheduler: guard against subdivision underflow on domain clear
c9f442ba6 imx8m: remove SDMA2
71ac0f97e component: do not reset component status in the case of xrun

so maybe 71ac0f97e and b50099947 could even remotely relate to this kind of issue, but also those seem quite unlikely. Other commits don't have for sure anything to do with this, I guess.

lgirdwood, 2021-05-27:
@zrombel thanks for the inputs, but I don't fully agree. Slim driver (Python scripts) does have different IO programming flows and timings than the kernel driver. These may expose issues or may not expose issues shown by the driver. The important part for me is "can this be reproduced at all by the driver ?" if yes, then we can fix, if no - it's more difficult to prioritize this against issues that are impacting users with the driver.
Ideally the Python script would invoke the driver low level functional IP programming flows for all it's IP programming, but this is not possible yet.

lgirdwood, 2021-06-03:
We really need to reproduce with the driver or figure out the differences in IP programming flows with Python code.

wszypelt, 2022-12-30:
unsupported platform
on TGL issue does not occur
Tested FW hash: https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74

closed 2022-12-30 by wszypelt