thesofproject/sof #4237: [BUG][HDA]free_block error when DMIC capture on TGLU_RVP_HDA and JSL_RVP_HDA

XiaoyunWu6666, 2021-05-25:
**Describe the bug**
Appeared on May 25 internal daily test  , 4171
When using DMIC to capture , cases failed and dmesg showed  dpcm_fe_dai_hw_params failed , free_block failed.

**To Reproduce**
 TPLG=sof-hda-generic-4ch-kwd.tplg ~/sof-test/test-case/check-capture.sh -d 1 -l 100 -r 1
OR
TPLG=sof-hda-generic-4ch-kwd.tplg ~/sof-test/test-case/check-capture.sh -d 3 -l 1 -r 3

**Reproduction Rate**
100%

**Model**
TGLU_RVP_HDA
JSL_RVP_HDA

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: dcea8052
SOF Branch: main
SOF Commit: 636cbefc62c9

**Screenshots or console output**
[console]
```
2021-05-25 03:13:27 UTC [COMMAND] arecord   -Dhw:0,6 -r 48000 -c 4 -f S32_LE -d 1 /dev/null -v -q
arecord: set_params:1407: Unable to install hw params:
```
[dmesg]
```
[ 7064.036004] sof-audio-pci-intel-icl 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[ 7064.036018] sof-audio-pci-intel-icl 0000:00:1f.3: error: hw params ipc failed for stream 2
[ 7064.036026] sof-audio-pci-intel-icl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[ 7064.036069]  DMIC: ASoC: soc_pcm_hw_params() failed (-22)
[ 7064.036090]  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-22)

```
[sof-logger]
```
[ 14116619145.462204] (           0.000000) c0 dma-trace             src/trace/dma-trace.c:335  ERROR FW ABI 0x3012001 DBG ABI 0x5003000 tag v1.8-rc1-114-g636cbefc62c9 src hash 0x85629837 (ldc hash 0x85629837)
[     2026644.502802] (     2026644.500000) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062080, cpu = 1
[     2027496.638185] (         852.135376) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062180, cpu = 1
[     4064586.140571] (     2037089.500000) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062100, cpu = 1
[     4065488.900952] (         902.760376) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062280, cpu = 1
[     6110554.444688] (     2045065.500000) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062200, cpu = 1
[     6111402.100905] (         847.656189) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062380, cpu = 1
[     8156107.696739] (     2044705.625000) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062300, cpu = 1
[     8158604.779973] (        2497.083252) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062480, cpu = 1
[    10195910.688601] (     2037305.875000) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062400, cpu = 1
[    10196890.688562] (         979.999939) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062580, cpu = 1
[    12254740.919290] (     2057850.250000) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062500, cpu = 1
[    12256093.158820] (        1352.239502) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062680, cpu = 1
[    12283775.084803] (       27681.925781) c1 dmic-dai     2.0   src/drivers/intel/dmic.c:1074 ERROR dmic_get_hw_params(): dai 0 not configured! &dmic_prm[di] 0x0x9e05b120
[    12283794.095219] (          19.010416) c1 dai          10.62          src/audio/dai.c:248  ERROR dai_comp_get_hw_params(): dai_get_hw_params failed ret -22
[    12283810.970219] (          16.875000) c1 pipe         10.63 ....../pipeline-params.c:162  ERROR pipeline_find_dai_comp(): comp_dai_get_hw_params() error.
[    12283943.209797] (         132.239578) c1 dmic-dai     2.0   src/drivers/intel/dmic.c:1074 ERROR dmic_get_hw_params(): dai 0 not configured! &dmic_prm[di] 0x0x9e05b120
[    12283960.397296] (          17.187500) c1 dai          10.62          src/audio/dai.c:248  ERROR dai_comp_get_hw_params(): dai_get_hw_params failed ret -22
[    12283977.584795] (          17.187500) c1 dai          10.62          src/audio/dai.c:297  ERROR dai_verify_params(): pcm rate parameter 48000 does not match hardware rate 36
[    12283996.386878] (          18.802082) c1 dai          10.62          src/audio/dai.c:459  ERROR dai_params(): pcm params verification failed.
[    12284016.438960] (          20.052082) c1 pipe         10.63 ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -22, host->comp.id = 60
[    12284034.928543] (          18.489582) c1 ipc                  src/ipc/handler-ipc3.c:282  ERROR ipc: pipe 10 comp 60 params failed -22
[    12284772.063930] (         737.135376) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062600, cpu = 1
[    12286085.970128] (        1313.906250) c1 memory                      src/lib/alloc.c:462  ERROR free_block(): invalid heap = 0x9e062780, cpu = 1
```

mengdonglin, 2021-05-25:
This issue was uncovered by a topology update with multicore in https://github.com/thesofproject/sof/pull/4200

fuyuntsuo, 2021-05-25:
> This issue was uncovered by a topology update with multicore in #4200. We can overlook multicore issues on JSL, but we need to fix this issue on TGL

Fix has been deployed here #4238.

keyonjie, 2021-05-25:
> > This issue was uncovered by a topology update with multicore in #4200. We can overlook multicore issues on JSL, but we need to fix this issue on TGL
> 
> Fix has been deployed here #4238.

Thanks, I will submit a PR to address the real issue at the same time.

keyonjie, 2021-05-25:
Here is the PR: https://github.com/thesofproject/sof/pull/4239

@XiaoyunWu6666 can you help to try if it can fix the issue here.

XiaoyunWu6666, 2021-05-26:
> Here is the PR: #4239
> 
> @XiaoyunWu6666 can you help to try if it can fix the issue here.

sure , will provide feedback asap :)

XiaoyunWu6666, 2021-05-26:
Keyon's PR seems able to fix this issue . 
Another issue hidden below appeared and its error message is as the following .
Already synced with keyon and he's debugging on it/ 

[sof-logger]
```
[  7941844383.481957] (          20.572916) c1 dmic-dai     2.0   src/drivers/intel/dmic.c:1074 ERROR dmic_get_hw_params(): dai 0 not configured! &dmic_prm[di] 0x0x9e077710
[  7941844403.481957] (          20.000000) c1 dai          10.62          src/audio/dai.c:248  ERROR dai_comp_get_hw_params(): dai_get_hw_params failed ret -22
[  7941844420.721539] (          17.239582) c1 pipe         10.63 ....../pipeline-params.c:162  ERROR pipeline_find_dai_comp(): comp_dai_get_hw_params() error.
[  7941844518.534036] (          97.812492) c1 dai          10.62       src/ipc/dai-ipc3.c:84   dai_data_config() dai type = 2 index = 0 dd 0x9e122000
[  7941844537.023619] (          18.489582) c1 dai          10.62       src/ipc/dai-ipc3.c:111  config->dmic.fifo_bits = 32 config->dmic.num_pdm_active = 2
[  7941844554.523618] (          17.500000) c1 dmic-dai     2.0   src/drivers/intel/dmic.c:1074 ERROR dmic_get_hw_params(): dai 0 not configured! &dmic_prm[di] 0x0x9e077710
[  7941844572.544450] (          18.020832) c1 dai          10.62          src/audio/dai.c:248  ERROR dai_comp_get_hw_params(): dai_get_hw_params failed ret -22
[  7941844591.659033] (          19.114582) c1 dai          10.62          src/audio/dai.c:297  ERROR dai_verify_params(): pcm rate parameter 48000 does not match hardware rate 36
[  7941844609.211116] (          17.552082) c1 dai          10.62          src/audio/dai.c:459  ERROR dai_params(): pcm params verification failed.
[  7941844629.836115] (          20.625000) c1 pipe         10.63 ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -22, host->comp.id = 60
[  7941844648.690281] (          18.854166) c1 ipc                  src/ipc/handler-ipc3.c:282  ERROR ipc: pipe 10 comp 60 params failed -22

```

keyonjie, 2021-05-26:
Thanks @XiaoyunWu6666 for check.

Yes that should be another issue, please help to file one in case see that again.

XiaoyunWu6666, 2021-05-26:
> Thanks @XiaoyunWu6666 for check.
> 
> Yes that should be another issue, please help to file one in case seeing that again.

hi @keyonjie 
I found how can I reproduce that issue.

If I use your patch by using 'git apply xx.patch' rather than 'git am xx.patch', that will occur .

juimonen, 2021-05-26:
@XiaoyunWu6666 are you talking about the 2 commits in #4239? Applying them which ever way should not produce any difference, at least I can't get any differences... are you sure you're applying both 2 commits?

lgirdwood, 2021-05-26:
@juimonen @keyonjie I think this is a case where the allocator needs to accept both cache and uncached aliases in rfree() and convert to the local alias before freeing. Zephyr uses uncached by default.

keyonjie, 2021-05-27:
> @juimonen @keyonjie I think this is a case where the allocator needs to accept both cache and uncached aliases in rfree() and convert to the local alias before freeing. Zephyr uses uncached by default.

@lgirdwood updated in my PR #4239

XiaoyunWu6666, 2021-05-27:
> @XiaoyunWu6666 are you talking about the 2 commits in #4239? Applying them which ever way should not produce any difference, at least I can't get any differences... are you sure you're applying both 2 commits?

I applied the whole patch from #4239

XiaoyunWu6666, 2021-05-27:
@keyonjie  pr #4239 is good to solve this issue.

closed 2021-06-01 by lgirdwood