thesofproject/sof #4259: [BUG] 09_02_TestCreateComplexTopologyPowerFlow Exception SofIpcErrorResponseException

wszypelt, 2021-05-28:
**Describe the bug**
Bug was found in 09_02_TestCreateComplexTopologyPowerFlow

the content of the error:
ERROR:Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':730 FW replied with error. Error code: -12 0x00000005 0x00000000 0x00004000

Last working build: master-2320

**Topology**
```
                pipe_plb
                +--------------------+
    +----------+    | +------+   +-----+ |   +----+
    | Host plb |----->|Buffer|-->| Dai |---->|SSP|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+    |  |
                    pipe_cap                  |  |  x SSP port count
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +----+
    | Host cap |<-----|Buffer|<--| Dai |---->|SSP|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+

    +----------+         +------------------+
    |    OS    |-------->|   AUDIO DEVICE   |------------------------+
    +----------+         +------------------+                        |
                                                                     |
                         +---------------------------+               |
    +-----------------+  | +------+   +------------+ |  +--------+   |
    | Host FIFO_A cap |<---|Buffer|<--| Dai Fifo_A |<---|        |   |
    +-----------------+  | +------+   +------------+ |  |        |   |
                         +---------------------------+  |  DMIC  |<--+
                         +---------------------------+  |        |
    +-----------------+  | +------+   +------------+ |  |        |
    | Host FIFO_B cap |<---|Buffer|<--| Dai Fifo_B |<---|        |
    +-----------------+  | +------+   +------------+ |  +--------+
                         +---------------------------+

                    pipe_plb
                    +--------------------+
    +----------+    | +------+   +-----+ |   +----+
    | Host plb |----->|Buffer|-->| Dai |---->|codec|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+    |  |
                                              |  |
                    pipe_cap                  |  |
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +----+
    | Host cap |<-----|Buffer|<--| Dai |---->|codec|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+

```
Description:
Test for D0/D3 entry with SSP, HDA and DMIC topology preparation
(playback and capture pipeline for every SSP port, playback and capture pipeline for HDA, capture pipeline for DMIC).

**To Reproduce**
Run test with diagnostic driver 09_02_TestCreateComplexTopologyPowerFlow

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: 5d9d2ddad8e0931b7d8c4273db821c9f160b1e88
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP, WHL RVP, 
3) Test environment:
BRANCH: master HASH: f8a5443a0e8242c3f00451ea7887545ccd251573

**Screenshots or console output**
[image]

**Logs**
[2021_05_28_08_54_17.zip](https://github.com/thesofproject/sof/files/6559627/2021_05_28_08_54_17.zip)

wszypelt, 2021-06-08:
Test: 09_01_TestCreateSspTopologyPowerFlow8000Hz32b32b8ch
has the same Error code on [TGL-013].

lgirdwood, 2021-06-08:
@kkarask @wszypelt would you guys also be able to paste the errors from the dma_trace.log (and the ~20 or so lines before the error) in the future on all issues. This allows the developers to triage and synchronize on this issue quicker. Thanks ! e.g.
```
[      240869.661458] (        1812.031250) c0 ipc                  src/ipc/handler-ipc3.c:547  ipc: dai 1.0 -> config 
[      240879.114583] (           9.453125) c0 dai                           src/lib/dai.c:164  dai_get type 1 index 0 new sref 3
[      240907.656250] (          28.541666) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:173  ssp_set_config(), config->format = 0x4304
[      240921.458333] (          13.802083) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:600  ssp_set_config(), sscr0 = 0xc7d0003f, sscr1 = 0xd0700000, ssto = 0x00000000, sspsp = 0x2010005
[      240929.843750] (           8.385417) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:602  ssp_set_config(), sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x07070000, ssioc = 0x00000020
[      240938.072917] (           8.229167) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:604  ssp_set_config(), ssrsa = 0x000000ff, sstsa = 0x000000ff
[      240947.838542] (           9.765625) c0 dai                           src/lib/dai.c:188  dai_put type 1 index 0 new sref 2
[      240956.822917] (           8.984375) c0 ipc                      src/ipc/dai-ipc3.c:159  ipc_comp_dai_config() dai type = 1 index = 0
[      240967.578125] (          10.755208) c0 dai          1.3         src/ipc/dai-ipc3.c:223  dai_config() dai type = 1 index = 0 dd 0x9e0e2000
[      240991.927083] (          24.348959) c0 dai          2.8         src/ipc/dai-ipc3.c:223  dai_config() dai type = 1 index = 0 dd 0x9e0e2100
[      242739.375000] (        1747.447876) c0 ipc                  src/ipc/handler-ipc3.c:547  ipc: dai 1.1 -> config 
[      242748.723958] (           9.348958) c0 dai                           src/lib/dai.c:164  dai_get type 1 index 1 new sref 3
[      242757.994792] (           9.270833) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:173  ssp_set_config(), config->format = 0x4304
[      242769.765625] (          11.770833) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:600  ssp_set_config(), sscr0 = 0xc7d0003f, sscr1 = 0xd0700000, ssto = 0x00000000, sspsp = 0x2010005
[      242778.906250] (           9.140625) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:602  ssp_set_config(), sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x07070000, ssioc = 0x00000020
[      242787.343750] (           8.437500) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:604  ssp_set_config(), ssrsa = 0x000000ff, sstsa = 0x000000ff
[      242797.031250] (           9.687500) c0 dai                           src/lib/dai.c:188  dai_put type 1 index 1 new sref 2
[      242806.640625] (           9.609375) c0 ipc                      src/ipc/dai-ipc3.c:159  ipc_comp_dai_config() dai type = 1 index = 1
[      242820.937500] (          14.296875) c0 dai          3.15        src/ipc/dai-ipc3.c:223  dai_config() dai type = 1 index = 1 dd 0x9e0e2200
[      242844.192708] (          23.255209) c0 dai          4.20        src/ipc/dai-ipc3.c:223  dai_config() dai type = 1 index = 1 dd 0x9e0e2300
[      242855.625000] (          11.432292) c0 dai          4.20        src/ipc/dai-ipc3.c:253  ERROR dai_config(): No memory for dai_config.
[      242863.984375] (           8.359375) c0 ipc                      src/ipc/dai-ipc3.c:184  ERROR ipc_comp_dai_config(): comp_dai_config() failed
Skipped 0 bytes after the last statement. Wrap possible, check the start of the output for later logs.
```

wszypelt, 2023-01-02:
TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt