thesofproject/sof #4266: [BUG] Multi-core DMIC capture I/O error

singalsu, 2021-05-31:
**Describe the bug**
Doing multiple overlapping arecord from DMIC devices ends to Input/output error.

**To Reproduce**
Use topology sof-hda-generic-2ch-dmic16kcore1.tplg from pull request https://github.com/thesofproject/sof/pull/4265 .

Then run this script dmic_arecord_test.sh.

```
$ cat bin/dmic_arecord_test.sh 
#!/bin/sh

set -e

DEV_A="-Dhw:0,6"
DEV_B="-Dhw:0,7"
FMT_A="-f S16_LE -c 2 -r 48000"
FMT_B="-f S16_LE -c 2 -r 16000"

# volume
amixer -c 0 cset name='Dmic0 Capture Switch' on
amixer -c 0 cset name='Dmic0 Capture Volume' 60
#amixer -c 0 cset name='Dmic1 2nd Capture Switch' on
amixer -c 0 cset name='Dmic1 2nd Capture Volume' 60

# separate  recordings
arecord $DEV_A $FMT_A -d 5 rec_a1.wav
arecord $DEV_B $FMT_B -d 5 rec_b1.wav

# overlap
arecord $DEV_A $FMT_A -d 10 rec_a2.wav & PID=$!
sleep 2
arecord $DEV_B $FMT_B -d 5 rec_b2.wav
wait $PID

# overlap
arecord $DEV_B $FMT_B -d 10 rec_b3.wav & PID=$!
sleep 2
arecord $DEV_A $FMT_A -d 5 rec_a3.wav
wait $PID

# overlap
arecord $DEV_A $FMT_A -d 10 rec_a4.wav & PID=$!
sleep 5
arecord $DEV_B $FMT_B -d 10 rec_b4.wav

# overlap
arecord $DEV_B $FMT_B -d 10 rec_b5.wav & PID=$!
sleep 5
arecord $DEV_A $FMT_A -d 10 rec_a5.wav

echo Done.

```
**Reproduction Rate**
Always after starting capturing rec_b2.wav.

**Expected behavior**
The script passes to end without errors.

**Impact**
The error does not happen with normal core0 topologies. Showstopper for multi-core, otherwise annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev 819e4e7eea07
    * SOF: main d62e9267f
2) Name of the topology file
    * Topology: sof-hda-generic-2ch-dmic16kcore1.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL HDA generic 2ch

**Screenshots or console output**
```
Recording WAVE 'rec_a1.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'rec_b1.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
Recording WAVE 'rec_a2.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'rec_b2.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
arecord: pcm_read:2152: read error: Input/output error
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 8000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 32000
TICK_TIME: 0

```

The tracing ends also to an error
`error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)
`

The logs are attached. So far I have tested this only in a TGL notebook.
[dmic_dmesg.txt](https://github.com/thesofproject/sof/files/6571147/dmic_dmesg.txt)
[dmic_etrace.txt](https://github.com/thesofproject/sof/files/6571149/dmic_etrace.txt)
[dmic_trace.txt](https://github.com/thesofproject/sof/files/6571150/dmic_trace.txt)

singalsu, 2021-05-31:
The address EPC1     0xbe02de27 corresponds to these code lines in sof-tgl.lst

```
be02de27:	03b180               	rsr.epc1	a8

	/* now save panic dump */
	/* TODO: we could invoke a GDB stub here */
	panic_dump(SOF_IPC_PANIC_EXCEPTION, NULL, &epc1);
be02de2a:	fffca1               	l32r	a10, be02de1c (dead006 <_core_s_size+0xde9d006>)
be02de2d:	00a0b2               	movi	a11, 0
be02de30:	20c110               	or	a12, a1, a1
```

The disassembly at EPC2     0xbe02d41a looks like this

```
	/* now wait */
	asm volatile("waiti 0");
be02d417:	007000               	waiti	0
}
be02d41a:	f01d                	retw.n

be02d41c <arch_wait_for_interrupt+0x1c>:

```

singalsu, 2021-05-31:
The error happens also with stable-v1.8 2cede3159 . The Stack dump in dmesg is:

```
[   42.464014] sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[   42.464120] sof-audio-pci-intel-tgl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   42.464126] sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[   42.464237] sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception
[   42.464241] sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000
[   42.464246] sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0
[   42.464250] sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[   42.464253] sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   42.464257] sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000024 PS       0x00060125 SAR     0x00000000
[   42.464262] sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe041385 EPC2     0xbe045e12 EPC3     0x00000000 EPC4    0x00000000
[   42.464267] sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0xbe02d40e EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   42.464271] sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00060920
[   42.464275] sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   42.464279] sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe091380
[   42.464284] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091380: be046cf3 be091570 00000001 be046bc4
[   42.464289] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091384: 7e041385 be091400 be091660 be091650
[   42.464293] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091388: d9200d00 55ee648d 000c0800 00000000
[   42.464297] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe09138c: 0dead000 00000000 078d9828 ffff956d
[   42.464301] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091390: c094739c ffffffff 0a699440 ffff956d
[   42.464305] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091394: 814cfb72 ffffffff 08740800 ffff956d
[   42.464309] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091398: 08740800 ffff956d 06ebda00 ffff956d
[   42.464313] sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe09139c: 00000000 00000000 06ebda00 ffff956d

```
EPC1 address points to 

```
/** See comp_ops::trigger */
static inline int comp_trigger(struct comp_dev *dev, int cmd)
{
	int ret = 0;

	assert(dev->drv->ops.trigger);
be041382:	1022d2               	l32i	a13, a2, 64
be041385:	092dd2               	l32i	a13, a13, 36

		return 0;
	}

```

lgirdwood, 2021-05-31:
@singalsu it's a NULL ptr deref and it looks like you are in IRQ level 5.

keyonjie, 2021-06-01:
@singalsu I think we need to fix this first, can you try apply my PR https://github.com/thesofproject/sof/pull/4239 and re-run?
```
ERROR free_block(): invalid heap = 0x9e082380, cpu = 1
```

keyonjie, 2021-06-01:
@singalsu I can reproduce this, and the panic issue will disappear if I change the pipelines from dynamic to static.

```
diff --git a/tools/topology/development/CMakeLists.txt b/tools/topology/development/CMakeLists.txt
index 704d5508bd16..21b91ce98c48 100644
--- a/tools/topology/development/CMakeLists.txt
+++ b/tools/topology/development/CMakeLists.txt
@@ -31,8 +31,8 @@ set(TPLGS_UP
        "sof-cml-rt5682\;sof-cml-eq-fir-rt5682\;-DPLATFORM=cml\;-DHSMICPROC=eq-fir-volume\;-DDMICPROC=eq-iir-volume\;-DDMIC16KPROC=eq-iir-volume"
        "sof-cml-rt5682\;sof-cml-eq-iir-rt5682\;-DPLATFORM=cml\;-DHSEARPROC=eq-iir-volume\;-DDMICPROC=eq-iir-volume\;-DDMIC16KPROC=eq-iir-volume"
        "sof-glk-da7219\;sof-glk-eq-da7219\;-DHEADPHONE=da7219\;-DDMICPROC=eq-iir-volume"
-       "sof-hda-generic\;sof-hda-generic-2ch-dmic16kcore1\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4\;-DDYNAMIC=1\;-DDMIC16KCORE=1"
-       "sof-hda-generic\;sof-hda-generic-2ch-dmic48kcore1\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4\;-DDYNAMIC=1\;-DDMIC48KCORE=1"
+       "sof-hda-generic\;sof-hda-generic-2ch-dmic16kcore1\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4\;-DDMIC16KCORE=1"
+       "sof-hda-generic\;sof-hda-generic-2ch-dmic48kcore1\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4\;-DDMIC48KCORE=1"
        "sof-hda-generic\;sof-hda-generic-tdfb_50mm-2ch\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMIC16KPROC=tdfb-eq-iir-volume\;-DDMIC16KPROC_FILTER1=tdfb/coef_line2_50mm_pm10deg_16khz.m4\;-DDMICPROC=tdfb-eq-iir-volume\;-DDMICPROC_FILTER1=tdfb/coef_line2_50mm_pm10deg_48khz.m4\;-DDMICPROC_FILTER2=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER2=eq_iir_coef_highpass_40hz_20db_16khz.m4"
 )

```

singalsu, 2021-06-01:
@keyonjie Yep, the error does not happen if I remove the -DDYNAMIC=1 from topology build.

lgirdwood, 2021-06-01:
> EPC1 address points to
> 
> ```
> /** See comp_ops::trigger */
> static inline int comp_trigger(struct comp_dev *dev, int cmd)
> {
> 	int ret = 0;
> 
> 	assert(dev->drv->ops.trigger);
> be041382:	1022d2               	l32i	a13, a2, 64
> be041385:	092dd2               	l32i	a13, a13, 36
> 
> 		return 0;
> 	}
> ```

Btw, I suspect ```dev->drv == NULL``` here, so the assertion itself is doing the NULL ptr deref.

lgirdwood, 2021-06-03:
@singalsu can you try with HEAD, @ranj063 has a fix that's now merged.

singalsu, 2021-06-03:
@lgirdwood This I/O error happens with "ipc: ipc3: Do not reset pipeline components during comp_free". There's no more FW panic but ipc timeout:

```
[   59.226053] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30200000 size 44
[   59.226156] sof-audio-pci-intel-tgl 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   59.226163] sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[   59.226272] sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   59.226284] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
[   59.226289] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   59.226304] sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   59.226310] sof-audio-pci-intel-tgl 0000:00:1f.3: error: waking up any trace sleepers
[   59.226327] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget BUF11.0

```

ranj063, 2021-06-04:
@singalsu, we should hold off on testing multi-core alongside dynamic pipelines. Im in the process of tackling bugs with dynamic piplines with single core and it is looking pretty promising. I will post the fixes for multi-core next week.

singalsu, 2021-06-04:
Yep, I added to #4265 build of non-dynamic pipelines. This issue does not happen with them.

lgirdwood, 2021-06-11:
@singalsu any update, can we close ?

singalsu, 2021-06-14:
@lgirdwood  Yes this issue still happens.

keyonjie, 2021-12-07:
suppose this issue should already be fixed with latest main or stable-v2.0?

lgirdwood, 2021-12-08:
Yep, think we fixed on tgl13 or earlier and have not seen on daily. closing.

closed 2021-12-08 by lgirdwood