thesofproject/sof #4268: [BUG] No sound on N4120 tablet (Chuwi Hi10X)

callegar, 2021-05-31:
**Describe the bug**

Cannot seem to be able to get sound working on the Chuwi Hi10X convertible/tablet. It looks like there are two versions of this tablet using the same Hi10X name. The problematic one uses a Celeron N4120 Gemini lake processor and related chipset.
 
Trying with manjaro linux that uses a quite recent kernel and alsa. Audio is not detected apart from the HDMI outputs.

Tried `options snd-intel-dspcfg dsp_driver=1` that does not seem to help.

Alto tried `options snd-intel-dspcfg dsp_driver=3` that creates a lot of activity on dmesg (with many errors). Will attach output here if relevant.

Audio works in Windows and that OS shows "Smart Sound Technology (Intel(R) SST) OED" for the audio part *not attached to HDMI*. The device codes I am getting there are 3198, 0222 and AE34. And for all of them the vendor code is 8086 (intel).

Before proceeding with an attempt to collect more data I would like to know if sound is expected to be working in linux on this hardware or if it is still a work in progress, so that testing is premature.

**To Reproduce**

`aplay -l` only reports HDMI outputs.

**Reproduction Rate**

Always.

**Expected behavior**

Sound should be available via the internal speakers and the audio jack (mic+headphones).

**Impact**

Total showstopper. Only way to use audio on this machine is to use bluetooth headphones.

**Environment**

Unclear to me how to get this for my system that is installed from a distro.

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**

Will provide all that may be needed if audio can be expected to be working on this system.
Unclear to me how to get the sof-logger to debug.

callegar, 2021-06-01:
Moving to a 5.12 kernel, with `snd-intel-dspcfg dsp_driver=3` the firmware appears to load correctly. However, a topology file for idisp only seems to be loaded and the local devices remain missing.

The situation seems similar (to some extent) to the one described in https://askubuntu.com/questions/1330755/no-sound-on-gemini-lake-with-xubuntu, that also refers to geminilake.

kv2019i, 2021-06-01:
@callegar  can you do a NHLT dump and attach the output to this bug? 

```
sudo bash
cat /sys/firmware/acpi/tables/NHLT >NHLT.bin
```

The NHLT.bin might reveal something about this machine. NHLT is a set of tables in ACPI providing the OS (Linux here) information about how the audio peripherals are connected. More info at https://01.org/sites/default/files/595976_intel_sst_nhlt.pdf