thesofproject/sof #4280: [BUG] [ADL][SDW] IO error when bluetooth aplay on ADLP_RVP_SDW

XiaoyunWu6666, 2021-06-02:
**Describe the bug**
 IO error happened when bluetooth aplay on ADLP_RVP_SDW.
happened in internal PR test 4322

**To Reproduce**
 aplay   -Dhw:0,14 -r 8000 -c 1 -f S16_LE -d 10 /dev/zero -v -q 
OR
TPLG=sof-adl-rt711-4ch.tplg ~/sof-test/test-case/check-playback.sh -d 10 -l 1 -r 1

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 819e4e7e
SOF Branch: main
SOF Commit: 19117f620e42
Topology: sof-adl-rt711-4ch.tplg
Platform:  ADLP_RVP_SDW

**Screenshots or console output**
[console]
```
2021-06-02 06:17:12 UTC [REMOTE_COMMAND] aplay   -Dhw:0,14 -r 8000 -c 1 -f S16_LE -d 10 /dev/zero -v -q
Hardware PCM card 0 'sof-soundwire' device 14 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 4000
  period_size  : 1000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1000
  period_event : 0
  start_threshold  : 4000
  stop_threshold   : 4000
  silence_threshold: 0
  silence_size : 0
  boundary     : 9007199254740992000
  appl_ptr     : 0
  hw_ptr       : 0
aplay: pcm_write:2061: write error: Input/output error
2021-06-02 06:17:14 UTC [REMOTE_ERROR] aplay on PCM hw:0,14 failed at 1/1.
```

plbossart, 2021-06-02:
@kv2019i @yongzhi1 any idea what could possibly cause this?

kv2019i, 2021-06-02:
@plbossart @XiaoyunWu6666 This is expected behaviour if the Bluetooth connection is not set up. The I2s is in slave and there will be no WS/FSYNC from the BT side, so audio side will get -EIO in this case.

We'll need to develop some way to cover this in the sof-test cases.

keqiaozhang, 2021-06-03:
@kv2019i ，I will try to disable the BT pipelines test in sof-test and let the chrome team to cover the BT validation.

kv2019i, 2021-06-03:
Ack @keqiaozhang . Should we move this to sof-test issues and track this there?