thesofproject/sof #4301: Audio no longer works with newest arch on Dell Vostro 7500

Footsiefat, 2021-06-05:
Audio used to work on my work laptop but after an update it is once again broken...

"neofetch" output:
```
                   -`                    james@workLaptop 
                  .o+`                   ---------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: Vostro 7500 
               `+oooooo:                 Kernel: 5.10.42-1-lts 
               -+oooooo+:                Uptime: 21 mins 
             `/:-:++oooo+:               Packages: 1301 (pacman) 
            `/++++/+++++++:              Shell: zsh 5.8 
           `/++++++++++++++:             Resolution: 1920x1080 
          `/+++ooooooooooooo/`           DE: Plasma 5.21.5 
         ./ooosssso++osssssso+`          WM: KWin 
        .oossssso-````/ossssss+`         WM Theme: WhiteSur-dark 
       -osssssso.      :ssssssso.        Theme: DeepinDark [Plasma], WhiteSur-dark [GTK2/3] 
      :osssssss/        osssso+++.       Icons: McMojave-circle-dark [Plasma], McMojave-circle-dark [GTK2/3] 
     /ossssssss/        +ssssooo/-       Terminal: konsole 
   `/ossssso+/:-        -:/+osssso+-     CPU: Intel i7-10750H (12) @ 2.600GHz 
  `+sso+:-`                 `.-/+oso:    GPU: NVIDIA GeForce GTX 1650 Ti Mobile 
 `++:.                           `-/+/   GPU: Intel UHD Graphics 
 .`                                 `/   Memory: 2717MiB / 15767MiB 

                                                                 
                                                                 
```

"sudo dmesg | grep 'snd\|soc\|sof'" output

```
[    0.674158] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.674160] software IO TLB: mapped [mem 0x0000000058876000-0x000000005c876000] (64MB)
[    0.704788] Initializing XFRM netlink socket
[    8.082863] systemd[1]: Listening on LVM2 poll daemon socket.
[    8.489602] Bluetooth: HCI socket layer initialized
[    8.489604] Bluetooth: L2CAP socket layer initialized
[    8.489607] Bluetooth: SCO socket layer initialized
[    8.895439] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    8.895576] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    8.895657] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    8.895865] snd_hda_intel 0000:01:00.1: Disabling MSI
[    8.895873] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    9.174527] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    9.174546] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    9.229884] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    9.229907] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    9.229927] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    9.230191] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   10.477859] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   10.484828] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   10.625994] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   10.625997] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   10.626001] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   10.629385] sof-audio-pci 0000:00:1f.3: warning: unknown sof_ext_man header type 6 size 0x20
[   10.629388] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:7:0-47d07
[   10.629390] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:17:0
[   10.629391] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   10.629397] sof-audio-pci 0000:00:1f.3: warning: unknown sof_ext_man header type 3 size 0x30
[   10.629398] sof-audio-pci 0000:00:1f.3: warning: unknown sof_ext_man header type 5 size 0x20
[   10.770405] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:7:0-47d07
[   10.770407] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:17:0
[   10.770408] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   10.896639] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:18:1 Kernel ABI 3:17:0
[   10.896641] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[   10.914547] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   10.982995] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC3204: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[   10.982997] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   10.982998] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   10.982999] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[   10.983000] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[   10.983001] snd_hda_codec_realtek ehdaudio0D0:      Headset Mic=0x19
[   10.983002] snd_hda_codec_realtek ehdaudio0D0:      Headphone Mic=0x1a
[   11.824738] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[   11.824744] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[   11.837359] input: sof-hda-dsp Headphone Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input26
[   11.837396] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input27
[   11.837426] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input28
[   11.837458] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input29
[   17.442495] wlan0: associate with dc:a6:32:21:d9:9e (try 1/3)
[   17.449134] wlan0: RX AssocResp from dc:a6:32:21:d9:9e (capab=0x411 status=0 aid=1)
[   17.454344] wlan0: associated

```

plbossart, 2021-06-07:
@Footsiefat you give us very little information.
a) what was the configuration before the update
b) what does not work?

Please check out https://thesofproject.github.io/latest/getting_started/index.html#debug-audio-issues-on-intel-platforms

FWIW, the Arch distribution had misleading documentation on modprobe quirks in the past, if you applied some of those in the past they may or may not work with the updates.

Footsiefat, 2021-06-08:
@plbossart hey dude, sorry for not giving enough into. Basically I wrote this all out properly but I accidentally closed the tab so I rushed remaking it.

Good news is that after switching from pulseaudio to pipewire audio now works fine again now.

closed 2021-06-08 by Footsiefat