thesofproject/sof #4302: [BUG] Sound on HP Spectre x360 aw2002 with Tiger Lake-H is not working on 1.8-rc2

by-cx, 2021-06-05:
**Describe the bug**

When I was trying to get speakers on my notebook working I noticed that it's working with version 1.7 but not 1.8-rc2, both from sof-bin repository. I cannot see any audio output related to the integrated sound hardware with 1.8-rc2.

My notebook is: HP Spectre x360 aw2002
Which uses CPU i7-1165G7
Here is lspci:

```
0000:00:00.0 Host bridge: Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers (rev 01)
0000:00:02.0 VGA compatible controller: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01)
0000:00:04.0 Signal processing controller: Intel Corporation TigerLake-LP Dynamic Tuning Processor Participant (rev 01)
0000:00:07.0 PCI bridge: Intel Corporation Tiger Lake-LP Thunderbolt 4 PCI Express Root Port #0 (rev 01)
0000:00:07.1 PCI bridge: Intel Corporation Tiger Lake-LP Thunderbolt 4 PCI Express Root Port #1 (rev 01)
0000:00:08.0 System peripheral: Intel Corporation GNA Scoring Accelerator module (rev 01)
0000:00:0d.0 USB controller: Intel Corporation Tiger Lake-LP Thunderbolt 4 USB Controller (rev 01)
0000:00:0d.2 USB controller: Intel Corporation Tiger Lake-LP Thunderbolt 4 NHI #0 (rev 01)
0000:00:0e.0 RAID bus controller: Intel Corporation Volume Management Device NVMe RAID Controller
0000:00:12.0 Serial controller: Intel Corporation Tiger Lake-LP Integrated Sensor Hub (rev 20)
0000:00:14.0 USB controller: Intel Corporation Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller (rev 20)
0000:00:14.2 RAM memory: Intel Corporation Tiger Lake-LP Shared SRAM (rev 20)
0000:00:14.3 Network controller: Intel Corporation Wi-Fi 6 AX201 (rev 20)
0000:00:15.0 Serial bus controller [0c80]: Intel Corporation Tiger Lake-LP Serial IO I2C Controller #0 (rev 20)
0000:00:15.1 Serial bus controller [0c80]: Intel Corporation Tiger Lake-LP Serial IO I2C Controller #1 (rev 20)
0000:00:16.0 Communication controller: Intel Corporation Tiger Lake-LP Management Engine Interface (rev 20)
0000:00:1c.0 PCI bridge: Intel Corporation Tigerlake PCH-LP PCI Express Root Port #6 (rev 20)
0000:00:1d.0 System peripheral: Intel Corporation Device 09ab
0000:00:1f.0 ISA bridge: Intel Corporation Tiger Lake-LP LPC Controller (rev 20)
0000:00:1f.3 Multimedia audio controller: Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller (rev 20)
0000:00:1f.4 SMBus: Intel Corporation Tiger Lake-LP SMBus Controller (rev 20)
0000:00:1f.5 Serial bus controller [0c80]: Intel Corporation Tiger Lake-LP SPI Controller (rev 20)
0000:57:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
10000:e0:1d.0 PCI bridge: Intel Corporation Tiger Lake-LP PCI Express Root Port #9 (rev 20)
10000:e0:1d.2 PCI bridge: Intel Corporation Device a0b2 (rev 20)
10000:e1:00.0 Non-Volatile memory controller: Intel Corporation Device 0975 (rev 03)
10000:e2:00.0 Non-Volatile memory controller: Intel Corporation Device 0975
```

**To Reproduce**

Working combination is:

* SOF 1.7
* Kernel 5.13-rc3
* [This version of alsa-ucm-conf](https://github.com/alsa-project/alsa-ucm-conf/archive/548036b8ecc92d4e8e2070046551fb0b8acb7d96.zip)

If I switch to SOF 1.8-rc2 the problem occurs.

**Reproduction Rate**

Always.

**Expected behavior**

Sound is working from all available outputs.

**Impact**

No sound coming from my notebook.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://bodhi.fedoraproject.org/updates/FEDORA-2021-4ee6ccd27a
    * SOF: https://github.com/thesofproject/sof-bin/commit/0a52f96bf3d5bf7cbd3af49ea26790de298597e0
2) Name of the topology file
    * Topology: sof-tgl-rt711-rt1308-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CannonLake+ (?)

**Screenshots or console output**

* [dmesg](https://gist.github.com/by-cx/de45871d0e4695f97174cb565527b538)
* [alsa-info](http://alsa-project.org/db/?f=7dd5410003137ff0c8aeffebe23392b44c78348e)

plbossart, 2021-06-07:
@thanks @by-cx for letting us know. I don't see anything wrong from the logs.

We'll look into this but can you confirm that you have 1.7 working? I see this in you alsa-info and I wonder if you did a manual install of 1.7?

alsa-sof-firmware-1.6.1-4.fc34.noarch

by-cx, 2021-06-07:
@plbossart That's system package Fedora installs along the other things. I backuped its files and copied version 1.7 from sof-bin repo manually. 1.6.1 doesn't work on my notebook.

[image]

I think the issue could relate to alsa-ucm-conf because the firmware is loaded, topology too and the only problem is I can't see the sound outputs.

plbossart, 2021-06-07:
Thanks @by-cx.
Could you try this:

sof-tplg -> sof-tplg-v1.7
sof->sof-v1.8-rc2

I see that we modified the PCM indices and renamed the PCMs in the topology between 1.7 and 1.8 and I wonder if this impacts UCM.

plbossart, 2021-06-07:
in theory UCM looks for the device number by name
https://github.com/alsa-project/alsa-ucm-conf/blob/a4cd64da90d01dc801b1887a7f835420512d0f17/ucm2/sof-soundwire/dmic.conf#L6

But if you have an earlier version of UCM that may not work. I see that in the initial version I used 
+		CapturePCM "hw:${CardId},3"

A temporary fix on your device could be to edit /usr/share/alsa/ucm2/sof-soundwire/dmic.conf and replace this number 3 by 10. Once you have the update to alsa-ucm-conf 1.2.5 things should work out of the box.