thesofproject/sof #4304: [BUG][ADL] DSP panic when multiple-pause-resume on ADLS_RVP_HDA

XiaoyunWu6666, 2021-06-07:
**Describe the bug**
 DSP panic when multiple-pause-resume on ADLS_RVP_HDA
found in daily-ADL id 4417

**To Reproduce**
TPLG=sof-hda-generic-2ch.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 3b025d22
SOF Branch: main
SOF Commit: d6e29ced2430
Topology: sof-hda-generic-2ch.tplg
Platform: ADLS_RVP_HDA

**Screenshots or console output**
[dmesg]

```
2021-06-07 07:22:09 UTC [ERROR] Caught kernel log error
===========================>>
[  149.670965] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[  149.670998] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[  149.671092] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception
[  149.671094] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000
[  149.671097] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0
[  149.671099] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[  149.671102] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  149.671104] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00010100 PS       0x00060725 SAR     0x00000000
[  149.671107] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe0495f2 EPC2     0xbe045cda EPC3     0x00000000 EPC4    0x00000000
[  149.671110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  149.671112] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  149.671115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  149.671118] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe091120
[  149.671121] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091120: be04ead6 00060b20 00000010 be09139c
[  149.671124] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091124: 7e0495f2 be0911a0 00000002 00010100
[  149.671126] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091128: ca6be100 0fbc84c8 000c0800 00000000
[  149.671129] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe09112c: 0dead000 00000000 51630028 ffff9cff
[  149.671131] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091130: c07be39c ffffffff 418f0c40 ffff9cff
[  149.671134] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091134: aaccfb72 ffffffff 41e5ec00 ffff9cff
[  149.671136] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe091138: 41e5ec00 ffff9cff 514d5f40 ffff9cff
[  149.671138] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe09113c: 00000000 00000000 514d5f40 ffff9cff

```

XiaoyunWu6666, 2021-06-07:
Also happened on TGLH_RVP_HDA which uses sof-hda-generic-4ch.tplg

kv2019i, 2021-06-07:
FYI @singalsu  and @ranj063 

Many dmic patches were merged on Friday. It does seem this only triggers on platforms with dynamic-pipelines enabled though but the error still occurs with the dmic pipeline:

```
[  149.574664] sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.10.DMIC0.IN freed
[  149.574695] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 6 dir 1
[  149.669904] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 4
[  149.670698] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x2024001e successful
[  149.670719] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  149.670965] sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
```

singalsu, 2021-06-07:
I wonder if this is duplicate https://github.com/thesofproject/sof/issues/4266

ranj063, 2021-06-07:
no this is not related to #4266. THis is with singel core

mengdonglin, 2021-06-08:
@XiaoyunWu6666 Can you please check if this issue can be reproduced with static pipelines?

XiaoyunWu6666, 2021-06-08:
@mengdonglin Yes I switch to stable-v1.8 branch and the issue doesn't reproduce

singalsu, 2021-06-08:
Does #4313 fix this?

keyonjie, 2021-06-10:
looks the case pass in all those 3 platforms with today's result 4487, @XiaoyunWu6666 shall we close it?

XiaoyunWu6666, 2021-06-10:
https://github.com/thesofproject/sof/issues/4328 looks similar but on different platforms and pipelines.
Since the case pass in all those 3 platforms with today's result 4487, let's close this issue

closed 2021-06-10 by XiaoyunWu6666