thesofproject/sof #4305: [BUG][APL]buffer_alloc() failed when  arecording in multiple-pipeline-all on APL_UP2_NOCODEC

XiaoyunWu6666, 2021-06-07:
**Describe the bug**
when  arecording in multiple-pipeline-all on APL_UP2_NOCODECs,  buffer_alloc() occurs 
arecord failed ,showing  'unable to install hw params'

Strangely if I make arecord ahead of aplay , this won't happen. which means if you try 'TPLG=sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f c -c 20 ', nothing wrong occurs.

**To Reproduce**
TPLG=sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 [playback first and then arecord]

**Reproduction Rate**
100%
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 3b025d22
SOF Branch: main
SOF Commit: d6e29ced2430
 Topology:  sof-apl-nocodec.tplg
 Platform: APL_UP2_NOCODEC

**Screenshots or console output**
[console]
```
2021-06-07 08:11:53 UTC [INFO] ===== Testing: (Loop: 1/1) =====
2021-06-07 08:11:53 UTC [INFO] /home/ubuntu/sof-test/test-case/multiple-pipeline.sh will use topology /lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg to run the test case
2021-06-07 08:11:53 UTC [INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital' in test case
2021-06-07 08:11:53 UTC [INFO] Run command to get pipeline parameters
2021-06-07 08:11:53 UTC [COMMAND] sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg -f 'type:playback' -b ' pcm:HDA Digital' -s 0 -e
2021-06-07 08:11:53 UTC [INFO] Testing: Port0 [hw:0,0]
2021-06-07 08:11:53 UTC [COMMAND] aplay   -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-06-07 08:11:53 UTC [INFO] Testing: Port1 [hw:0,1]
2021-06-07 08:11:53 UTC [COMMAND] aplay   -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-06-07 08:11:53 UTC [INFO] Testing: Port3 [hw:0,3]
2021-06-07 08:11:53 UTC [COMMAND] aplay   -D hw:0,3 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-06-07 08:11:53 UTC [INFO] Testing: Port5 [hw:0,5]
2021-06-07 08:11:53 UTC [COMMAND] aplay   -D hw:0,5 -c 2 -r 48000 -f S16_LE /dev/zero -q
2021-06-07 08:11:53 UTC [INFO] Testing: Media Playback 7 [hw:0,7]
2021-06-07 08:11:53 UTC [COMMAND] aplay   -D hw:0,7 -c 2 -r 8000 -f S16_LE /dev/zero -q
2021-06-07 08:11:53 UTC [INFO] /home/ubuntu/sof-test/test-case/multiple-pipeline.sh will use topology /lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg to run the test case
2021-06-07 08:11:53 UTC [INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital' in test case
2021-06-07 08:11:53 UTC [INFO] Run command to get pipeline parameters
2021-06-07 08:11:53 UTC [COMMAND] sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg -f 'type:capture' -b ' pcm:HDA Digital' -s 0 -e
2021-06-07 08:11:54 UTC [INFO] Testing: Port0 [hw:0,0]
2021-06-07 08:11:54 UTC [COMMAND] arecord   -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/null -q
2021-06-07 08:11:54 UTC [INFO] Testing: Port1 [hw:0,1]
2021-06-07 08:11:54 UTC [COMMAND] arecord   -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/null -q
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 02021-06-07 08:11:54 UTC [INFO] Testing: Port3 [hw:0,3]

2021-06-07 08:11:54 UTC [COMMAND] arecord   -D hw:0,3 -c 2 -r 48000 -f S16_LE /dev/null -q
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
```
[dmesg]
[dmesg_apl.txt](https://github.com/thesofproject/sof/files/6607116/dmesg_apl.txt)
[sof-logger]
```
[ 13489906315.573868] (           0.000000) c0 dma-trace             src/trace/dma-trace.c:335  ERROR FW ABI 0x3012001 DBG ABI 0x5003000 tag v1.8-rc1-162-gd6e29ced2430 src hash 0xaa1a3d92 (ldc hash 0xaa1a3d92)
[      205234.575178] (      205234.578125) c0 buffer                   src/audio/buffer.c:60   ERROR buffer_alloc(): could not alloc size = 768 bytes of type = 113
[      205240.512678] (           5.937500) c0 ipc                   src/ipc/ipc3/helper.c:623  ERROR ipc_buffer_new(): buffer_new() failed
[      205246.137678] (           5.625000) c0 ipc                  src/ipc/ipc3/handler.c:1208 ERROR ipc: pipe 4 buffer 24 creation failed -12
[      219258.116287] (       14011.978516) c0 buffer                   src/audio/buffer.c:60   ERROR buffer_alloc(): could not alloc size = 768 bytes of type = 113
[      219264.210037] (           6.093750) c0 ipc                   src/ipc/ipc3/helper.c:623  ERROR ipc_buffer_new(): buffer_new() failed
[      219269.574620] (           5.364583) c0 ipc                  src/ipc/ipc3/handler.c:1208 ERROR ipc: pipe 8 buffer 36 creation failed -12
[      233286.032397] (       14016.458008) c0 buffer                   src/audio/buffer.c:60   ERROR buffer_alloc(): could not alloc size = 768 bytes of type = 113
[      233291.917813] (           5.885417) c0 ipc                   src/ipc/ipc3/helper.c:623  ERROR ipc_buffer_new(): buffer_new() failed
[      233297.490730] (           5.572917) c0 ipc                  src/ipc/ipc3/handler.c:1208 ERROR ipc: pipe 12 buffer 48 creation failed -12
[      247168.636012] (       13871.145508) c0 buffer                   src/audio/buffer.c:60   ERROR buffer_alloc(): could not alloc size = 768 bytes of type = 113
[      247174.573511] (           5.937500) c0 ipc                   src/ipc/ipc3/helper.c:623  ERROR ipc_buffer_new(): buffer_new() failed
[      247180.198511] (           5.625000) c0 ipc                  src/ipc/ipc3/handler.c:1208 ERROR ipc: pipe 13 buffer 53 creation failed -12
[     1252269.429406] (     1005089.250000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 34376 mask 63 val 0 us 937
[     1253942.866839] (        1673.437378) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 32840 mask 63 val 0 us 937
[     1255658.752188] (        1715.885376) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[     1256987.450052] (        1328.697876) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 35400 mask 63 val 0 us 937
```

marc-hb, 2021-06-07:
@ranj063 This looks similar to #4273 that was fixed by https://github.com/thesofproject/linux/pull/2964

Sorry if it's unrelated.

ranj063, 2021-06-07:
no this is not related to 4273. We;re failing with -ENOMEM here @keyonjie any insights?

XiaoyunWu6666, 2021-06-08:
can be reproduced on ubuntu@sh-apl-up2-nocodec-04

lgirdwood, 2021-06-08:
@ranj063  @keyonjie We are probably forgetting to free something in the pipeline. Two options

1) cmocka can run on host we should be able to use/extend it to load topologies and run valgrind at the same time. leaks found !!
2) load this topology in testbench (agian under valgrind). Leaks found !!

closed 2021-06-10 by lgirdwood