thesofproject/sof #4309: [BUG][SDW]no PCM in topology for HDMI converter when hda_dsp_hdmi_build_controls, on tglh-0a5e-sdw

XiaoyunWu6666, 2021-06-08:
**Describe the bug**
in daily test 4434 , verify-kernel-boot-log failed on TGLH-0A5E-SDW.
when creating new PCM HDMI , error occured

**To Reproduce**
 TPLG=sof-tgl-rt711-rt1308-rt715.tplg ~/sof-test/test-case/verify-kernel-boot-log.sh
**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: d07af131
SOF Branch: main
SOF Commit: d6e29ced2430
 Topology: sof-tgl-rt711-rt1308-rt715.tplg
 Platform:TGLH_0A5E_SDW

**Screenshots or console output**
[dmesg]
```
[    4.754027] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: creating new PCM HDMI 1
[    4.754028] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spcm: allocate Passthrough Playback 5 playback DMA buffer size 0x10000 max 0x10000
[    4.754043] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: creating new PCM HDMI 2
[    4.754044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spcm: allocate Passthrough Playback 6 playback DMA buffer size 0x10000 max 0x10000
[    4.754068] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: creating new PCM HDMI 3
[    4.754069] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: spcm: allocate Passthrough Playback 7 playback DMA buffer size 0x10000 max 0x10000
[    4.754123] kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[    4.754126] kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 4
[    4.754128] kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 5
[    4.754130] kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 6
[    4.754132] kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 7
[    4.754133] kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 8
[    4.766395] kernel: SDW: Invalid device for paging :0
[    4.766403] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766411] kernel: SDW: Invalid device for paging :0
[    4.766413] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766425] kernel: SDW: Invalid device for paging :0
[    4.766426] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766430] kernel: SDW: Invalid device for paging :0
[    4.766431] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766435] kernel: SDW: Invalid device for paging :0
[    4.766436] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766440] kernel: SDW: Invalid device for paging :0
[    4.766441] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766445] kernel: SDW: Invalid device for paging :0
[    4.766446] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766450] kernel: SDW: Invalid device for paging :0
[    4.766451] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
[    4.766832] kernel: rt715 sdw:3:025d:0715:00: [rt715_sdw_write] 3607 <= 0000
[    4.766906] kernel: rt715 sdw:3:025d:0715:00: [rt715_sdw_write] 3608 <= 0000
[    4.766957] kernel: rt715 sdw:3:025d:0715:00: [rt715_sdw_write] 3609 <= 0000
[    4.767016] kernel: rt715 sdw:3:025d:0715:00: [rt715_sdw_write] 3627 <= 0000
[    4.767018] kernel: SDW: Invalid device for paging :0
[    4.767020] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at snd_soc_component_update_bits on sdw:2:025d:1308:00: -22
```

kv2019i, 2021-06-08:
The "kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 7" prints are normal and will not cause errors. I'll remove "Display Audio" and replace with Soundwire tag. This seems to be a Soundwire specific issue.

plbossart, 2021-06-08:
@XiaoyunWu6666 can you check for a possible ACPI issue? I've seen this sort of errors when we have a bad ACPI config.

The error says 
````
[    4.766403] kernel: rt1308 sdw:2:025d:1308:00: ASoC: error at soc_component_read_no_lock on sdw:2:025d:1308:00: -22
````

but the only rt1308 is this:
````
[    3.738711] kernel: rt1308 sdw:1:025d:1308:00: rt1308_io_init m_btl_l=0xffe2, m_btl_r=0x25
````

There is zero trace of any rt1308 activity on link 2 (sdw:2 prefix).