thesofproject/sof #4310: [BUG] Glitches and interrupted recording on channels 3 and 4 with Mux component

kkarask, 2021-06-08:
**Describe the bug**
Glitches was found in test 14_11_TestMuxSimpleStreamOperations48000Hz32b32b4ch. 
While pipelines are working pause and resume operation are done on playback stream. During this operation recording was interrupted on channels 3 and 4.

**Topology**
```
 pipe_plb
    +----------------------+
    | +-------+   +------+ |
    | | Host1 |-->| Buff |----+
    | +-------+   +------+ |  |      pipe_plb
    +----------------------+  |  +--------------------------------+
                              +---->+-----+   +------+   +-----+  |     +------+
                                 |  | Mux |-->| Buff |-->| Dai |------->| SSPx |
        pipe_plb              +---->+-----+   +------+   +-----+  |     +------+
    +----------------------+  |  +--------------------------------+        |
    | +-------+   +------+ |  |                                            |
    | | Host2 |-->| Buff |----+                                            |
    | +-------+   +------+ |                                               |
    +----------------------+                                               |
                                                                           |
                                                                           |
        pipe_cap                                                           |
    +-------------------------------+                                      |
    | +------+   +------+   +-----+ |   +------+                           |
    | | Host |<--| Buff |<--| Dai |<----| SSPx |<--------------------------<
    | +------+   +------+   +-----+ |   +------+
    +-------------------------------+
```

**To Reproduce**
Run test 14_11_TestMuxSimpleStreamOperations48000Hz32b32b4ch with Diagnostic driver

**Reproduction Rate**
Sporadic

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: d6e29ced243079e69beb055730a33ad103b377d7
    
2) Name of the platform(s) on which the bug is observed.
    * Platform: APL RVP
    * 
3) Test environment:
    BRANCH: master HASH: a71a6d87b75a0f64e2924fae30a31ed8a0208014

**Screenshots or console output**

[image]

[image]

[14_11_TestMuxSimpleStreamOperations48000Hz32b32b4ch.zip](https://github.com/thesofproject/sof/files/6614141/14_11_TestMuxSimpleStreamOperations48000Hz32b32b4ch.zip)

slawblauciak, 2021-06-08:
@kkarask @wszypelt can you please attempt to bisect this, so we have a rough idea what might be causing this?

Also, does this occur on any other platforms?

lgirdwood, 2021-06-08:
@slawblauciak the interesting part is that the glitch switches from chans 1, 2 - > 3 4 after 40ms, could be a result of host22 starting ?

kkarask, 2021-06-09:
@slawblauciak yes, issue exists only on APL. This bug was hidden and it is hard to find good version. Here 71ac0f97e9d98df7e5679fbb6669bb07229d4170 reproduction rate is 3/100.

wszypelt, 2021-07-28:
Glitches are visible in test 14_00_TestDemuxEchoSynchronized48000Hz32b32b4ch without start/stop operations . The topology is the same.

ICL RVP platform (near the start of the capture:)

[image]

lgirdwood, 2021-08-03:
@wszypelt are you guys seeing anything in the FW logs here ? Is it possible to attach the logs ?

wszypelt, 2021-08-03:
@lgirdwood 
This is very sporadic issue:
on SOF_FW_MASTER 01.01.00.2506 
hash: 9648fb65a5b8b86707e0a160efa992f3f3c97f59
Reproduction 1/75
**Logs:** 
[2021_07_27_03_32_35.zip](https://github.com/thesofproject/sof/files/6924734/2021_07_27_03_32_35.zip)

New Stress test's on: SOF_FW_MASTER 01.01.00.2520 **without issue**
hash: d38c3c39129149c1e0adf3501b7f950ba3f0611c

lgirdwood, 2021-08-03:
No errors in the logs, warning only point to capture DAI running out of data at stream start so would not explain the glitches at 100ms above.
```
[      381961.171875] (      326292.406250) c0 pipe         4.13  ....../pipeline-stream.c:184  pipe trigger cmd 1
[      381983.802083] (          22.630209) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:1059 ssp_trigger() cmd 1
[      381993.020833] (           9.218750) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:876  ssp_pre_start()
[      382003.411458] (          10.390625) c0 mn                 ./drivers/intel/ssp/mn.c:258  mclk_rate 38400000, mclk_source_clock 1
[      382011.822917] (           8.411458) c0 mn                 ./drivers/intel/ssp/mn.c:220  mclk_id 0 mdivr_val 1
[      382022.291667] (          10.468750) c0 mn                 ./drivers/intel/ssp/mn.c:328  find_mn for freq 38400000 bclk 4096000
[      382032.005208] (           9.713542) c0 mn                 ./drivers/intel/ssp/mn.c:367  find_mn m 64 n 75
[      382041.640625] (           9.635417) c0 mn                 ./drivers/intel/ssp/mn.c:625  bclk_rate 4096000, *out_scr_div 8, m 64, n 75
[      382051.250000] (           9.609375) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:215  ssp_bclk_prepare_enable(): sscr0 = 0xc7d00777
[      382061.223958] (           9.973958) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:970  ssp_start(): SSE set for SSP0
[      382070.000000] (           8.776042) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:974  ssp_start()
[      382127.057292] (          57.057293) c0 ll-schedule        ./schedule/ll_schedule.c:390  task add 0xbe080200 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[      382135.364583] (           8.307292) c0 ll-schedule        ./schedule/ll_schedule.c:394  task params pri 0 flags 0 start 0 period 1000
[      382147.760417] (          12.395833) c0 ll-schedule        ./schedule/ll_schedule.c:306  new added task->start 553209747 at 553194660
[      382156.067708] (           8.307292) c0 ll-schedule        ./schedule/ll_schedule.c:309  num_tasks 3 total_num_tasks 3
[      382567.057292] (         410.989594) c0 dw-dma                 src/drivers/dw/dma.c:1048 dw_dma_avail_data_size() size is 0!
[      382577.187500] (          10.130208) c0 dai          4.11           src/audio/dai.c:957  WARN dai_copy(): nothing to copy
[      382588.333333] (          11.145833) c0 host         4.10          src/audio/host.c:381  no bytes to copy, 0 avail in buffer, 1024 free in DMA
[      383565.546875] (         977.213562) c0 dw-dma                 src/drivers/dw/dma.c:1048 dw_dma_avail_data_size() size is 0!
[      383575.000000] (           9.453125) c0 dai          4.11           src/audio/dai.c:957  WARN dai_copy(): nothing to copy
[      383585.520833] (          10.520833) c0 host         4.10          src/audio/host.c:381  no bytes to copy, 0 avail in buffer, 1024 free in DMA
[      384565.234375] (         979.713562) c0 dw-dma                 src/drivers/dw/dma.c:1048 dw_dma_avail_data_size() size is 0!
[      384574.583333] (           9.348958) c0 dai          4.11           src/audio/dai.c:957  WARN dai_copy(): nothing to copy
[      384585.234375] (          10.651042) c0 host         4.10          src/audio/host.c:381  no bytes to copy, 0 avail in buffer, 1024 free in DMA
[      384953.802083] (         368.567719) c0 pipe         1.2   ....../pipeline-stream.c:184  pipe trigger cmd 1
[      384968.177083] (          14.375000) c0 mux          3.6        src/audio/mux/mux.c:703  mux_trigger(), command = 1
[      384981.562500] (          13.385417) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:1059 ssp_trigger() cmd 1
[      384991.536458] (           9.973958) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:876  ssp_pre_start()
[      385001.171875] (           9.635417) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:970  ssp_start(): SSE set for SSP0
[      385010.182292] (           9.010417) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:974  ssp_start()
[      385067.005208] (          56.822918) c0 ll-schedule        ./schedule/ll_schedule.c:390  task add 0xbe080080 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[      385075.312500] (           8.307292) c0 ll-schedule        ./schedule/ll_schedule.c:394  task params pri 0 flags 0 start 0 period 1000
[      385087.864583] (          12.552083) c0 ll-schedule        ./schedule/ll_schedule.c:306  new added task->start 553324947 at 553307565
[      385096.250000] (           8.385417) c0 ll-schedule        ./schedule/ll_schedule.c:309  num_tasks 4 total_num_tasks 4
[      385106.015625] (           9.765625) c0 ll-schedule        ./schedule/ll_schedule.c:390  task add 0xbe0800c0 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[      385113.567708] (           7.552083) c0 ll-schedule        ./schedule/ll_schedule.c:394  task params pri 1 flags 0 start 0 period 1000
[      385125.026042] (          11.458333) c0 ll-schedule        ./schedule/ll_schedule.c:306  new added task->start 553324947 at 553309005
[      385133.151042] (           8.125000) c0 ll-schedule        ./schedule/ll_schedule.c:309  num_tasks 5 total_num_tasks 5
[      385569.036458] (         435.885406) c0 dw-dma                 src/drivers/dw/dma.c:1048 dw_dma_avail_data_size() size is 0!
[      385578.828125] (           9.791667) c0 dai          4.11           src/audio/dai.c:957  WARN dai_copy(): nothing to copy
[... cut ...]
[     3616678.645833] (           9.609375) c0 ll-schedule        ./schedule/ll_schedule.c:336  num_tasks 5 total_num_tasks 5
[     3619318.359375] (        2639.713623) c0 pipe         2.5   ....../pipeline-stream.c:184  pipe trigger cmd 2
[     3619328.854167] (          10.494792) c0 mux          3.6        src/audio/mux/mux.c:703  mux_trigger(), command = 2
[     3619339.505208] (          10.651042) c0 dw-dma                 src/drivers/dw/dma.c:370  dw_dma_pause(): dma 0 channel 0 pause
[     3619348.723958] (           9.218750) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:1059 ssp_trigger() cmd 2
```