thesofproject/sof #4312: [BUG] Exception SofIpcErrorResponseException

wszypelt, 2021-06-08:
**Describe the bug**
Bug was found in 09_01_TestCreateSspTopologyPowerFlow8000Hz32b32b8ch

the content of the error:
ERROR:Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':730 FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000

Last working build: master-2320

**Topology**
```
                pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSP|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |  x SSP port count
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSP|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+
```
Description:
Description:
Test for D0/D3 entry with SSP topology preparation (playback and capture pipeline for every SSP port available on platform).

**To Reproduce**
Run test with diagnostic driver 09_01_TestCreateSspTopologyPowerFlow8000Hz32b32b8ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: d6e29ced243079e69beb055730a33ad103b377d7
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP
3) Test environment:
BRANCH: master HASH: ffe0639e5c22d19aaa83629b55831487ca84010d

**Screenshots or console output**
[image]

**Logs**
[2021_06_08_08_30_06.zip](https://github.com/thesofproject/sof/files/6615681/2021_06_08_08_30_06.zip)

lgirdwood, 2021-06-08:
```
[      116977.421875] (        1313.802124) c0 dma-trace                 src/trace/trace.c:133  Suppressed 6 similar messages:   block %d base 0x%x size %d
[      116984.270833] (           6.848958) c0 dma-trace                 src/trace/trace.c:133  Suppressed 6 similar messages:    count %d free %d
[      116990.703125] (           6.432292) c0 pipe         8.31  ......./pipeline-graph.c:300  pipeline complete, clock freq 400000000Hz
[      118848.437500] (        1857.734375) c0 ipc                  src/ipc/ipc3/handler.c:547  ipc: dai 1.3 -> config 
[      118857.942708] (           9.505208) c0 dai                           src/lib/dai.c:164  dai_get type 1 index 3 new sref 3
[      118867.421875] (           9.479167) c0 ssp-dai      1.3   /drivers/intel/ssp/ssp.c:173  ssp_set_config(), config->format = 0x4304
[      118879.531250] (          12.109375) c0 ssp-dai      1.3   /drivers/intel/ssp/ssp.c:600  ssp_set_config(), sscr0 = 0xc7d0003f, sscr1 = 0xd0700000, ssto = 0x00000000, sspsp = 0x2010005
[      118888.750000] (           9.218750) c0 ssp-dai      1.3   /drivers/intel/ssp/ssp.c:602  ssp_set_config(), sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x07070000, ssioc = 0x00000020
[      118897.083333] (           8.333333) c0 ssp-dai      1.3   /drivers/intel/ssp/ssp.c:604  ssp_set_config(), ssrsa = 0x000000ff, sstsa = 0x000000ff
[      118906.979167] (           9.895833) c0 dai                           src/lib/dai.c:188  dai_put type 1 index 3 new sref 2
[      118916.458333] (           9.479167) c0 ipc                      src/ipc/ipc3/dai.c:159  ipc_comp_dai_config() dai type = 1 index = 3
[      118934.843750] (          18.385416) c0 dai          7.25        src/ipc/ipc3/dai.c:223  dai_config() dai type = 1 index = 3 dd 0x9e0e4200
[      118958.463542] (          23.619791) c0 dai          8.29        src/ipc/ipc3/dai.c:223  dai_config() dai type = 1 index = 3 dd 0x9e0e4800
[      121988.697917] (        3030.234375) c0 component             src/ipc/ipc3/helper.c:422  comp new host <8b9d100c-6d78-418f-90a3-e0e805d0852b> type 1 id 9.32
[      122010.312500] (          21.614584) c0 dma                           src/lib/dma.c:121  dma_get() ID 5 sref = 5 busy channels 0
[      123621.484375] (        1611.171875) c0 buffer                src/ipc/ipc3/helper.c:886  buffer new size 0x200 id 9.34 flags 0x0
[      125573.932292] (        1952.447876) c0 component             src/ipc/ipc3/helper.c:422  comp new dai <c2b00d27-ffbc-4150-a51a-245c79c5e54b> type 2 id 9.33
[      125594.479167] (          20.546875) c0 dai                   src/ipc/ipc3/helper.c:429  ERROR comp_new(): unable to create the new component
[      125602.630208] (           8.151042) c0 ipc                   src/ipc/ipc3/helper.c:829  ERROR ipc_comp_new(): component cd = NULL
Skipped 0 bytes after the last statement. Wrap possible, check the start of the output for later logs.
```

wszypelt, 2023-01-02:
unsupported platform
on TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt