thesofproject/sof #4314: [BUG][TGL-13] Expression check failed, TestPtpSimpleStreamOperations

wszypelt, 2021-06-08:
**Describe the bug**

Wave file without glitch

Bug was found in:
02_12_TestPtpSimpleStreamOperations8000Hz16b16b1ch
02_12_TestPtpSimpleStreamOperations8000Hz24b32b4ch
02_12_TestPtpSimpleStreamOperations8000Hz32b32b8ch
02_12_TestPtpSimpleStreamOperations16000Hz16b16b1ch
02_12_TestPtpSimpleStreamOperations16000Hz32b32b4ch
02_12_TestPtpSimpleStreamOperations24000Hz16b16b1ch
02_12_TestPtpSimpleStreamOperations24000Hz24b32b4ch
02_12_TestPtpSimpleStreamOperations24000Hz32b32b8ch
02_12_TestPtpSimpleStreamOperations48000Hz16b16b8ch
02_12_TestPtpSimpleStreamOperations48000Hz24b32b4ch
02_12_TestPtpSimpleStreamOperations48000Hz32b32b2ch

the content of the error:
Expression check failed Expected: 1.01 Got: 3.01 & 5.03 & 7.03

**Topology**
```
                pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+

```
Description:
Simultaneous playback and capture on external loopback (same SSP).
While pipelines are working pause and resume operation are done on playback stream.

**To Reproduce**
Run test with diagnostic driver 02_12_TestPtpSimpleStreamOperations8000Hz16b16b1ch and other

**Reproduction Rate**
~90%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: 9a35f65111b9f6603d8ed26a6178511e6cebaa75
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL RVP - [TGL-13]
3) Test environment:
BRANCH: master HASH: a71a6d87b75a0f64e2924fae30a31ed8a0208014

**Screenshots or console output**

[image]

[image]

[image]

**Logs**
[2021_06_02_15_53_33.zip](https://github.com/thesofproject/sof/files/6616715/2021_06_02_15_53_33.zip)

lgirdwood, 2021-06-08:
@slawblauciak @wszypelt this looks like a slim driver or CI script issue if no glitches are observed ?

antonirx, 2022-10-14:
Reproduced on TGL-SSP
5594ad3512830feaa478677718036ffabf51a454

antonirx, 2022-10-21:
Reproduced on TGL:
c807d6844b535ff4ad90ac6429a2fef46f398cb4

wszypelt, 2022-12-30:
Test issue

closed 2022-12-30 by wszypelt