thesofproject/sof #4317: [BUG][DMIC]TGLU_RVP_NOCODEC failed to enable target core for widget PCM5C

XiaoyunWu6666, 2021-06-09:
**Describe the bug**
found in internal daily test 4457 on  TGLU_RVP_NOCODEC， in case verify-kernel-boot-log
unable to set up DMIC1 pipeline 

**To Reproduce**
TPLG=sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/verify-kernel-boot-log.sh
OR 
dmesg

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: c2e91674
SOF Branch: main
SOF Commit: 5abd8a9ea572
 Topology: sof-tgl-nocodec-ci.tplg
Platform: TGLU_RVP_NOCODEC

**Screenshots or console output**
[dmesg]
```
[    3.492213] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    3.492426] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    3.492434] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    3.493061] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    3.493063] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget DMIC1.IN setup complete
[    3.493067] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    3.493207] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    3.493208] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.8.DMIC1.IN setup complete
[    3.493211] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to enable target core: -2 for widget PCM5C
[    3.493327] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
[    3.493518] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[    3.493520] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    3.493529] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -2
[    3.493539] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    3.493546] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    3.493578] kernel: sof-nocodec sof-nocodec: ASoC: failed to instantiate card -22
[    3.493737] kernel: sof-nocodec: probe of sof-nocodec failed with error -22
```
[TPLG png]
[image]

keyonjie, 2021-06-09:
Thanks for reporting @XiaoyunWu6666 , new PR https://github.com/thesofproject/sof/pull/4318 submitted to address this.

XiaoyunWu6666, 2021-06-09:
> Thanks for reporting @XiaoyunWu6666 , new PR #4318 submitted to address this.

#4318 fixed it

closed 2021-06-09 by keyonjie