thesofproject/sof #4324: Undefined AM_GNU_GETTEXT when building alsa-utils [WSL2 Ubuntu 20.04 LTS (Windows)]

edward-a, 2021-06-09:
I am following Getting Started instructions to build SOF on WSL2 Ubuntu 20.04 LTS (Windows) and i was able to successfully compile ALSA libraries. I am however stuck on compiling alsa-utils with the following errors:

configure.ac:10: error: possibly undefined macro: AM_GNU_GETTEXT
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:11: error: possibly undefined macro: AM_GNU_GETTEXT_VERSION

Here below is the complete output after executing ./gitcompile

configure.ac:10: warning: macro 'AM_GNU_GETTEXT' not found in library
configure.ac:11: warning: macro 'AM_GNU_GETTEXT_VERSION' not found in library
./gitcompile: line 10: gettextize: command not found
configure.ac:15: installing './compile'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
alsactl/Makefile.am: installing './depcomp'
axfer/test/Makefile.am:9: warning: source file '../container.c' is in a subdirectory,
axfer/test/Makefile.am:9: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
axfer/test/Makefile.am:9: warning: source file '../container-riff-wave.c' is in a subdirectory,
axfer/test/Makefile.am:9: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:9: warning: source file '../container-au.c' is in a subdirectory,
axfer/test/Makefile.am:9: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:9: warning: source file '../container-voc.c' is in a subdirectory,
axfer/test/Makefile.am:9: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:9: warning: source file '../container-raw.c' is in a subdirectory,
axfer/test/Makefile.am:9: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../container.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../container-riff-wave.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../container-au.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../container-voc.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../container-raw.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../mapper.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../mapper-single.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
axfer/test/Makefile.am:20: warning: source file '../mapper-multiple.c' is in a subdirectory,
axfer/test/Makefile.am:20: but option 'subdir-objects' is disabled
parallel-tests: installing './test-driver'
configure.ac:10: error: possibly undefined macro: AM_GNU_GETTEXT
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:11: error: possibly undefined macro: AM_GNU_GETTEXT_VERSION
CFLAGS=-O2 -Wall -pipe -g
./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
./configure: line 2937: syntax error near unexpected token `external'
./configure: line 2937: `AM_GNU_GETTEXT(external)'

paulstelian97, 2021-06-10:
Did you set the LD_LIBRARY_PATH to use the newly compiled ALSA libraries? The system ones are invariably older than the ones you need.

lgirdwood, 2021-06-10:
@edward-a also worth trying your local alsa libraries and utils as code is frequently upstreamed to ALSA. We do have a copy for code that is not upstream or not yet available in a ALSA release (so it's possible you have all dependencies).
Fwiw, I dont really recommend LTS releases for desktop SW development reasons as LTS releases can fall quickly behind with developer tools / packages and become stale.