thesofproject/sof #4329: [BUG][BYT] Recurring beeps when DMA trace is not used

plbossart, 2021-06-10:
**Describe the bug**
Users such as @jwrdegoede, @HoshiYamazaki and @fxthomas reported 'beeps' that occur regularly with SOF.
This does not seem to happen when the DMA trace is used.

A suggested workaround is to use 
options snd_sof sof_debug=1
in /etc/modprobe.d/alsa-base.conf

**To Reproduce**
Take SOF release and use in a distro

**Reproduction Rate**
Difficult to assess, but 3 independent reports show something is seriously wrong.

**Expected behavior**
valid audio rendered on speakers/headset

**Impact**
showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.10+
    * SOF: release 1.7
2) Name of the topology file
    * Topology: sof-byt* (they are all the same)
3) Name of the platform(s) on which the bug is observed.
    * Platform: Baytrail and Cherrytrail

plbossart, 2021-06-10:
links:
https://github.com/thesofproject/sof/issues/3868#issuecomment-850845612

lgirdwood, 2021-06-10:
Does trace have to be enabled only or need to be captured by sof-logger. i.e. it it enough that the trace is DMAed to host buffer only.

plbossart, 2021-06-10:
> Does trace have to be enabled only or need to be captured by sof-logger. i.e. it it enough that the trace is DMAed to host buffer only.

it's the latter as I understand it. @fxthomas please correct me it I am wrong.

plbossart, 2021-06-10:
Closing, this is a duplicate of https://github.com/thesofproject/sof/issues/3868 which I missed to to missing labels. same problem with DMA trace.

closed 2021-06-10 by plbossart