thesofproject/sof #4330: [BUG] Glitch in wav file with DMIC interface FIFO B

kkarask, 2021-06-10:
**Describe the bug**
Glitches was found in test 05_04_TestDmicFifoBSimpleStreamOperator48000hz32b32b4ch

**Topology**
```
 +----------+      +------------------+
    |    OS    |----->|   AUDIO DEVICE   |-----------------+
    +----------+      +------------------+                 |
                                                           |
                  +---------------------------+            |
    +----------+  | +------+   +------------+ |  +------+  |
    | Host cap |<---|Buffer|<--| Dai Fifo_B |<---| DMIC |<-+
    +----------+  | +------+   +------------+ |  +------+
                  +---------------------------+
```

**To Reproduce**
Run test 05_04_TestDmicFifoBSimpleStreamOperator48000hz32b32b4ch with Diagnostic driver
with parameters:
--playback_iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: d6e29ced243079e69beb055730a33ad103b377d7
   
2) Name of the platform(s) on which the bug is observed.
    * Platform: APL RVP

**Screenshots or console output**
[image]

[image]

Logs:
[05_04_TestDmicFifoBSimpleStreamOperator48000hz32b32b4ch.zip](https://github.com/thesofproject/sof/files/6631015/05_04_TestDmicFifoBSimpleStreamOperator48000hz32b32b4ch.zip)

lgirdwood, 2021-06-10:
@kkarask I assume this works with FIFO A ? Can you stress ?

kkarask, 2021-06-14:
@lgirdwood yes, it works with FIFO A. I run stress tests and reproduction rate is 5/200.

lgirdwood, 2021-06-14:
@singalsu any difference here between A (working) and B (glitch 5/200) ?

singalsu, 2021-06-15:
@lgirdwood The A and B paths in HW are identical except DMA configuration.

@kkarask If the configuration of A and B is swapped in topology, does the issue swap too?

kkarask, 2021-06-15:
@singalsu In that test we have configured only FIFO B and there are glitches. 
We have another test where only FIFO A is configured and it works well. 
We have also a test where FIFO A and FIFO B are configured and the problem doesn't exist.

singalsu, 2021-06-15:
I was thinking a case where FIFO A is configured to 16 kHz, does the FIFO A make then glitches?

keyonjie, 2021-06-16:
@lgirdwood we have run out the 1.8 time, so this have to be pushed out to 1.9.

kkarask, 2021-06-17:
@singalsu In case where FIFO A is configured as 16 kHz problem doesn't occur.

wszypelt, 2022-12-30:
unsupported platform
on TGL issue does not occur
Tested FW hash: https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74

closed 2022-12-30 by wszypelt