thesofproject/sof #4353: [BUG]  [v1.8] glitch in test 02_19_TestPipelineCreateDelete16000Hz32b32b8ch

kkarask, 2021-06-15:
**Describe the bug**
Glitch was found at the beginning of the wav file in test 02_19_TestPipelineCreateDelete16000Hz32b32b8ch

**Topology**
```
pipe_plb
                        +--------------------+
        +----------+    | +------+   +-----+ |   +----+
        | Host plb |----->|Buffer|-->| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+    |  |
                                                  |  |
                        pipe_cap                  |  |
                        +--------------------+    |  |
        +----------+    | +------+   +-----+ |   +----+
        | Host cap |<-----|Buffer|<--| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+
```

**To Reproduce**
Run test 02_19_TestPipelineCreateDelete16000Hz32b32b8ch with Diagnostic driver
with parameters:
--playback_iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * Branch: main, Hash: 9e7a8f921efba5979129950c64bca835f8ce982f, v1.8
    
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP

**Screenshots or console output**
[image]

[02_19_TestPipelineCreateDelete16000Hz32b32b8ch.zip](https://github.com/thesofproject/sof/files/6653913/02_19_TestPipelineCreateDelete16000Hz32b32b8ch.zip)

lgirdwood, 2021-06-15:
@kkarask when we do see glitches with slim driver, are they spread evenly through the file or always near the start of the test. i.e. above I can see the glitches near the start of the capture.

keyonjie, 2021-06-16:
@lgirdwood push out to 1.9?

wszypelt, 2022-12-30:
unsupported platform

closed 2022-12-30 by wszypelt

antonirx, 2023-01-05:
on TGL issue does not occur
Tested FW hash: 4943aaa